realsenseai/librealsense #9340: Mapping a 2D texture point into a pointcloud index and vice versa

goloste, 2021-07-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { Intel RealSense L515 } | 
| Firmware Version                   | { 01.05.01.03 } | 
| Operating System & Version | {Linux Ubuntu 20.04.2 LTS} | 
| Kernel Version (Linux Only)    |  (5.8.0-59-generic) | 
| Platform                                 | Dell XPS-13-9350 |
| Language                            |  {python} |  

### Issue Description
I want to generate a point cloud, and be able to  locate every pixel in texture in the pointcloud.
In other words, given a set of 2D [x, y] coordinates in the texture image coordinate system, i want to retrieve their correspondent index in the pointcloud.

To illustrate the problem i Prepared the snippet below:

    # Import Modules
    import pyrealsense2 as rs
    import open3d as o3d
    import numpy as np
    import cv2
    
    # Configure camera streams
    pipeline = rs.pipeline()
    config = rs.config()
    config.enable_stream(rs.stream.color, 1920, 1080, rs.format.rgb8, 30)
    config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
    
    # Create an align object
    # rs.align allows us to perform alignment of depth frames to others frames
    # The "align_to" is the stream type to which we plan to align depth frames.
    align_to = rs.stream.color
    align = rs.align(align_to)
    
    # Start streaming
    profile = pipeline.start(config)
    
    # Main Loop
    while True:
        # Get frames
        frames = pipeline.wait_for_frames()                     # Get frames from pipeline
    
        aligned_frames = align.process(frames)                  # Align depth frame to rgb frame
    
        aligned_depth_frame = aligned_frames.get_depth_frame()  # Get aligned depth frame
        aligned_color_frame = aligned_frames.get_color_frame()  # Get aligned color frame
    
        img = np.asarray(aligned_color_frame.get_data())    # Get aligned RGB image
        img = cv2.cvtColor(img, cv2.COLOR_RGB2BGR)          # Change color mapping from RGB to BGR
        cv2.namedWindow("Video", cv2.WINDOW_GUI_EXPANDED)   # Opens a custom sized window to show image
        cv2.imshow("Video", img)                            # Shows image in window
    
        # Exits Loop by pressing 'ESC' or 'Q'
        k = cv2.waitKey(7) % 0x100
        if k == 27 or k == 10:
            break
    # Stop streaming
    pipeline.stop()
    
    # Generate pointcloud
    
    pc = rs.pointcloud()                        # Create Pintcloud object
    pc.map_to(aligned_color_frame)              # Use color frame for texturing
    points = pc.calculate(aligned_depth_frame)  # Compute pointcloud by depth frame
    points.export_to_ply("issue_cloud.ply", aligned_color_frame)  # Save with texturing provided by aligned_color_frame
    
    texture = np.asarray(aligned_color_frame.get_data())    # Retrieve texture as a 2D image
    texture = cv2.cvtColor(texture, cv2.COLOR_RGB2BGR)      # Change color mapping from RGB to BGR
    
    # Draw a red square on retrieved texture with edge l and upper left vertex in (x, y)
    x = 1000                                                 # Set x coordinate
    y = 700                                                  # Set y coordinate
    l = 100                                                  # Set edge
    texture[y:y+l, x:x+l, :] = [0.0, 0.0, 255.0]             # Set pixel x,y value as pure red
    
    cv2.namedWindow("Texture", cv2.WINDOW_GUI_EXPANDED)     # Opens a custom sized window to show image
    cv2.imshow("Texture", texture)                          # Shows texture in window
    cv2.waitKey()                                           # Close window by pressing any key
    
    loaded_pcd = o3d.io.read_point_cloud("issue_cloud.ply") # Load saved pointcloud
    o3d.visualization.draw_geometries([loaded_pcd])         # Show loaded cloud in viewer
    
    print("Shape of texture image [y, x]: ", [texture.shape[0], texture.shape[1]])
    print(loaded_pcd)

The Main Loop produces a window displaying the currently captured video stream (Note taht the camera is phisically tilted -90° along is z axis)

[image]

When pressing 'Esc' execution breaks from main loop and saves the point cloud shown below:

[image]

As you can see the pointcloud consists in a list of 858308 points, while the color texture is a 1920x1080 image.
  
[image]

After the pointcloud has been created i retrieve the 2D color image used as texture and draw a red square starting
from the upper left vertex locatex in [x, y] = [1000, 700]

[image]

Is there any method to get the pointcloud indexes corresponding to the red square pixels?

MartyG-RealSense, 2021-07-05:
You could perhaps use the instruction **RS2_PROJECT_COLOR_PIXEL_TO_DEPTH_PIXEL** to convert a single pixel in the color frame within the red area to a depth pixel instead of aligning the whole image.  Example code for this instruction in the Python language can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008

[image]

goloste, 2021-07-08:
Hi @MartyG-RealSense thank you for your answer, but I'm failing to understand how to retrieve the corresponding pointcloud index by RS2_PROJECT_COLOR_PIXEL_TO_DEPTH_PIXEL method:
If I understand correctly, when computing
    depth_point = rs.rs2_project_color_pixel_to_depth_pixel(
       aligned_depth_frame.get_data(), depth_scale, depth_min, depth_max,
       depth_intrin, color_intrin, depth_to_color_extrin, color_to_depth_extrin, color_point)

depth_point now contains an array [y, x] of the corresponding coordinates in the depth image.

I was trying to get the corresponding point in pointcloud:
say a given point [x, y] = [1000, 700], corresponds to point pcd.points[2301] in pointcloud. How do I retrieve index 2301 given  coordinates [x, y]?

You meant there is a method to retreive this index out of depth image coordinates?
What am I missing?

Thank you

MartyG-RealSense, 2021-07-08:
Hi @goloste  I was suggesting to use **RS2_PROJECT_COLOR_PIXEL_TO_DEPTH_PIXEL** to convert a red color-pixel to a depth pixel on the **2D color image** that you retrieved.  

I wonder whether this is an over-complicated approach though.  If you know the approximate coordinates where the red square would be then you should be able to check those coordinates directly on the point cloud.  If you are usin the **pc.map_to** approach to generating a point cloud in Python then you could perhaps retrieve the **vertices** of the point cloud (see the link below).

https://github.com/IntelRealSense/librealsense/issues/1653

MartyG-RealSense, 2021-07-14:
Hi @goloste  Do you require further assistance with this case, please?  Thanks!

goloste, 2021-07-14:
After reading about your vertices method, maybe I found what I was Looking for in this comment: https://github.com/IntelRealSense/librealsense/issues/1783#issuecomment-392536795

It seems the method to map from pixel (x, y) to pointcloud index is
    pcd[y * IMAGE_WIDTH + x]

But I still have to test the method, I will post an update eventually

MartyG-RealSense, 2021-07-14:
Okay, thanks very much @goloste for this update - it's excellent to hear that you may have made progress.  Please do update again after your testing.   Good luck!