realsenseai/librealsense #9341: Very low precision, is there something wrong with my camera?

JerryGadd, 2021-07-06:
Hi All,
I have an L515, and using the Intel RealSense Viewer (v2.43.0), i get really poor precision of depth.
If i shut my Office Curtains andset to No Ambient Light, it improves a tiny amount, but still much much poorer than i would expect. Is there something wrong with myh device or settings?
See attached image.

----------------------------------------------------------------------------------------------------
| Camera Model                       | L515
| Firmware Version                   | 01.05.05.00 
| Operating System & Version | Win 10 
| Platform                                 | PC 

[image]

MartyG-RealSense, 2021-07-06:
Hi @JerryGadd  Could you disable Depth and RGB in the 3D point cloud mode, and then enable Depth first and RGB second.  This should result in a **textured point cloud** where RGB color information is mapped onto the depth points.

Although you have RGB enabled in the image that you kindly supplied, it has no **RGB** color (the color in the image is depth points that are color-shaded according to their depth values.  There are a couple of ways in which this kind of RGB-less image could be produced in the Viewer's 3D point cloud mode. 

1.  If you had Depth and RGB enabled in the Viewer's 2D mode and then switched to 3D mode.

2.  If you entered 3D mode and then enabled RGB first and depth secondly.  If Depth is enabled first and RGB second in 3D mode then the RGB image should automatically be mapped onto the depth points.

JerryGadd, 2021-07-07:
HI Marty
I am not worried about the RGB, I am concerned about the accuracy/noise…
My image is very spikey… not smooth at all… nothing like what I see from examples from online searches

https://www.google.com/search?q=realsense+viewer+images+for+l515&tbm=isch&ved=2ahUKEwjCzfft29DxAhUoJLcAHSHzAUwQ2-cCegQIABAA&oq=realsense+viewer+images+for+l515&gs_lcp=CgNpbWcQDFAAWABg3fhOaABwAHgAgAEAiAEAkgEAmAEAqgELZ3dzLXdpei1pbWc&sclient=img&ei=d33lYIK2M6jI3LUPoeaH4AQ&bih=1038&biw=2048&rlz=1C1GCEA_enBE932BE932

***@***.***D77323.C2DD94B0]

Siemens Industry Software Limited is a limited company registered in England and Wales.
Registered number: 3476850.
Registered office: Faraday House, Sir William Siemens Square, Frimley, Surrey, GU16 8QD.

MartyG-RealSense, 2021-07-07:
Would it be possible to provide an RGB image of the scene that the camera is observing please so that I can see whether there are factors in the location that may be affecting the depth image?

Could you also try disabling the two **GLSL** options in the settings of the Viewer please to see whether that makes a positive difference.  Instructions for performing this procedure can be found in the link below.

JerryGadd, 2021-07-07:
Hi Marty,
GLSL settings were already disabled.
Here is the 2D View, that includes a 2D image.
***@***.***D77342.A2FC52C0]
FYI, I have just upgraded to FW to 1.5.8.1, and downloaded Intel RealSense Viewer 2.45.0

MartyG-RealSense, 2021-07-07:
Thanks very much for checking the GLSL settings.

Next, please return to the settings interface and try resetting the Viewer software's defaults by (1) selecting the **General** option tab, (2) clicking on the **Restore Defaults** button, and finally (3) clicking on the **Apply** button.

JerryGadd, 2021-07-07:
Interesting… I have just noticed something…
I just started the Viewer and turned on Depth Camera and the precision was more like what I expected for a period of a few second, and then it jumped to being terrible.
Restoring Defaults made no difference.

MartyG-RealSense, 2021-07-07:
When the L515 is enabled, there is a period of 4 or 5 seconds before the **APD Temperature** under the 'Controls' category of the 'L500 Depth Sensor' options updates from its default of '-9999'.  When my own L515 first updates, it has a temperature of around 27 or 28.  What does your own APD temperature read as after 5 seconds please?

[image]

JerryGadd, 2021-07-07:
It running at around ‘38.5’ and rizing

[Image removed by sender. image]<https://user-images.githubusercontent.com/41145062/124790077-c459ff00-df42-11eb-8f61-89e0a5518b89.png>

-----------------

MartyG-RealSense, 2021-07-07:
It is normal for the L515's temperatures to rise over time after the camera is enabled.  A temperature of 35 does not seem excessive enough to cause problems so soon after streaming commences.

What is the depth image like in the Viewer's 2D mode?

JerryGadd, 2021-07-07:
Same as, smooth (on occasion at start) then jumping to very noisy.

MartyG-RealSense, 2021-07-07:
Please check whether the **Temporal Filter** post-processing filter is responsible by left-clicking on the blue icon beside the **Post-Processing** option to turn the icon from blue (On) to red (off).

[image]

MartyG-RealSense, 2021-07-13:
Hi @JerryGadd  Do you require further assistance with this case, please?  Thanks!

JerryGadd, 2021-07-13:
HI Marty
I am still struggling, but it believe it may be being caused by ambient light. Is there a way to confirm?
I have the camera in my office, and during the day, even with the curtains closed, I have the problem.
However I tried it the other evening when it was dark outside, and I did not.
Kind Regards
Jerry Gadd

MartyG-RealSense, 2021-07-13:
The L515 is an indoor camera that is suited to controlled lighting conditions where the light level remains consistent instead of varying and there is not sunlight coming through a window (as it is disrupted by infrared sources).

Do you have any devices near the L515 that might be emitting infrared, such as a RealSense 400 Series camera?

closed 2021-07-26 by MartyG-RealSense