realsenseai/librealsense #9344: Intel D455 undetected by Jetson Xavier NX

RicLincio, 2021-07-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    | Jetpack 4.4 (L4T 32.4.3) | 
| Platform                                 | NVIDIA Jetson Xavier NX |
| SDK Version                            | 2.45.0 |
| Language                            | - | 
| Segment			| Robot |

After following the guide on [installation for Jetson](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) for patching the kernel and building with:
```shell
cmake ../ \
    -DBUILD_EXAMPLES=off \
    -DBUILD_GRAPHICAL_EXAMPLES=off \
    -DBUILD_PYTHON_BINDINGS=on \
    -DBUILD_WITH_CUDA=true \
    -DCMAKE_BUILD_TYPE=release \
    -DCMAKE_CUDA_COMPILER=/usr/local/cuda/bin/nvcc \
    -DFORCE_RSUSB_BACKEND=off \
    -DOpenGL_GL_PREFERENCE=GLVND
```
when I plug my D455, neither `realsense-viewer` nor `rs-enumerate-devices` are able to detect the camera.
On the same setup, my T265 is detected.
Also, D455 is detected on PC with the same version of librealsense, so it seems to be a problem specific to Jetson.

MartyG-RealSense, 2021-07-06:
Hi @RicLincio  The build flags **-DBUILD_EXAMPLES** and **-DBUILD_GRAPHICAL_EXAMPLES** determine whether the SDK's examples and tools (including realsense-viewer) and the support for them are built or not when librealsense is installed.  If the flags are set to false then the SDK's examples and tools are not included in the build.

If you were building for a 'headless' configuration (a computer without a display screen) then you could set **-DBUILD_EXAMPLES** to true to include support for text based non-graphical examples and tools such as rs-enumerate-devices, and set **-DBUILD_GRAPHICAL_EXAMPLES** to false so that examples and tools that use graphics (such as the Viewer) are not built.  Otherwise, if you want the Viewer to be included in the build then set DBUILD_GRAPHICAL_EXAMPLES to true.

RicLincio, 2021-07-07:
Hello @MartyG-RealSense, you're right.
Unfortunately I made I mistake in reporting the cmake configuration. The actual command I used is
```shell
cmake ../ \
    -DBUILD_EXAMPLES=on \
    -DBUILD_GRAPHICAL_EXAMPLES=on \
    -DBUILD_PYTHON_BINDINGS=on \
    -DBUILD_WITH_CUDA=true \
    -DCMAKE_BUILD_TYPE=release \
    -DCMAKE_CUDA_COMPILER=/usr/local/cuda/bin/nvcc \
    -DFORCE_RSUSB_BACKEND=off \
    -DOpenGL_GL_PREFERENCE=GLVND
```
otherwise, as you said yourself, I wouldn't be able to open `realsense-viewer` or `rs-enumerate-devices`, which I can do: without those, I wouldn't have been able to report the real problem, i.e. the undetected D455.

MartyG-RealSense, 2021-07-07:
As you are building with CMake, it may be worth looking at the Jetson installation guide in the link below that installs librealsense from source code along with the Python wrapper and targets Python 3.6.  

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

This guide has worked consistently for Jetson users, including Xavier NX, and once solved a case (in the link below) where D455 was not being detected.

https://github.com/IntelRealSense/librealsense/issues/7766#issuecomment-726882630

RicLincio, 2021-07-08:
> As you are building with CMake, it may be worth looking at the Jetson installation guide in the link below that installs librealsense from source code along with the Python wrapper and targets Python 3.6 (if you are not using Python 3.6, edit the '3'6' number in the procedure to the Python version number that you are using).
> 
> #6964 (comment)

I followed this guide already, without success. The main point here is just updating cmake. I tried with version 3.13.0 as indicated at point one, but nothing changed.

> This guide has worked consistently for Jetson users, including Xavier NX, and once solved a case (in the link below) where D455 was not being detected.
> 
> #7766

The detection problem seems to be solved with the same guide above, so this is not an additional way of solving the issue.

MartyG-RealSense, 2021-07-08:
Hi @RicLincio  Can you confirm whether your D455 detection problem was solved using https://github.com/IntelRealSense/librealsense/issues/7766 please?

RicLincio, 2021-07-13:
> Hi @RicLincio Can you confirm whether your D455 detection problem was solved using #7766 please?

No, my problem has not been solved with this guide. In fact that is not even a guide, just a bunch of tips, none of which seem to apply to my case. My compilation seems to be successful since the executables are working with T265, but the problem with D455 is still there.

MartyG-RealSense, 2021-07-14:
Do you have the D455 and T265 plugged into the Jetson at the same time?  If so, can the D455 be detected if you unplug the T265 and boot the Jetson only with the D455?

RicLincio, 2021-07-15:
> Do you have the D455 and T265 plugged into the Jetson at the same time? If so, can the D455 be detected if you unplug the T265 and boot the Jetson only with the D455?

No, I tried with both and with each singularly, the result is always the same: T265 detected, D455 not.

MartyG-RealSense, 2021-07-15:
Is it able to detect D455 if you follow the official Jetson guide exactly (for example, not changing the CMake process from the one described in the guide).

```
sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev -y
./scripts/setup_udev_rules.sh  
mkdir build && cd build  
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true && make -j$(($(nproc)-1)) && sudo make install
```

Although the above process does not install Python support, at this point the most important goal is to achieve detection of the D455 at all.

You could also test installing with Debian packages using the information for doing so in the guide instead of building with CMake if you have not tried doing so already.

MartyG-RealSense, 2021-07-21:
Hi @RicLincio  Do you require further assistance with this case, please?  Thanks!

RicLincio, 2021-07-21:
> Is it able to detect D455 if you follow the official Jetson guide exactly (for example, not changing the CMake process from the one described in the guide).
>
> ```
> sudo apt-get install git libssl-dev [...]

Yes, please, but let me try this out first.

MartyG-RealSense, 2021-07-21:
Okay, no problem @RicLincio if you require more time for further testing.  Thanks for the update!

RicLincio, 2021-07-21:
> Is it able to detect D455 if you follow the official Jetson guide exactly (for example, not changing the CMake process from the one described in the guide).
>
> ```
> sudo apt-get install git libssl-dev [...]

This strategy with standard installation is not working. The same conditions are met: T265 is detected, D455 is not, regardless of having only one or both plugged.
When launching realsense-viewer, I collect these messages on the terminal, after connecting the D455:
```
user@xavier:~$ realsense-viewer 
 21/07 10:24:46,471 WARNING [547792506656] (ds5-factory.cpp:1127) DS5 group_devices is empty.
 21/07 10:25:25,518 ERROR [547792506656] (types.h:315) get_dev_capabilities: Cannot open '/dev/video1 Last Error: Device or resource busy
 21/07 10:25:25,519 ERROR [547792506656] (types.h:315) get_dev_capabilities: Cannot open '/dev/video2 Last Error: Device or resource busy
 21/07 10:25:25,520 ERROR [547792506656] (types.h:315) get_dev_capabilities: Cannot open '/dev/video0 Last Error: Device or resource busy
 21/07 10:25:25,531 WARNING [547792506656] (ds5-factory.cpp:1127) DS5 group_devices is empty.
 21/07 10:25:30,547 WARNING [547792506656] (ds5-factory.cpp:1127) DS5 group_devices is empty.
```

RicLincio, 2021-07-21:
> You could also test installing with Debian packages using the information for doing so in the guide instead of building with CMake if you have not tried doing so already.

This doesn't solve the problem. The same behaviour as above is observed.

MartyG-RealSense, 2021-07-21:
Do the results on Jetson change if you downgrade to firmware **5.12.13.50** on the D455, which is one version older than the currently newest one and is also the recommended firmware for SDK 2.45.0?

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

guglielmodev, 2021-07-21:
Same problem here with official librealsense version 2.47 and D455 on Jetson Xavier NX

MartyG-RealSense, 2021-07-22:
@RicLincio @RicLincio  Does the USB insertion sound effect in Ubuntu ever trigger if you unplug the camera, wait a couple of seconds and then re-insert the connector, repeating this up to 10 times?

RicLincio, 2021-07-22:
> Do the results on Jetson change if you downgrade to firmware **5.12.13.50** on the D455, which is one version older than the currently newest one that you have installed in the camera and is also the recommended firmware for SDK 2.45.0?
> 
> https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

No, results are the same.

RicLincio, 2021-07-22:
> @RicLincio @RicLincio Does the USB insertion sound effect in Ubuntu ever trigger if you unplug the camera, wait a couple of seconds and then re-insert the connector, repeating this up to 10 times?

No idea, I don't have an audio device available. I think it is more relevant checking the system messages with `dmesg`
```
[   46.720629] usb 2-3.3: new SuperSpeed USB device number 4 using tegra-xusb
[   46.742114] usb 2-3.3: New USB device found, idVendor=8086, idProduct=0b5c
[   46.742136] usb 2-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   46.742144] usb 2-3.3: Product: Intel(R) RealSense(TM) Depth Camera 455 
[   46.742156] usb 2-3.3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 455 
[   46.742214] usb 2-3.3: SerialNumber: 032623060566
[   46.749737] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
[   46.758960] uvcvideo: Unable to create debugfs 2-4 directory.
[   46.759209] uvcvideo 2-3.3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[   46.759434] uvcvideo 2-3.3:1.0: Entity type for entity Processing 2 was not initialized!
[   46.759580] uvcvideo 2-3.3:1.0: Entity type for entity Camera 1 was not initialized!
[   46.760040] input: Intel(R) RealSense(TM) Depth Ca as /devices/3610000.xhci/usb2/2-3/2-3.3/2-3.3:1.0/input/input6
[   46.761577] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
[   46.765715] uvcvideo: Unable to create debugfs 2-4 directory.
[   46.765941] uvcvideo 2-3.3:1.3: Entity type for entity Processing 7 was not initialized!
[   46.766150] uvcvideo 2-3.3:1.3: Entity type for entity Extension 8 was not initialized!
[   46.766304] uvcvideo 2-3.3:1.3: Entity type for entity Camera 6 was not initialized!
[   54.216338] usb 2-3.3: usb_suspend_both: status 0
[   54.272289] usb 2-3: usb_suspend_both: status 0
[   54.272535] usb usb2: usb_suspend_both: status 0
```

MartyG-RealSense, 2021-07-22:
Another RealSense user who had a non-detecting camera on Ubuntu (an L515, not a D455) reported today that their problem was solved if they disabled USB auto-suspend.

https://github.com/IntelRealSense/librealsense/issues/9455#issuecomment-884781316

RicLincio, 2021-07-23:
> Another RealSense user who had a non-detecting camera on Ubuntu (an L515, not a D455) reported today that their problem was solved if they disabled USB auto-suspend.
> 
> [#9455 (comment)](https://github.com/IntelRealSense/librealsense/issues/9455#issuecomment-884781316)

The command `sudo sed -i 's/GRUB_CMDLINE_LINUX_DEFAULT="/&usbcore.autosuspend=-1 /' /etc/default/grub` fails because there's no file `/etc/default/grub`. Any tip on how to avoid this?

MartyG-RealSense, 2021-07-23:
I will tag the author of the advice into this discussion.

Hi @Consultaing  Thanks very much again for your advice about disabling auto-suspend.  Could you kindly provide @RicLincio above with advice on how they might avoid an error regard **/etc/default/grub** being absent when trying to recreate the advice, please?  Thanks!

Consultaing, 2021-07-23:
The Jetson platform uses a Linux distribution for ARM architecture. The ARM architecture has no BIOS and therefore does not use GRUB. What it usually uses are some configuration .conf files in the boot folder. In these files you have to check if the automatic suspension flag exists. (This depends on the capabilities of the hardware and if the distribution contains the option)

dgr1992, 2021-07-28:
I faced the same problem that realsense-viewer and python were not detecting the Intel RealSense D455 on my Jetson Nano 4GB running Jetpack 4.5 and Ubuntu 18.04. I finally managed to fix the problem an here how i got it working:

1. Either clone or get the latest source code release zip/tar of the librealsense
2. If you got the zip/tar extract it, the name of the folder will be something like librealsens-v{version_number}
3. change into the directory
4. Install all the required dependencies: sudo ./scripts/installDependencies.sh
5. Next create a build directory and change into it: 
`
mkdir build
`
`
cd build
`
7.  The next step is to prepare some environment variables guiding to the cuda directory. Note: change cuda-10.2 according your installation
`
export CUDACXX=/usr/local/cuda-10.2/bin/nvcc
export PATH=$PATH:/usr/local/cuda/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/cuda/lib64
`
8. No it is time to call cmake
`
cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true
`
9.  Next is a very time consuming step, the actual compiling. I set it to use two cores.
`
make -j2
`
10.  After this finished it is time to copy all the files to its destinations which is done by
`
sudo make install
`
11. Then apply the kernel patch. Therefore go up one level 
`
cd ..
`
and call the script
`
./scripts/patch-realsense-ubuntu-L4T.sh
`
12. It turned out that there are some copy commands missing in the "make install" which cause Python to allow `import pyrealsense2 as rs` but when calling a method getting an not implemented error.  In the directory
`
librealsense-{version}/build/wrappers/python
`
(replace {version} with the version number if from zip/tar extracted) are some .so files an links which need to be moved to the 
`
/usr/local/lib
`
copy the files "pybackend2.cpython-36m-aarch64-linux-gnu.so.{versionnumber}" and "pyrealsense2.cpython-36m-aarch64-linux-gnu.so.{versionnumber}". Inside the /usr/local/lib now create the links.
`
sudo ln -s pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.47.0 pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.47
sudo ln -s pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.47 pyrealsense2.cpython-36m-aarch64-linux-gnu.so
sudo ln -s pybackend2.cpython-36m-aarch64-linux-gnu.so.2.47.0 pybackend2.cpython-36m-aarch64-linux-gnu.so.2
sudo ln -s pybackend2.cpython-36m-aarch64-linux-gnu.so.2 pybackend2.cpython-36m-aarch64-linux-gnu.so
` 
Note: change the version number according to your version
13.  Now some paths need to be added which I did by adding following lines to the .bashrc
`
export PATH=$PATH:~/.local/bin
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2
`
After adding do a
`
source .bashrc
`
14. The last step was to add udev rules to give access to the camera
`
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger
`

If you now run "rs-enumerate-devices" the camera should be listed. Running realsens-viewer should open the application and the camera should be detected.

To check if the python library is working I run the python-tutorial-1-depth.py in the folder wrappers/python/examples.

Hope this helps!

MartyG-RealSense, 2021-07-28:
Thanks so much @dgr1992 for sharing your guide with the RealSense community  :)

yzkzero, 2021-08-02:
@dgr1992 I think why you can find device in Jetson is because you are using RSUSB. And patching kernel (run patch-realsense-ubuntu-L4T.sh) is not necessary in your case.

If you want to use V4L (-DFORCE_RSUSB_BACKEND=false) as @RicLincio tried, you have to patch the kernel. Otherwise, camera device cannot be found in Jetson.

dgr1992, 2021-08-02:
> @dgr1992 I think why you can find device in Jetson is because you are using RSUSB. And patching kernel (run patch-realsense-ubuntu-L4T.sh) is not necessary in your case.
>
> If you want to use V4L [...]

I didn't know that, thanks for letting me know!

guglielmodev, 2021-08-03:
I can confirm that this issue with D455 on Jetson Xavier NX still has not a solution for me.
Moreover I have same problem with D435i and D515, while T265 is recognized.

`rs-enumerate-devices` says "No device detected"

Tried also with multiple cables and usb ports.

dmesg output with D435i:
```
[  422.755463] usb 2-3.4: new SuperSpeed USB device number 8 using tegra-xusb
[  422.776137] usb 2-3.4: New USB device found, idVendor=8086, idProduct=0b3a
[  422.776148] usb 2-3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  422.776155] usb 2-3.4: Product: Intel(R) RealSense(TM) Depth Camera 435i
[  422.776161] usb 2-3.4: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[  422.776167] usb 2-3.4: SerialNumber: 903223053376
[  422.780370] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  422.789735] uvcvideo: Unable to create debugfs 2-8 directory.
[  422.789934] uvcvideo 2-3.4:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  422.790144] uvcvideo 2-3.4:1.0: Entity type for entity Processing 2 was not initialized!
[  422.790285] uvcvideo 2-3.4:1.0: Entity type for entity Camera 1 was not initialized!
[  422.790686] input: Intel(R) RealSense(TM) Depth Ca as /devices/3610000.xhci/usb2/2-3/2-3.4/2-3.4:1.0/input/input9
[  422.792879] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  422.797437] uvcvideo: Unable to create debugfs 2-8 directory.
[  422.797686] uvcvideo 2-3.4:1.3: Entity type for entity Processing 7 was not initialized!
[  422.797868] uvcvideo 2-3.4:1.3: Entity type for entity Extension 8 was not initialized!
[  422.798009] uvcvideo 2-3.4:1.3: Entity type for entity Camera 6 was not initialized!
[  430.578983] usb 2-3.4: usb_suspend_both: status 0
[  430.634971] usb 2-3: usb_suspend_both: status 0
[  430.635172] usb usb2: usb_suspend_both: status 0
```
  
  

dmesg output with D515:
```
[  596.460955] usb 2-3.2: new SuperSpeed USB device number 10 using tegra-xusb
[  596.481732] usb 2-3.2: New USB device found, idVendor=8086, idProduct=0b64
[  596.481743] usb 2-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  596.481750] usb 2-3.2: Product: Intel(R) RealSense(TM) 515
[  596.481755] usb 2-3.2: Manufacturer: Intel(R) RealSense(TM) Camera
[  596.481760] usb 2-3.2: SerialNumber: 00000000F0265395
[  596.486765] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
[  596.496639] uvcvideo: Unable to create debugfs 2-10 directory.
[  596.501603] uvcvideo: Unable to create debugfs 2-10 directory.
[  596.501773] uvcvideo 2-3.2:1.0: Entity type for entity Extension 3 was not initialized!
[  596.501959] uvcvideo 2-3.2:1.0: Entity type for entity Processing 2 was not initialized!
[  596.502105] uvcvideo 2-3.2:1.0: Entity type for entity Camera 1 was not initialized!
[  596.502734] input: Intel(R) RealSense(TM) 515 as /devices/3610000.xhci/usb2/2-3/2-3.2/2-3.2:1.0/input/input11
[  596.505214] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
[  596.509970] uvcvideo: Unable to create debugfs 2-10 directory.
[  596.510147] uvcvideo 2-3.2:1.4: Entity type for entity Processing 10 was not initialized!
[  596.510445] uvcvideo 2-3.2:1.4: Entity type for entity Extension 11 was not initialized!
[  596.510638] uvcvideo 2-3.2:1.4: Entity type for entity Camera 9 was not initialized!
[  603.632667] usb 2-3.2: usb_suspend_both: status 0
[  603.688651] usb 2-3: usb_suspend_both: status 0
[  603.688880] usb usb2: usb_suspend_both: status 0
```

RiccardoLincetto, 2021-08-09:
Hello, yes, but I won’t be able to work on it for 2 weeks. I kindly ask you to leave the issue open in the meantime.

MartyG-RealSense, 2021-08-09:
That is no problem at all to leave the case open for that further time period.  Thanks for the update!

closed 2021-11-03 by MartyG-RealSense