realsenseai/librealsense #9346: "ERROR [140293504141120] (backend-hid.cpp:1060) sensor gyro_3d not found!" and "RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x5597ffe65590):     Frame didn't arrive within 15000"

whizbuzzer, 2021-07-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (05.12.14.50) | 
| Operating System & Version |   { Linux (Ubuntu 20.04.2 LTS) } | 
| Kernel Version (Linux Only)    |  (5.8.0-59-generic)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.48.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
**ERROR [140293504141120] (backend-hid.cpp:1060) sensor gyro_3d not found!** appears when I enable the _Motion module_ in the realsense-viewer. Sometimes I can see neither the Accel stream nor the Gyro stream. Sometimes, I can see the Accel stream, but the Gyro stream says "No frames received". Sometimes, I can see both, Accel stream and the Gyro stream, but the Gyro stream is blank, while the error written above still pops up in the terminal.

Speaking of the next error,
**RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x5597ffe65590):     Frame didn't arrive within 15000**

It appeared when I tried to execute _rs-capture_ and _rs-hello-realsense_ Anyone knows how to fix these?

There is another list of errors attached in the screenshot below:
[image]

MartyG-RealSense, 2021-07-07:
Hi @whizbuzzer   The error messages suggest to me that the kernel is unpatched.  Which method did you use to build the librealsense SDK please - was it with Debian packages or with source code?

whizbuzzer, 2021-07-07:
Hi @MartyG-RealSense I installed the librealsense SDK and all the libraries given on _https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md_ in the exact way they instructed (sudo apt-get install) since I do not have much experience in building from source. I even updated the firmware with the latest version from Intel's website and made sure that the FW was supported.

MartyG-RealSense, 2021-07-07:
Thanks for the information.  When you have problems with the Motion Module, do you also have the depth and RGB streams enabled at the same time, or only the Motion Module?

whizbuzzer, 2021-07-07:
So, if I were to enable the Motion module alone, it would show me this:
[image]
But, if I were to turn on the motion module with the RGB and Depth camera, it would show me this:
[image]
Or this:
[image]

MartyG-RealSense, 2021-07-07:
Could you try running the **rs-capture** example program with your D455 please (making sure that the Viewer is not currently open when you launch it) and see whether the depth, RGB and gyro / accel streams can be accessed normally.  This test will help to demonstrate whether the problem is with the Viewer specifically or RealSense programs in general.

[image]

whizbuzzer, 2021-07-07:
I tried running rs-capture, rs-hello-realsense and rs-pointcloud, but all of them showed a black screen followed by _"RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x5597ffe65590): Frame didn't arrive within 15000"_

MartyG-RealSense, 2021-07-07:
Can you perform a check that the kernel was updated during the DKMS package installation process please by inputting the command below into Ubuntu:

**modinfo uvcvideo | grep "version:"**

Does the output of this command include the string **RealSense** ?

whizbuzzer, 2021-07-08:
@MartyG-RealSense  yes:
[
[image]
](url)

MartyG-RealSense, 2021-07-09:
The section of the **backend-hid.cpp** SDK script that is triggering the message **(backend-hid.cpp:1060) sensor gyro_3d not found!** is shown below.

[image]

I interpret that it is saying that it is not identifying the presence of the gyro sensor, as the 'sensor gyro_3d not found' message suggests.  And in the Viewer images that you supplied, it was the Accel that was okay whilst the Gyro was unavailable.

If it was only an issue with the gyro though then I would still expect programs such as **rs-hello-realsense** that do not access the Motion Module to function.

Next, can you check the **rs-motion** IMU example (which shows a camera graphic that should rotate when the camera is tilted and turned) and confirm whether it is functioning when you move the camera.  The image below is from my own D455.

[image]

whizbuzzer, 2021-07-12:
Hi @MartyG-RealSense sorry for the late reply. "rs-motion.cpp" is indeed working as yours.
[image]

MartyG-RealSense, 2021-07-12:
No problem at all - thanks very much for the update.  So **rs-motion** can function normally, indicating that there is not a hardware fault with the camera's IMU component.  Do other examples such as rs-hello-realsense continue to not work?

whizbuzzer, 2021-07-12:
Ok, so I used a previous Linux kernel and now my camera is working all fine. So, it must have been an issue with the kernel. Previously, I had used "linux-image-5.8.0-59-generic", where there were gyro issues. The camera however worked properly for "linux-image-5.8.0-43-generic"

MartyG-RealSense, 2021-07-12:
That's great news that a kernel change solved your problem. :)

MartyG-RealSense, 2021-07-18:
Case closed due to solution achieved and no further comments received.

closed 2021-07-18 by MartyG-RealSense

zhh2005757, 2022-01-05:
I've got the same problem with "sensor gyro_3d not found" and it bothered me until I turn off the SECURE BOOT!

MartyG-RealSense, 2022-01-05:
Thanks so much @zhh2005757 for sharing the solution that worked for you with the RealSense community!