realsenseai/librealsense #9347: How to frame synchronize the depth camera and color camera?

nevynwong, 2021-07-07:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } |
| Operating System & Version |   {Win 10 | 
Set depth, color frame rate to 60, how to fully synchronize frames?

MartyG-RealSense, 2021-07-07:
Hi @nevynwong   If you are asking how to synchronize RGB and depth frames, or how to enforce a constant FPS for both streams, then the resources listed in the link below may be helpful to you in answering both of these subjects.

https://github.com/IntelRealSense/librealsense/issues/8726#issuecomment-812556071

nevynwong, 2021-07-08:
@MartyG-RealSense ，Set the same frame rate, turn off auto-exposure priority, I've done that, I have one more question, is this sync useful for depth and color sync as well?
[image]

MartyG-RealSense, 2021-07-08:
Inter Cam Sync Mode is for syncing the timestamps of more than one camera using **hardware sync**.  So it would not be helpful for a single-camera project.

nevynwong, 2021-07-09:
This means that the depth and color cameras in one D415 camera will not be effective, but will be helpful for multiple D415 cameras.

MartyG-RealSense, 2021-07-15:
Hi @nevynwong  Do you require further assistance with this case, please?  Thanks!

nevynwong, 2021-07-15:
No, Thanks!

MartyG-RealSense, 2021-07-15:
Okay, thanks very much @nevynwong for the update!   I will close this case.  Thanks again!

closed 2021-07-15 by MartyG-RealSense