realsenseai/librealsense #9349: Apt server is broken/down, gives 403 Forbidden

spanaRe, 2021-07-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | N/A | 
| Firmware Version                   | N/A | 
| Operating System & Version |  Ubuntu 16.04/18.04/20.04  | 
| Kernel Version (Linux Only)    |  various                                        | 
| Platform                                 | PC/ NVIDIA Jetson |
| SDK Version                            |  N/A }                          | 
| Language                            |  N/A                          | 
| Segment			|  N/A                        | 

### Issue Description
I'm trying to install the SDK.
I tried according to this. 
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
I was able to register the public key, but when I run add-apt-repository, I get a 403 error.
Steps to repro:
```
sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
```
Example result: 
```
Hit:1 http://jp.archive.ubuntu.com/ubuntu focal InRelease
Get:2 http://jp.archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]     
Get:3 http://jp.archive.ubuntu.com/ubuntu focal-backports InRelease [101 kB]   
Hit:4 http://packages.ros.org/ros2/ubuntu focal InRelease                      
Get:5 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]      
Err:6 https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease
  403  Forbidden [IP: 52.218.91.82 443]
Hit:7 https://librealsense.intel.com/Debian/apt-repo focal InRelease           
Err:7 https://librealsense.intel.com/Debian/apt-repo focal InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C8B3A55A6F3EFCDE
Get:8 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [741 kB]
Hit:9 https://librealsense.intel.com/Debian/apt-repo bionic InRelease
Err:9 https://librealsense.intel.com/Debian/apt-repo bionic InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY C8B3A55A6F3EFCDE
Get:10 http://security.ubuntu.com/ubuntu focal-security/main i386 Packages [254 kB]
Get:11 http://security.ubuntu.com/ubuntu focal-security/universe i386 Packages [502 kB]
Get:12 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [628 kB]
Reading package lists... Done                                                  
E: Failed to fetch https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease  403  Forbidden [IP: 52.218.91.82 443]
E: The repository 'https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

I referred to past cases, but I can't solve it.
Help me!

MartyG-RealSense, 2021-07-07:
Hi @spanaRe  The keyserver system was changed in late June to phase out the keys.gnupg.net keyserver address, but Intel had already prepared for this.  An alternative keyserver address provided after the '||' symbol offers **keyserver.ubuntu.com** as an alternate address.

sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE **|| sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE**

The subject was discussed during the past week at https://github.com/IntelRealSense/librealsense/issues/9181#issuecomment-874163120  

I see that you inputted the full keyserver command in the installation instructions though, so I am uncertain why the command would not work.

In the related case https://github.com/IntelRealSense/librealsense/issues/9289#issuecomment-873457374 a RealSense user foud that replacing keys.gnupg.net with keyserver.ubuntu.com in the first half of the command worked for them.

**sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE**

spanaRe, 2021-07-09:
Hello, ＠MartyG-RealSense
Thanks for your comment.

I deleted the public key once and started over.

```
sudo apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
```
```
Executing: /tmp/apt-key-gpghome.k3lgVFPVKY/gpg.1.sh --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: keyserver receive failed: No name
Executing: /tmp/apt-key-gpghome.6TlrqgPbDq/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A 55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: public key ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" imported
gpg: Total number processed: 1
gpg: imported: 1
```

The proxy settings are fine, but for some reason I am getting a No name error.
When I change the key server and execute, I get a timeout.

```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
```
```
Executing: /tmp/apt-key-gpghome.GHfx68U99q/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: keyserver receive failed: Connection timed out
Executing: /tmp/apt-key-gpghome.aJDijd1r9Q/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: public key ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" imported
gpg: Total number processed: 1
gpg: imported: 1
```

I also tried running both with the proxy settings added for the year, and the results were similar.
```
--keyserver-options http-proxy=http://localhost:8080
```

Do you have a solution for this?

MartyG-RealSense, 2021-07-09:
1.  Do you have any luck with the command below?

**sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 (hkp://keys.gnupg.net:80) --recv-key C8B3A55A6F3EFCDE**

2.  Is there the possibility that a **firewall** is blocking access from your computer?

hwh-nearearth, 2021-07-12:
@MartyG-RealSense 

We are also seeing this problem while trying to install packages. I was following the instructions based on the 2.47.0 tagged [release documentation](https://github.com/IntelRealSense/librealsense/blob/v2.47.0/doc/distribution_linux.md). I modified the process based on your previous recommendation to just hit `keyserver.ubuntu.com`.  

Note - the key is updated just fine. See in my terminal output below, key was "not changed". So it is not an issue with the key unless the hosting system is not accepting that key any longer.

```
apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
Executing: /tmp/apt-key-gpghome.B8SnwzTKGN/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
gpg: key C8B3A55A6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
```

Here is the key listing:
```
apt-key list
/etc/apt/trusted.gpg
--------------------
pub   rsa2048 2018-01-09 [SC] [expires: 2028-01-08]
      F6E6 5AC0 44F8 31AC 80A0  6380 C8B3 A55A 6F3E FCDE
uid           [ unknown] "CN = Intel(R) Intel(R) Realsense", O=Intel Corporation
```

Here is the add repository command with the 403 error
```
add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo bionic main" -u
Err:9 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease
  403  Forbidden [IP: 52.218.121.145 80]
```

Additionally I try to browse to the repository location (which usually allows me to browse versions, etc) and I still get a 403 response.
```
<Error>
    <Code>AccessDenied</Code>
    <Message>Access Denied</Message>
    <RequestId>49WDSY7S16CVCDZ5</RequestId>
    <HostId>
        Qw/xkoMcqkagjbw5YiDg23UrRWew+...
    </HostId>
</Error>
```

MartyG-RealSense, 2021-07-18:
Hi @spanaRe  Do you require further assistance with this case, please?  Thanks!

ttencate, 2021-07-18:
**I don't think this has anything to do with keys.** The issue report shows a "403 Forbidden" error. (Note that normally, keys are not needed to _access_ an apt repository; they are only used to verify package contents after downloading packages.) Also notice [above](https://github.com/IntelRealSense/librealsense/issues/9349#issuecomment-876942330) that @spanaRe got a "No name" error from the first command, but thanks to the `||` construct, the second one was then executed, which succeeded with "gpg: imported: 1". So they received the key successfully in the end.

**I think the actual problem is that _something_ is using the old hostname `realsense-hw-public.s3.amazonaws.com` instead of the new `librealsense.intel.com`.** Indeed, https://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease gives a 403, but https://librealsense.intel.com/Debian/apt-repo/dists/xenial/InRelease works. But @spanaRe was using the latest repository URL, `https://librealsense.intel.com/Debian/apt-repo`, so how can this be? If it's not a caching issue on their side, it must be something on the the Intel side...

ev-mp, 2021-07-18:
@spanaRe,
As noted by @ttencate , the PPA server address you use is not up-to-date.  
The PPA transition to the secure server has happened [in March](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#debian-distribution-ppa-change---user-notification), yet the original address was maintained till the end of May to allow for a gradual transition, when it was finally disconnected.

Follow the instructions and replace the deprecated link with data from https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

Duplicate of #9102, #9105, #9181, #9302

MartyG-RealSense, 2021-07-24:
Case closed due to no further comments received.

closed 2021-07-24 by MartyG-RealSense

outbreak-sen, 2021-09-07:
sorry ，I havn‘t got your point. I just follow the github 's instruction. And I wonder if you could give me some detail instruction about my problem . thanks