realsenseai/librealsense #9354: Can't run PCL examples due to DLL errors

suomi-miartem, 2021-07-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |   Win 10 x64  | 
| Platform                                 | PC  |
| SDK Version                            |  2.45.0        | 
| Language                            |  C++/PCL                      | 

### Issue Description
I put together a solution with a PCL module in VS2019 following the instructions in the PCL section using the latest versions of RealSense (2.45.0) and PCL (1.11.1). The solution was assembled without errors, but when I ran the example, the exception popped up "cannot continue the execution of the code because the system did not find" pcl_segmentation.dll ", and the console displayed the inscription" Program "[15236] rs-pcl.exe" completed with the code -1073741515 (0xc0000135) 'The dependent DLL file was not found.' "What could be the error, and can I fix it somehow? Should I add DLLs directly to the project folder as a temporary solution, or do something similar with DLLs? No advice ([here](https://github.com/IntelRealSense/librealsense/issues/7549), [here](https://github.com/IntelRealSense/librealsense/issues/3378) and here) could help me.
 
The algorithm by which I created the assembly and ran the example inside:
1. Installed PCL using the All-In-One installer according to the [instructions](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl) from the corresponding section of repository
2. Downloaded the source code of RealSense
3. Collected RealSense with the PCL module using Cmake (3.21.0), checking BUILD_SHARED_LIBS and BUILD_PCL_EXAMPLES (x64 configuration for VS2019)
4. Collected the assembly solution in Release x64 configuration (ALL_BUILD by default)
5. Launched debugging of the "rs-pcl" project (the same Release-x64 configs)

There is a suspicion that the error is contained in one of the cmake files, but I could not find it myself due to my inexperience (т-т).

suomi-miartem, 2021-07-07:
**UPD**
Solved it by adding the requested DLLs from {PCL_ROOT}/bin to project folder, but the feature to add DLLs in project while assembling is still relevant.

MartyG-RealSense, 2021-07-08:
Excellent news @suomi-miartem that you were able to make progress with the DLLs.  I look forward to your next report   :)

suomi-miartem, 2021-07-09:
Thank you, @MartyG-RealSense :)
So, I continue.

Trying to catch the moment of the output of the error in the console, I decided to start the project in debug mode so that the program stops at the breakpoint given by me, which displays an error. As a result, I found that the debug version of the project refers to the release name of the glfw3 library (glfw3.lib), although it must refer to the library name with the end of "d", which gives an error when trying to start the project in debug mode. 

As a temporary solution, I removed this letter from the library name in the debug folder, the path to which was set in the project cmakelists (the name of the "glfw3d.pdb" file, as I found out, is not necessary to change), and the project has launched. 

Oddly enough, after returning to the issue release mode, the project also began to work properly. 

So, current fix requests:
1. To add rebug/release PCL DLLs to the associated project while building
2. Incorrect name of the linked debug library in PCL projects ("glfw3.lib" instead of "glfw3d.lib")

Since my problems are eliminated, I close this issue.

closed 2021-07-09 by suomi-miartem

MartyG-RealSense, 2021-07-09:
Thanks again for providing such detailed feedback @suomi-miartem for the benefit of the RealSense community!