realsenseai/librealsense #9357: l515 multi-cam inter-cam sync crashes on some cameras jetson only

Kevin-Delnoije, 2021-07-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | l515| 
| Firmware Version                   | 1.5.4.1 (newest version firmware crashes camera )| 
| Operating System & Version |   both jetpack 4.4 and 4.51 | 
| Kernel Version (Linux Only)    |  tegra 4.9                                      | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.4.7 /2.4.8                     | 
| Language                            | realsense viewer + gpio script                         | 
| Segment			| robot                  | 

### Issue Description
I open the realsense viewer and i can turn on both camera's, then i click the inter-cam checkbox on both control panels.
No problem so far if there is no gpio signal, the moment i send a gpio signal one of the camera always crashes and the other one works. I checked the wiring and swapped everything still same issue and when cameras connected to laptop both camera's can be toggled using gpio from jetson.

sometimes if i dont close realsense-viewer after crash the whole jetson crashes and i need to reboot.

some logs:
(at start) uvc metadata payloads are not available for stream
backend-v4l2.cpp incomplete frame detected  
frames didn't arrive within 5 seconds

I installed realsense from source using the native backend and kernel patching. 

might be related to this https://github.com/IntelRealSense/librealsense/issues/9329, since we cant upgrade all cameras we tried to the newest firmware without crashing

RealSenseCustomerSupport, 2021-07-09:
----------------------------------------------

Hi
What was the configuration for each L515? resolution? frame rate? ...
What does the trigger looks like for each L515? pulse level and frequency, duty circle? are they using same trigger?
Are you saying this issue doesn't happen on Windows platform? if so, that sounds camera HW and FW should be fine.

Thanks

Kevin-Delnoije, 2021-07-15:
At the beginning i had the trigger similar to the multi-camera example docs, but after it crashed i switched to a breadboard for easier testing where i manually moved the pins from +3.3v and 0 from the jetson gpio. Same crashes were observed using breadboard manually or using a clock signal that was slow enough.
on my ubuntu 18.04 LTS laptop it works with realsense viewer 2.47 without crashes to turn both camera's on and off. When i try on jetson xavier nx (jetpack 4.3) and xavier agx (jetpack 4.4 and 4.5) i get the crash when moving from high to low. (except one camera out of the 4 i tried keeps working without problems). The moment the camera's get stuck i need to plug/unplug usb connection for it to connect again.

on both hardware setups the exact same electric signals are send (from jetson) and i have measured the voltage with a multimeter and swapped the cables between cameras. (only one magic camera always seems to work)

standard settings when turning on depth stream are used (besides inter-cam checkbox):
resolution: 640*480 depth stream only
fps: 30

also the camera is active when signal is low and turns off when signal is high.

I have not tried on windows yet, but the fact that it works on a laptop and not on the jetson suggest that there is some incompatibility there.

RealSenseCustomerSupport, 2021-07-16:
----------------------------------------------

Thank you for the feedback.
Do you have a chance to try using an external USB3 hub instead of directly connecting camera to Xavier's USB port? try 1x camera first, then 2x and more.
Regarding the "magic camera", is its FW same as others? can you list the first 4 digits of the Serial Number for all cameras?

closed 2021-07-30 by RealSenseCustomerSupport