realsenseai/librealsense #9358: Mac installation instructions out of date

Timmmm, 2021-07-08:
[This page](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation_osx.md) says to run this command:

    brew cask install apenngrace/vulkan/vulkan-sdk

But `brew cask` is no longer a valid command. It seems like you can just do:

    brew install apenngrace/vulkan/vulkan-sdk

Is the vulkan SDK definitely needed? Because [this other page](https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk) doesn't mention it and it asks for glfw instead.

Also when you do `brew install homebrew/core/glfw3` it tells you to use `glfw` not `glfw3`. I think both of those pages need checking.

MartyG-RealSense, 2021-07-09:
Hi @Timmmm  Thanks very much for your report.  I have added a **Documentation** tag to this case so that RealSense team members responsible for documentation can consider it at a future date.  This case should be kept open in the meantime.  Thanks again!

MartyG-RealSense, 2023-07-18:
Case closed due to Mac no longer being officially supported and modern community-provided Mac installation instructions available elsewhere.

https://lightbuzz.com/realsense-macos/