realsenseai/librealsense #9360: Kernel support for librealsense.

Ronic01, 2021-07-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | {D400 } | 
| Firmware Version                    | (None at the moment) | 
| Operating System & Version  | {Ubuntu 20.04} | 
| Kernel Version (Linux Only)    | (5.9) | 
| Platform                                  | PC/GeForce RTX 2080 SUPER |
| SDK Version                            | { None at the moment } | 
| Language                                | {python } | 
| Segment			         |{Robot} | 

### Issue Description
Hello,

 

I would like to clarify on kernel support for librealsense.

 

I am working on machine with relatively new hardware, because of this the graphics card is only compatible with more new Linux kernels (starting from 5.8), as well the system I am working with requires a real-time patch (meaning I am limited to Linux kernels 5.9, 5.10, 5.11, 5.12).

 

Therefore, I would like to ask is there support planned for Linux kernels 5.9, 5.10, 5.11  or 5.12?

 

As well maybe there is some sort of work-around to get existing library version working with more new kernels? Thank you very much.

MartyG-RealSense, 2021-07-09:
Hi @Ronic01  Whilst I do not have information about future plans for support of specific Linux kernels or time schedules for implementation, support for newer kernels does get added over time, such as the recent addition of 5.8 support.

More recent kernels than the officially supported ones have been used by RealSense users in the past (such as when 5.4 was used for a while when 5.3 was the most recent officially supported version), though the results will be unpredictable in regards to stability.

There is the option of building the librealsense SDK from source code with the **RSUSB backend** method.  This installation method is not dependent on Linux versions or kernel versions and does not require patching.  It does have some limitations compared to the patched-kernel method though, such as being suited to single-camera applications rather than multiple cameras.  

For a comparison of the advantages and disadvantages of these two installation methods, you can visit the link below and scroll down through the linked-to comment to the section headed **What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

Ronic01, 2021-07-09:
Hello @MartyG-RealSense. Thank you for the information and suggestions. It is highly appreciated.

MartyG-RealSense, 2021-07-16:
Hi @Ronic01  Do you require further assistance with this case, please?  Thanks!

Ronic01, 2021-07-16:
Hi @MartyG-RealSense, I am really grateful for the help solving the issue. I would just like to clarify on uninstalling. I need to uninstall and freshly install the library, however with the RSUSB backend method I cannot seem to find build directory to run **sudo make uninstall** command, could you please advise what approach should I take to re-install the library? Thank you.

MartyG-RealSense, 2021-07-16:
As far as I know, you would simply return to the build directory in the librealsense source code folder (the one that was originally created with **mkdir build && cd build**) and then input **sudo make uninstall && make clean**

Ronic01, 2021-07-16:
Yes this seemed to work. Thank you!

MartyG-RealSense, 2021-07-16:
Great to hear that it worked for you @Ronic01 - you are very welcome   :)

MartyG-RealSense, 2021-07-22:
Case closed due to solution achieved and no further comments recieved.

closed 2021-07-22 by MartyG-RealSense

doisyg, 2021-08-13:
Any update on this ? Ubuntu 20.04 LTS with hwe just upgraded to kernel 5.11. It would be nice to have path and DKMS for this.
As a general rule, following Ubuntu LTS kernel version should be, I believe, the target (as it is also the main ROS dev OS)

MartyG-RealSense, 2021-08-13:
Hi @doisyg   The librealsense SDK does receive kernel support updates over time but not on a regular schedule.  As mentioned earlier in https://github.com/IntelRealSense/librealsense/issues/9360#issuecomment-876995830 the SDK can be used without dependence on particular kernel versions by building librealsense with the **RSUSB** method.