realsenseai/librealsense #9361: How to get the true depth value

JZhu1208, 2021-07-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   |v2.45.0 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                    | 
| Platform                                 |PC  |
| SDK Version                            |v2.45.0                     | 
| Language                            | v2.45.0                          | 
| Segment			|              | 

Hello,

When I used the L515 camera, I saved the 2D depth map (.png) observed by the realsense viewer and read it in with matlab (using the ‘imread’ function) to get a two-dimensional matrix. Is the value in the matrix the intensity value of the pixel in the depth map? If so, how to convert it to a true depth value. If not, how can I get the true depth value from the L515 camera. thank you very much

[image]

MartyG-RealSense, 2021-07-09:
Hi @JZhu1208  The MATLAB documentation about images states that most images that MATLAB handles have a two-dimensional matrix, whilst an **RGB** image may have a three-dimensional matrix.

https://uk.mathworks.com/help/matlab/creating_plots/working-with-images-in-matlab-graphics.html

My understanding is that the matrix represents intensity values of pixels, but I am not familiar enough with MATLAB's data structure to describe more than that with confidence in accuracy.

It is worth bearing in mind that when depth is saved into a PNG image file in RealSense, most of the depth information is lost.  It may therefore be better to use the **.raw** image file exported by the Viewer's snapshot function, as this preserves depth information  An Intel RealSense team member once suggested using the code below to read a .raw into MATLAB.

```
fid = fopen("FilesName.Raw");
Depth = fread(fid, [1280 720], '*uint16')';
fclose(fid);
```

In regard to ply import, there is a ply import tool available for MATLAB that converts a ply to a MAT file with XYZ coordinates and RGB color channels.

https://uk.mathworks.com/matlabcentral/fileexchange/54041-plytomat 

If you would like to explore live image acquisition in MATLAB with a RealSense camera, the discussion in the link below may be helpful.

https://support.intelrealsense.com/hc/en-us/community/posts/1500001233981-Shooting-and-saving-videos-with-RealSense-d435-using-MATLAB

JZhu1208, 2021-07-09:
Hello, thank you for your patient answers.

Regarding your answer, can the Z in XYZ coordinates be understood as the depth value?

> 
> In regard to ply import, there is a ply import tool available for MATLAB that converts a ply to a MAT file with XYZ coordinates and RGB color channels.
> 
> https://uk.mathworks.com/matlabcentral/fileexchange/54041-plytomat

MartyG-RealSense, 2021-07-09:
As the ply file exported from RealSense stores 3D point cloud data, I would assume that Z does represent 3D depth when XYZ are extracted from the ply into MATLAB.

closed 2021-07-22 by MartyG-RealSense