realsenseai/librealsense #9362: invalid key/value pair in file /etc/udev/rules.d/99-realsense-libusb.rules

yzkzero, 2021-07-09:
**Operating System & Version:** Linux 5.4.70 (aarch64) built by yoctor
**udevd version:** 3.2.8
**udevadm version:** 228

I'm deploying the realsense SDK 2.45 to my iMX8 board. And follow the installation guideline, I copied 99-realsense-libusb.rules and run the command of "udevadm control --reload-rules && udevadm trigger".

Then I got following error:
[ 1589.181217] udevd[261]: invalid key/value pair in file /etc/udev/rules.d/99-realsense-libusb.rules on line 81, starting at character 154 ('.')

But it seems the SDK and camera (D455) work as expected. However every time the system is booting, the above error will be shown when starting udevd.

MartyG-RealSense, 2021-07-09:
Hi @yzkzero   Regarding the information **Linux 5.4.70 (aarch64) built by yoctor** - do you mean Yocto, please?

I haven't seen this particular error in librealsense before.  There has been a past reported issue with using **udevadm control --reload-rules && udevadm trigger** in a Docker container on an Arm-based Nvidia Jetson board though.  The i.MX8 board is Arm-based too.  Is your project using a Docker container?

https://github.com/IntelRealSense/librealsense/issues/7849

yzkzero, 2021-07-11:
Hi MartyG,

Sorry for the typo. Yes, it's built by Yocto. 
I'm not using docker or any container. The error only displays in serial console (serial port connected). If I execute the command in ssh or telnet console, there's no error displayed. But the error is immediately displayed in serial console once I execute the command in ssh/telnet console. And as I mentioned, each time the system is booting up, the error is also shown. It can also be observed in boot log.

MartyG-RealSense, 2021-07-11:
Would it be possible to provide a link to the installation instructions that you used, please?  The process that you used sounds similar to the one in the link below.

https://stackoverflow.com/questions/62134563/how-to-give-permission-to-intel-realsense-camera-on-ubuntu/62369143#62369143

yzkzero, 2021-07-14:
I referred to installation instruction of raspberry pi : https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

What I did are:
$ git clone https://github.com/IntelRealSense/librealsense.git
$ cd librealsense
$ sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/ 
$ sudo udevadm control --reload-rules && udevadm trigger

MartyG-RealSense, 2021-07-14:
The official raspbian_md guide is suited to Pi 3.  If you have a Raspberry Pi 4 then the similarly structured **Acrobotic** installation guide for Pi 4 and Raspbian is a better choice.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

 RealSense users who have performed installation with these guides have also recommended using a more modern version of **Protobuf** than the ones in the guides, such as version **3.14** in order to reduce the likelihood of build errors.

yzkzero, 2021-07-15:
Hi MartyG,

I just referred to the part of applying udev rules from raspberry doc.

The only things I did on my iMX8 board are:
$ git clone https://github.com/IntelRealSense/librealsense.git
$ cd librealsense
$ sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
$ sudo udevadm control --reload-rules && udevadm trigger

MartyG-RealSense, 2021-07-15:
If the **invalid key/value pair** message is not affecting the D455's operation then would it be an option to simply ignore the message?

yzkzero, 2021-07-16:
> 
> 
> If the **invalid key/value pair** message is not affecting the D455's operation then would it be an option to simply ignore the message?

Yes, it currently doesn't affect the RGB and depth function on D455 so far. But I'm not sure if it will affect the IMU or any other function I am not aware of. 

The comment for the error line is:
`# For products with motion_module, if (kernels is 4.15 and up) and (device name is "accel_3d") wait, in another process, until (enable flag is set to 1 or 200 mSec passed) and then set it to 0.
KERNEL=="iio*", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0ad5|0afe|0aff|0b00|0b01|0b3a|0b3d|0b64|0b68", RUN+="/bin/sh -c '(major=`uname -r | cut -d \".\" -f1` && minor=`uname -r | cut -d \".\" -f2` && (([ $major -eq 4 ] && [ $minor -ge 15 ]) || [ $major -ge 5 ])) && (enamefile=/sys/%p/name && [ `cat $enamefile` = \"accel_3d\" ]) && enfile=/sys/%p/buffer/enable && echo \"COUNTER=0; while [ \$COUNTER -lt 20 ] && grep -q 0 $enfile; do sleep 0.01; COUNTER=\$((COUNTER+1)); done && echo 0 > $enfile\" | at now'"`

So it looks like the rule is for the motion module.

MartyG-RealSense, 2021-07-16:
If you are using a text-only display in a terminal and are not able to run graphical tools such as the RealSense Viewer, it may be difficult to validate the Motion Module's performance (though it is likely that it is a one-off warning rather than something that impacts Motion Module performance).

You could perhaps profile the IMU streams and collect 60 seconds worth of data into a log file using the SDK's **rs-data-collect** tool.

https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect

MartyG-RealSense, 2021-07-28:
Case closed due to no further comments received.

closed 2021-07-28 by MartyG-RealSense

nalithlakshan, 2022-08-03:
I have the exact same issue when doing this on Xilinx ZCU104 running Petalinux 2020.2. I get the error in doing the exact same steps as below.
```
$ git clone https://github.com/IntelRealSense/librealsense.git
$ cd librealsense
$ sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
$ sudo udevadm control --reload-rules && udevadm trigger
```
After finally building and installing the SDK my 435i camera is not being detected. I beleive it is because of this error in /etc/udev/rules.d/99-realsense-libusb.rules on line 81, starting at character 154 ('.'). 

Is the cause of this error found yet? If so please guide me to fix it.

MartyG-RealSense, 2022-08-03:
Hi @nalithlakshan  There are very few references for installing the librealsense SDK on Xilinx products, and only for **Xilinx Ultra96 FPGA**.  These references are listed at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/6673895808915/comments/6674992093459

I would recommend trying the **libuvc backend** installation method, as it bypasses the Linux kernel and so is not dependent on Linux versions of kernel versions and does not require kernel patching.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md