realsenseai/librealsense #9363: RealSense Viewer, wrong visualization of the top row of the point cloud

Marenix, 2021-07-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400| 
| Firmware Version                   |-| 
| Operating System & Version |   Win10, Ubuntu 20  | 
| Kernel Version (Linux Only)    | -                                      | 
| Platform                                 | PC  |
| SDK Version                            | 2.48.0 }                          | 
| Language                            |   | 
| Segment			|         | 

Hey. I have more of an observation than an issue. I was using RealSense Viewer, and while looking at the point cloud I saw something strange in the top row. Here are the screenshots I took:
[image]
[image]
As you can see, the top row of the point cloud looks a bit unexpected. The value of each point is okay, so, as I said, it's not really an issue, except maybe as a visualization issue. I also noticed that this "problem" only occurs when you disable GLSL for rendering. 
My guess is that this has something to do with how the values of each point of the quadrangle are chosen. Maybe, as the top row doesn't have any value "above" it, it takes the values from the bottom most point? On the other hand, if that's the case, I don't understand why it's only one vertex, not both of the top ones, and also, why this only happens on the top row, and not the bottom one too (and left and right column). 
Do you have any idea why this is happening only for that one row? Was this noticed before? And, if it was, are there any plans for fixing it, or as it's not really causing any problems, it'll be left as is (I know that GLSL rendering removes it, but as disabling GLSL rendering is an option, that would still be a valid way of using the Viewer)?
Thanks for the help!

MartyG-RealSense, 2021-07-09:
Hi @Marenix  I have not noticed that particular graphical trait at the top of the screen in the Viewer's 3D mode before, though I believe that you are the first to draw attention to it by providing a zoomed image.  Thanks!

A consequence of disabling GLSL rendering is that access to the **Measure** option at the top of the Viewer window in 3D mode for measuring between two points on the cloud is disabled until GLSL rendering is re-enabled.  This is not a problem if you are not interested in measuring between two points though.  Other than that, I am not aware of negative effects of disabling rendering other than a possible negative effect on overall performance (though a reduction in CPU utilization).

[image]

RealSense team member @ev-mp may be able to offer advice on why this effect at the top of the screen may occur when GLSL rendering is disabled.

MartyG-RealSense, 2021-07-16:
Hi @Marenix  Do you require further assistance with this case, please?  Thanks!

Marenix, 2021-07-16:
Hi @MartyG-RealSense I reported this to draw attention to it, in case you'd want to fix it. Thanks for the info and help!

MartyG-RealSense, 2021-07-16:
I performed testing in the Viewer's 3D point cloud mode with both GLSL modes disabled and was able to recreate a micro-triangular effect on the top edge that resembles the one that you reported.  It went away if enabling the GLSL **Rendering** option but having the **Processing** option disabled.  So it does seem to be focused around the Rendering aspect.  

Ultimately, it does not seem to be a serious enough effect to warrant a fix, since the effect does not disrupt the depth image.

[image]

Marenix, 2021-07-21:
Okay, thanks for the info!

Marenix, 2021-07-27:
Hi @MartyG-RealSense Thanks for the help! I'll close the issue now :)

closed 2021-07-27 by Marenix

MartyG-RealSense, 2021-07-27:
You are very welcome @Marenix - thanks for the update!