realsenseai/librealsense #9364: Cannot enable VGA resolution on the L515 throught .cpp code

ParadoxRobotics, 2021-07-09:
### Issue Description

Hi, 
I would to configure my L515 to output depth and color streams at VGA (640x480) resolution.
Unfortunately, at execution I get an rs2::error : 
```
terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
Abandon (core dumped)
```
from the code : 
```
  rs2::pointcloud pc;
  rs2::points RSCloud;
  // Realsense filters
  rs2::spatial_filter spatialFilter;
  spatialFilter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 5);
  spatialFilter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.25);
  spatialFilter.set_option(RS2_OPTION_FILTER_SMOOTH_DELTA, 25);
  rs2::threshold_filter thresholdFilter;
  thresholdFilter.set_option(RS2_OPTION_MAX_DISTANCE, maxDepth);
  rs2::config configL515;
  configL515.enable_stream(rs2_stream::RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
  configL515.enable_stream(rs2_stream::RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Y16, 30);
  rs2::pipeline pipe;
  pipe.start(configL515);
```
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 2.48.0| 
| Operating System & Version |   Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.8.0-59-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2                        | 
| Language                            |  C++                         | 
| Segment			|  Robot                         |

MartyG-RealSense, 2021-07-09:
Hi @ParadoxRobotics  The **Couldn't resolve requests** error occurs because a camera configuration that is currently unsupported was requested or there was a mistake in the code that causes the request not to be understood.  In your case it appears to be that the requested RGB mode was unsupported by the firmware version installed in your L515.

Support for 640x480 and 640x360 RGB resolutions was added in L515 firmware **1.5.8.1**.  Please update your L515's firmware to this version to access 640x480 RGB.

[image]

ParadoxRobotics, 2021-07-09:
Hi, Thanks for your answer.
I update my L515 to the latest firmware release; I can access the VGA mode only in the realsense-viewer but in my .cpp code.
I still get an : 
```
terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
Abandon (core dumped)
```
with the code 
```
// RealSense specific library
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
int main (int argc, char** argv){
  // Realsense filters
  rs2::spatial_filter spatialFilter;
  spatialFilter.set_option(RS2_OPTION_FILTER_MAGNITUDE, 5);
  spatialFilter.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 0.25);
  spatialFilter.set_option(RS2_OPTION_FILTER_SMOOTH_DELTA, 25);
  rs2::threshold_filter thresholdFilter;
  thresholdFilter.set_option(RS2_OPTION_MAX_DISTANCE, maxDepth);
  // Start the Realsense camera
  rs2::pipeline pipe;
  rs2::config configL515;
  configL515.enable_stream(rs2_stream::RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
  configL515.enable_stream(rs2_stream::RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Y16, 30);
  pipe.start(configL515);
  // get intrinsic matrix
  auto const intrinsic_depth = pipe.get_active_profile().get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>().get_intrinsics();
  auto const intrinsic_color = pipe.get_active_profile().get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>().get_intrinsics();
  // Skips some frames to allow for auto-exposure stabilization
  std::cerr << " PipeLine started -> running auto-exposure" << std::endl;
  for (int i = 0; i < 10; i++) pipe.wait_for_frames();
  while(true){
    // acquiring initial depth state
    rs2::frameset rgbdSet = pipe.wait_for_frames();
    rs2::depth_frame depthFrame = rgbdSet.get_depth_frame();
    // filtering using temporal, spatial and threshold filter
    rs2::depth_frame FilteredDepthFrame = depthFrame;
    FilteredDepthFrame = thresholdFilter.process(FilteredDepthFrame);
    FilteredDepthFrame = spatialFilter.process(FilteredDepthFrame);
    // get filtered realsense point cloud
    RSCloud = pc.calculate(FilteredDepthFrame);
 }
  // stop camera process
  pipe.stop();
  return (0);
}
```

MartyG-RealSense, 2021-07-09:
Could you try using **RGB8** instead of BRG8 as the color format please.

MartyG-RealSense, 2021-07-09:
I see you have the **depth** format set as Y16.  That format is usually used for infrared.  Could you change the depth format to **Z16** please.

MartyG-RealSense, 2021-07-16:
Hi @ParadoxRobotics  Do you require further assistance with this case, please?  Thanks!

ParadoxRobotics, 2021-07-16:
Hi,

I'm terribly sorry; I completely forget to write an answer...
Your are totally right, the depth format was wrong; I simply change the Y16 to Z16.
The code works perfectly.

Best regards.

Munch Quentin.

MartyG-RealSense, 2021-07-16:
It's no trouble at all, @ParadoxRobotics - it's great to hear that the advice worked for you.  Thanks for the update!

closed 2021-07-22 by MartyG-RealSense