realsenseai/librealsense #9366: realsense wrong

xuyixiO, 2021-07-09:
| | |
|---|---|
|**librealsense**|2.47.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:569

```

MartyG-RealSense, 2021-07-09:
Hi @xuyixiO  Could you use the instructions in the link below to run an example called **rs-capture** and let me know whether it works, please?  As rs-capture and the RealSense Viewer both depend on **OpenGL** graphics to function, if rs-capture works correctly then this would suggest that the problem is not with the OpenGL graphics support on your ocmputer.

https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

xuyixiO, 2021-07-09:
thank  you so much. it works，however，when I click the "Intel RealSense Viewer",it says something went wrong.
I thought it might caused by computer,however,no matter my thinkpad or Microsoft 7 both can't open it.

 

 
#8024 (comment)

MartyG-RealSense, 2021-07-09:
You are very welcome!  Is the **Depth Quality Tool** able to be launched?  This program has a very similar design to the Viewer.

xuyixiO, 2021-07-09:
both are not able to be launched

MartyG-RealSense, 2021-07-09:
Are you using Windows 10 or Windows 7, please?

xuyixiO, 2021-07-09:
I’m using Win10 now

MartyG-RealSense, 2021-07-09:
Could you update the graphics drivers on your computer please?  Although the ability of rs-capture to work correctly suggests that it is not a graphics driver issue, updating the drivers should help to eliminate this possibility.

xuyixiO, 2021-07-09:
Here it is, I have been updated all the drivers as I can. you can see that the 
error it said the path is not real exist，
Including hidden files，if I can find which file is wrong, I can change it, is this situation have connected with win10? I have found some post that said opengl can’t work well on it.
Thank you for your guidance.

MartyG-RealSense, 2021-07-09:
You are not the first RealSense user who has experienced this error recently.  It was also reported in https://github.com/IntelRealSense/librealsense/issues/9176 - though that case did not have a clear outcome that indicated success or failure.

RealSense usually works fine with OpenGL on Windows 10, and the OpenGL-using rs-capture program worked correctly for you.

 Have you installed the full RealSense SDK for Windows 10 using the **Intel.RealSense.SDK-WIN10-2.47.0.3309.exe** installer file from the RealSense SDK's **Releases** page?  Or did you install the SDK on Windows using a different method?

[image]

xuyixiO, 2021-07-09:
Sorry to bother you again， I have to say that the thinkpad t14 is the computer that I just bought, so~you know, it can't be any thing wrong with my computer

xuyixiO, 2021-07-09:
I installed everything include debug, I sincerely hope that we can solve this problem,or I have to go to Ubuntu to solve it.

MartyG-RealSense, 2021-07-09:
If it is the two **GLSL** settings (Rendering and Processing) in the Viewer and the Depth Quality being enabled by default that is preventing launch of those two programs then I'm not sure that there is anything that can be done about it if the programs cannot be launched first in order to access the Settings interface to disable GLSL.

[image]

Being unable to launch the Viewer would not mean that the camera cannot be used, since the other SDK programs apparently work and you could develop your own applications too.  I appreciate that it would be very undesirable not to be able to use the Viewer though.

There is a way to check whether the GLSL settings may be the cause.  Please go to the old **2.21.0** SDK at the link below and download the **Intel.RealSense.Viewer.exe** file (the standalone Viewer version) from its Assets list.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.21.0

This was the version before GLSL was introduced and so does not have the GLSL settings in the Viewer. 

[image]

The standalone Viewer should be launched from the computer location that it is downloaded to (not the launch icon on the Windows desktop), or by left-clicking on the file download tab at the bottom of the browser window after the file has finished downloading.

xuyixiO, 2021-07-09:
ok，I will try it，Then I'll tell you the results

 

 

 
 
 
https://github.com/IntelRealSense/librealsense/releases/tag/v2.21.0

xuyixiO, 2021-07-10:
bother you again.官方给的注册服务器公钥只有一个可以使用，并且添加服务器列表时显示403，显示该网站没有数字前面，请问我该怎么办啊

 

 

 
 
 
https://github.com/IntelRealSense/librealsense/releases/tag/v2.21.0

MartyG-RealSense, 2021-07-10:
Translation: "Only one public key of the registered server given by the official can be used, and 403 is displayed when the server list is added, indicating that the website does not have a number in front.  What should I do?"

It is no trouble.  It sounds as though you may be experiencing the problem described in the recent case linked to below.

https://github.com/IntelRealSense/librealsense/issues/9349

********

這沒有問題。 聽起來您可能遇到了與下面鏈接的最近案例中描述的問題。

xuyixiO, 2021-07-10:
thanks again，old version is acceptable，i run it pretty well！！！

closed 2021-07-22 by MartyG-RealSense