realsenseai/librealsense #9367: Dropped frames while reading bag file of  455 Depth camera

okhsallam, 2021-07-09:
Hi All  

I am trying to read data using Python on windows 10 of a bag file recorded by D455 camera. when I play the bag file on the RealSense viewer, all frames appears.  However when I read it using the python code it only read even frames and drops the odd frames.

A code is shown 

for frame_num in range(frame_to_show-10):
#while True:

    pipe.wait_for_frames()
    profile.get_device().as_playback().set_real_time(False)  #Needed so frames don't get dropped during processing:
    frameset = pipe.wait_for_frames()
    playback.pause()   # Added
    depth_frame = frameset.get_depth_frame()  # Get the depth frame
    Z.append(depth_frame.get_distance(int(120),int(250)))
    Real_frame_num= depth_frame.frame_number
    print(Real_frame_num-5153)
    playback.resume()   # Added

and the Output for portion of printed frames is shown below  

0
2
4
6
8
10
12
14
16

Thanks in advance for help

MartyG-RealSense, 2021-07-10:
Hi @okhsallam   Using the **append** instruction in a pyrealsense2 Python script can be problematic unless the SDK's **Keep()** instruction is used to store frames in memory.  This is because once 16 frames are reached in an append operation then the program cannot process further frames.  This phenomenon is described in the link below.

https://github.com/IntelRealSense/librealsense/issues/946

You could instead try the frame-printing script in the link below.  It has had an issue in some librealsense versions of printing duplicate frame numbers, but may be worth trying to see what results it gives you.

https://github.com/IntelRealSense/librealsense/issues/8949#issuecomment-832507598

okhsallam, 2021-07-11:
@MartyG-RealSense 
Thanks a lot for your reply, I discovered the problem, I did a silly mistake by calling pipe.wait_for_frames() 2 times, that is why I only got the even frames.  But using append did not cause any issue with code and all frames are captured. 

Thanks again

closed 2021-07-11 by okhsallam

MartyG-RealSense, 2021-07-11:
I'm pleased to hear that you found a solution @okhsallam - thanks very much for the update!