realsenseai/librealsense #9370: How to project lidar points obtained from L515 to color image?

kinggreat24, 2021-07-10:
[image]

My projection result seems to wrong.　Is there any solutions to fix it?

MartyG-RealSense, 2021-07-10:
Hi @kinggreat24  What method are you using to align depth to color please?  Is it a point cloud in the **3d** mode of the RealSense Viewer program or with a script that you have created yourself?

kinggreat24, 2021-07-10:
> Hi @kinggreat24 What method are you using to map depth to color please? Is it a point cloud in the **3d** mode of the RealSense Viewer program or with a script that you have created yourself?

I wrote a c++ program myself to project the lidar points obtained with the topic **/camera/depth/color/points**, the detailed processes were  as follows.
- Firstly, I transformed the points to camera coordinate using the extrinsic obtained by the topic **/camera/extrinsics/depth_to_color**. Then the points were projected to the image plane using the camera Intrinsic matrix got from the topic **/camera/color/camera_info**. Finally, those 2D points were drawn to color image(**/camera/color/image_raw**) which had been undistorted already. 

Can you give me some tips?  Thanks very much.

kinggreat24, 2021-07-10:
[image]
The point cloud looks good, but the projection result is not well.

MartyG-RealSense, 2021-07-10:
I see from the image that you have kindly provided that you are visualizing the topics in RViz.  It sounds as though you are using the RealSense ROS wrapper and wrote a C++ node script.  Is that correct, please?

If you are using the RealSense ROS wrapper, an easier approach to produce a pointcloud where color is mapped to depth may be to use the pointcloud example in the wrapper documentation and visualize the pointcloud in RViz.

https://github.com/IntelRealSense/realsense-ros#point-cloud

kinggreat24, 2021-07-10:
> I see from the image that you have kindly provided that you are visualizing the topics in RViz. It sounds as though you are using the RealSense ROS wrapper and wrote a C++ node script. Is that [...]

MartyG-RealSense, 2026-10-23:
This open GitHub issue relating to the RealSense L515 camera is being closed because the L515 model is End of Life (EOL) and no longer supported.  We welcome the creation of new issues relating to supported cameras.

closed 2026-10-23 by MartyG-RealSense