realsenseai/librealsense #9371: realSense2.0 & OpenCV4.5.2, How to get Mat appropriately?

liuluyangdev, 2021-07-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Can someone tell me how to force Frame(realSense) to Mat(OpenCV) appropriately?
Thanks.

MartyG-RealSense, 2021-07-10:
Hi @liuluyangdev   If you are aiming to convert RealSense rs2::frame to Mat in the RealSense Android wrapper using Java, the scripting in the link below may be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/5185#issuecomment-550266254

liuluyangdev, 2021-07-10:
@MartyG-RealSense Thanks.I will test as soon as possible.

liuluyangdev, 2021-07-12:
@MartyG-RealSense Do you know this error?

`E/librs: Exception was thrown during user processing callback: Out of frame resources!
E/librs: Exception was thrown during user processing callback: Out of frame resources!
E/librs: Exception was thrown during user processing callback: Out of frame resources!
E/librs: Exception was thrown during user processing callback: Out of frame resources!
.......`

liuluyangdev, 2021-07-12:
I found the answer.
invoke frame.close(); at the end.

MartyG-RealSense, 2021-07-12:
Thanks very much @liuluyangdev for the update.  Is your problem now solved, please?  Thanks!

liuluyangdev, 2021-07-13:
Yes,@MartyG-RealSense Thanks very much.

MartyG-RealSense, 2021-07-13:
Great to hear, @liuluyangdev - thanks for the confirmation!  As you have achieved a solution, I will close this case.

closed 2021-07-13 by MartyG-RealSense