realsenseai/librealsense #9372: rs-server works, pyrealsense2-net fails to connect: Out of range value for argument def

nixone, 2021-07-11:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Ubuntu  | 
| Kernel Version (Linux Only)    |  4.9.201-tegra | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            | 2.47.0                         | 
| Language                            |  Python                          | 
| Segment			| others                          | 

### Issue Description
On the same jetson nano, I am trying to run rs-server and connect to it using pyrealsense2-net module. `rs-server` seems to be running fine (see below), but connecting to it from python throws an error: `RuntimeError: out of range value for argument "def"`. When `rs-server` is not running, the same code fails with different error: `RuntimeError: [getControls] error: Connection to server failed: Connection refused - -107`, so I assume the python wrapper shouldn't be at fault, since there is some communication obviously going on. Is there anything I can try to make this work in Python?

rs-server running:
```
jetson@jetson:~$ rs-server
Rs-server is running
```

python execution when `rs-server` is running:
```
jetson@jetson:~/workspace/librealsense/build$ python3.8
Python 3.8.0 (default, Feb 25 2021, 22:10:10)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> rsnet = __import__("pyrealsense2-net")
>>> rsnet.net_device("localhost")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: out of range value for argument "def"
```

python execution when `rs-server` is not running:
```
jetson@jetson:~/workspace/librealsense/build$ python3.8
Python 3.8.0 (default, Feb 25 2021, 22:10:10)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> rsnet = __import__("pyrealsense2-net")
>>> rsnet.net_device("localhost")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: [getControls] error: Connection to server failed: Connection refused - -107
```

MartyG-RealSense, 2021-07-11:
Hi @nixone  Intel's white-paper document for the open-source networking system states "Integration with Python, Android and other wrappers in librealsense ecosystem will be provided in future releases based on feedback from the community".  

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-2-building-from-source

The out **of range value for argument "def"** error  that you experienced with re-served has been encountered in a past case in the link below though, where a RealSense team member provides advice.

https://github.com/IntelRealSense/librealsense/issues/7459

If you require a RealSense networking solution that can be integrated with Python, the earlier **EtherSense** system may suit your needs instead, as it was designed specifically for Python and Ubuntu.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

MartyG-RealSense, 2021-07-18:
Do you require further assistance with this case, please?  Thanks!

nixone, 2021-07-22:
Hi @MartyG-RealSense . I went through the white paper document you mentioned also before I opened this issue. Also I went throuth the issue #7459 you mentioned, unfortunately it doesn't provide a solution for me.

It's true that I am using python net device, which is not to be claimed officially supported. Based on the error and the behaviour though, I thought that it might work, I am just doing some mistake in my setup, since I am getting an error from the communication itself, not that I would not be able to even call the methods because of the wrapper not being ready.

MartyG-RealSense, 2021-07-22:
I haven't made a claim about pyrealsense2-net specifically in this discussion, just highlighted that that the open-source ethernet networking system that uses rs-server states in its paper that Python integration with that system may be added in future.

I researched this subject again but was not able to find any references about using rs-server with pyrealsense2.  I would encourage any RealSense users reading this who have had success with it to contribute information about their method in a comment on this discussion.

nixone, 2021-07-22:
Oh yeah, sorry, I meant "Intel", as on https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras it's mentioned: "Integration with python, Android and other wrappers in librealsense ecosystem will be provided in future releases based on feedback from the community." Yeah, let's keep this opened if possible for anyone to see maybe a different solution. Thanks.

MartyG-RealSense, 2021-08-04:
Hi @nixone  Do you require further assistance with this case, please?  Thanks!

nixone, 2021-08-04:
@MartyG-RealSense feel free to close this. I didn't manage to make it work, but since this is so far not officially supported, I am ok with you closing this issue.

MartyG-RealSense, 2021-08-04:
Thanks very much for the update @nixone - I'm sorry to hear that you weren't successful yet.  Good luck in the future!

closed 2021-08-04 by MartyG-RealSense

MartyG-RealSense, 2021-08-11:
Hi @nixone  I thought that you may be interested to know that the new **2.49.0** librealsense version that is now available adds an lrs-net viewer for Python called **net-viewer.py** to the Examples section of the Python wrapper.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/net_viewer.py