realsenseai/librealsense #9373: pyrealsense is slow on jetson

gidisza, 2021-07-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |   Linux Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  (4.9.201-tegra)                                         | 
| Platform                                 |NVIDIA Jetson|
| SDK Version                            |  2.48.0                          | 
| Language                            |  {CPP/python}                          | 
| Segment			|  Robot                        | 

Hi,
I'm using pyrealsense in my project and I have noticed that the execution time of the decimation filter on Jetson Xavier NX is very slow (80 ms for frame), while on the same machine a corresponding CPP code runs much faster(5ms).
The same python code on PC runs fast as the CPP code does.
What can I do to improve the performance of pyrealsense python module on the jetson machine?

MartyG-RealSense, 2021-07-12:
Hi @gidisza  Are you building librealsense with CMake and setting the build flag **BUILD_WITH_OPENMP** to True?  I have seen a number of cases on Python where this maximizes CPU usage at 100%, whilst setting BUILD_WITH_OPENMP to False in those cases caused a significant reduction in CPU usage.

If you have built librealsense with BUILD_WITH_OPENMP = true then you could compare CPU usage % during running of your Python and C++ applications by using a system performance monitoring tool such as **htop** in Ubuntu to see whether there is a significant difference in % usage between the two languages.

gidisza, 2021-07-13:
I compiled it without OPENMP. Anyway, I'll try to compare the CPU usage of the CPP and python apps. 
Is there anything else that can cause that behavior?

MartyG-RealSense, 2021-07-13:
Did you use **-DCMAKE_BUILD_TYPE=Release** in your CMake instruction on Jetson to ensure that the SDK is built with compiler optimizations?  Otherwise, librealsense would be built in the slower Debug mode.

closed 2021-07-26 by MartyG-RealSense