realsenseai/librealsense #9374: Issue with depth sensor on L515

GitNzino, 2021-07-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    |  5.4.0-77-generic                                 | 
| Platform                                 | PC |
| SDK Version                            |   2.48.0                     | 
| Language                            |  C++/python/ROS                         | 
| Segment			|  Lidar                   | 

### Issue Description
The depth sensor seems to not work properly either in code and in realsense app. I suggest that resolving the problem for realsense app the L515 will work also with my script because some week before all worked properly.

Error showe from realsense app:
[image]

Same error but terminal output:
[image]

Camera info:
[image]

MartyG-RealSense, 2021-07-12:
Hi @GitNzino  Some previous occurrences of **xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error** on the L515 camera model when enabling the depth sensor can be found in the link below:

https://github.com/IntelRealSense/librealsense/issues/8723

Could you first try unplugging the end of the USB cable that is inserted in the camera, turning it around the opposite way and re-inserting it please (the connector is two-way insertion on the camera end of the cable).  Does doing so stop the error from appearing when enabling the depth sensor?

GitNzino, 2021-07-12:
Thank you for reading and responding!

I have already read and this issue, but also trying to rotate the USB type C connectors nothing change.

I want also tell you that I'm using the original Type C connectors and when I start Realsense exe the Red ligth on camera does not appear.

GitNzino, 2021-07-12:
In addiction I have
 tried the Realsense exe also for Windows, but not works neither

MartyG-RealSense, 2021-07-12:
Thanks very much for the additional details.  I will seek advice from Intel about your particular case.  Thank you for your patience!

MartyG-RealSense, 2021-07-12:
As the camera is able to be detected in the Viewer, and display its details, let's explore the possibility of a firmware-related problem first.  I see that you are using firmware **1.5.8.1** - if you have updated the firmware recently, did the firmware update coincide with the start of the problem or happen some tine later?

GitNzino, 2021-07-12:
I have tried to installe the 1.5.4.1 firmware, but with Ubuntu does not work.

At the start I had no problem with camera, after I leave the camera to another person but I cannot known if he done a firmware upgrade meanwhile, and then the camera returned with this problem.

MartyG-RealSense, 2021-07-12:
Did you download the firmware file from the L515 firmware page in the link below and then use the **Update Firmware** option in the RealSense Viewer program to select the downloaded firmware file to initiate the update?

https://dev.intelrealsense.com/docs/firmware-releases-l500

GitNzino, 2021-07-12:
Yes I followed all the step in that page

MartyG-RealSense, 2021-07-12:
If the firmware that is shown in the Viewer's **Info** display is currently **1.5.8.1** then you have the most recent firmware installed.

Although the depth stream will not enable, is the RGB stream able to be enabled correctly?

GitNzino, 2021-07-12:
I tried with firmware 1.5.4.1 but when I saw that not worked I re-installed 1.5.8.1.

Yes the other sensor work fine, the only that not works i the depth sensor.

Honestly I don't remember what version of SDK I have, but on Linux i update the libraries very often so if in June you released a new version of SDK then I changed SDK version

MartyG-RealSense, 2021-07-12:
2.48.0 was released at the very end of June .  Before that, 2.47.0 was released on 6 June.

GitNzino, 2021-07-12:
Well fine!

Can you suggest a correct method to downgrade the SDK to 2.46 or less?

MartyG-RealSense, 2021-07-12:
Do you build librealsense from packages or from source code please?

GitNzino, 2021-07-12:
I build it with apt install

MartyG-RealSense, 2021-07-12:
Before 2.47.0, the next oldest version is 2.45.0 as there is not a 2.46.0.

My understanding is that installing a specific librealsense package version is more complex than just installing the latest one, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785

Installing from source code may be more straightforward, as you just select the source-code folder version that you want from the 'Assets' file lists on the SDK **Releases** page and build it with CMake.

https://github.com/IntelRealSense/librealsense/releases

Changing versions is considerably easier if you have a Windows computer available instead of Ubuntu, as you can use the automated installer file in the 'Assets' list and the installation simply overwrites the previously installed version.

[image]

GitNzino, 2021-07-13:
I get the SDK 2.47.0 on Ubuntu, but the error is still present.

GitNzino, 2021-07-13:
[image]

MartyG-RealSense, 2021-07-13:
Could you try the **rs-capture** SDK example program please to test whether the problem is only in the RealSense Viewer or affecting depth streaming in general?

You can launch this example from the Ubuntu terminal with the command **rs-capture** or on Windows using the instructions in the link below.

https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

GitNzino, 2021-07-13:
Ok thank you!

I have tried, and i obtained this error:
```
13/07 11:17:09,850 ERROR [139888354895744] (types.h:310) xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
RealSense error calling rs2_pipeline_start(pipe:0x55c697fa2640):
    xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
 13/07 11:17:09,850 ERROR [139888354895744] (l500-depth.cpp:576) Exception caught in l500_depth_sensor::open
```

And no window opened

MartyG-RealSense, 2021-07-13:
I will check with Intel again for further advice and update you when I have information.  I recommend putting the problem aside until then.  I appreciate your continued patience.

GitNzino, 2021-07-13:
Thank you for support and patience too!

See you soon, have a nice time

MartyG-RealSense, 2021-07-13:
Hi @GitNzino  After discussing your case with Intel, they requested that you return your L515 camera to wherever you purchased it from, using the **RMA** (Return Merchandise Authorization) returns process of that supplier.

GitNzino, 2021-07-14:
Ok, thank you!

I'm closing the issue

closed 2021-07-14 by GitNzino