realsenseai/librealsense #9375: Video Resolution Of Realsense D455

Try-Hello, 2021-07-12:
The resolution corresponds to the angle of the field of view。I know （640，480），（1280， 720）. Are there other resolutions?
Is there a resolution corresponding to 90°?

W = 1280
H = 720   # 1280 720  （424, 240）   （640, 480）

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, W, H, rs.format.z16, 5)
config.enable_stream(rs.stream.color, W, H, rs.format.bgr8, 15)
print("[INFO] start streaming...")
pipeline.start(config)

MartyG-RealSense, 2021-07-12:
Hi @Try-Hello  Support for RGB resolutions 640x360 and 640x480 were added in L515 firmware version **1.5.8.1**.  A list of supported resolutions for depth and color at the time of writing this when using the latest firmware is shown below.

[image]

In regard to the question about an L515-supported resolution corresponding to 90° , I do not have information about that.

MartyG-RealSense, 2021-07-12:
Hi @Try-Hello   There is a table of supported resolutions for D455 (extracted from the data sheet document for the 400 Series cameras) in the image below.

[image]

In regard to a resolution that corresponds to 90° FOV, I do not have information available about that.

Try-Hello, 2021-07-13:
> Hi @Try-Hello There is a table of supported resolution and FPS modes for D455 (extracted from the data sheet document for the 400 Series cameras) in the image below. The RGB modes are listed in the **Color stream from RGB camera** section.
> 
> [image]
> 
> In regard to a resolution that corresponds to 90° FOV, I do not have information available about that.

thanks a lot。

closed 2021-07-24 by MartyG-RealSense