realsenseai/librealsense #9376: net_device : No device connected when starting pipeline

Lblenner, 2021-07-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |  Ubuntu 20.04.2 LTS  | 
| Kernel Version (Linux Only)    |  5.8.0-59-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.48.0                          | 
| Language                            |  C                          | 
| Segment			|  others                         | 

### Issue Description

Following [this tutorial](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras), I created a simple program to get frames from another PC.

I compiled librealsenses from master (2.48.0) with the flags `-DBUILD_NETWORK_DEVICE=ON` 
`-DFORCE_RSUSB_BACKEND=ON` on two pc.

I start `rs-server` on one of the pc.
The `realsense-viewer` is able to connect and display the image from the other pc.
The program works when run on the same pc as `rs-server`.

But when the program is run on the other pc, it is only able to retrieve metadata about the camera (serial number, name...).

I get the following error message when calling `start` on the pipeline :

```
terminate called after throwing an instance of 'rs2::error'
  what():  No device connected
```

The two pc were at first just part of the same wifi, but connecting them using an ethernet cable did not change the behavior.
What could be the cause of the problem ? 

Here is the program

```cpp
#include <librealsense2/rs.hpp>
#include <librealsense2-net/rs_net.hpp>
#include <librealsense2/h/rs_pipeline.h>
#include <iostream>

using namespace std;

int main()
{
	rs2::net_device dev("ip");
	rs2::context ctx;
	dev.add_to(ctx);

	/* Metadata from context */
	auto g = ctx.query_devices();
	string serial_number;

	for (auto t : g)
	{
		auto name = rs2_camera_info::RS2_CAMERA_INFO_NAME;
		auto info = t.get_info(name);
		auto info_descriptor = rs2_camera_info_to_string(name);

		cout << info_descriptor << " : " << info << endl;

		name = rs2_camera_info::RS2_CAMERA_INFO_SERIAL_NUMBER;
		info = t.get_info(name);
		info_descriptor = rs2_camera_info_to_string(name);

		cout << info_descriptor << " : " << info << endl;

		serial_number= info;
	}

	/* Starting pipeline */
	rs2::pipeline p(ctx);

	rs2::config cfg;
	cfg.enable_device(serial_number);
	cfg.enable_stream(RS2_STREAM_COLOR, 0);

	p.start(cfg);

	while(true) {
		p.wait_for_frames();
	}
}
```

MartyG-RealSense, 2021-07-12:
Hi @Lblenner  There is a past rs-server C++ case in the link below that also had problems with starting the pipeline when a cfg custom configuration was used (the script worked when 'cfg' was removed from the brackets of the pipe start instruction).  The problem in that case was found to be with the cfg definition for the color stream.

https://github.com/IntelRealSense/librealsense/issues/8423

The cfg instruction below worked for the RealSense user in that particular case:

**cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8 , 30);**

https://github.com/IntelRealSense/librealsense/issues/8423#issuecomment-795557815

Lblenner, 2021-07-13:
I did not find this issue while searching for solutions.

Using this format for the configurations solved the issue.

Thanks.

closed 2021-07-13 by Lblenner

MartyG-RealSense, 2021-07-13:
Excellent news @Lblenner that the suggested solution worked for you - thanks for the update!