realsenseai/librealsense #9381: Recognized as usb 2.1 when inserted into usb 3.0 port

hyebin-rdv, 2021-07-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 5.4.0-77-generic | 
| Platform                                 | PC |
| SDK Version                            |  2.45.0 | 
| Language                            | c++, python  | 

### Issue Description
It's exactly the title. It was connected to the pc's 3.0 port using the L515 Genuine 3.0 cable, but is recognized as 2.1.
How to fix it?

MartyG-RealSense, 2021-07-13:
Hi @hyebin-rdv   Please try unplugging the USB cable from the end that is inserted into the L515 camera, turning the connector around the other way and re-inserting it (the Type-C cable is two-way insertion on the camera end).

hyebin-rdv, 2021-07-13:
@MartyG-RealSense Oh, it's working! Do you happen to know why? Is it affected by the direction?

MartyG-RealSense, 2021-07-13:
Great to hear that the solution worked for you!  

It is just a characteristic of this particular camera model that the orientation of the USB cable at the camera end can have an influence on its operation.

closed 2021-07-13 by hyebin-rdv