realsenseai/librealsense #9384: L515 Hardware Sync Mode - Erratic data

korroktheslavemaster, 2021-07-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.05.00 | 
| Operating System & Version |  Ubuntu 20.04.2 LTS | 
| Kernel Version (Linux Only)    |  5.8.0-43-generic | 
| Platform                                 | PC  |
| SDK Version                            |  2.42.0 on system, 2.44.0 in realsense-viewer                          | 
| Language                            | C++                      | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello
I'm trying to run multiple L515 cameras in hardware sync mode, as described here https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup . The issue I am facing happens with even one camera connected so I'm not going to discuss multiple cameras.
I am using a Jetson Nano and some custom circuit to generate 160ms on/off triggers at 3.3V. Validity of this signal I have verified using an oscilloscope. A single L515 is connected to my laptop. When I open `realsense-viewer` and enable `Inter Cam Sync Mode`, I am able to see that the camera turns on/off based on the pulses and keeps sending pointcloud data for atleast several minutes without any issue.
However, when I try to run my own code with hardware sync, I have issues. PFB my sample code that is basically copied from https://dev.intelrealsense.com/docs/rs-distance but with inter cam sync enabled.
```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2015-2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API

#include <algorithm>            // std::min, std::max
#include <iostream>
// Helper functions

int main(int argc, char * argv[]) try
{

    // Declare pointcloud object, for calculating pointclouds and texture mappings
    rs2::pointcloud pc;
    // We want the points object to be persistent so we can display the last cloud when a frame drops
    rs2::points points;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH);
    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    auto r = cfg.resolve(pipe);

    r.get_device().query_sensors().front().set_option(RS2_OPTION_INTER_CAM_SYNC_MODE, 1.f);

    // Start streaming with default recommended configuration
    pipe.start();

    while (true) // Application still alive?
    {
        // Wait for the next set of frames from the camera
        auto frames = pipe.wait_for_frames();

        auto depth = frames.get_depth_frame();
        // Get the depth frame's dimensions
        float width = depth.get_width();
        float height = depth.get_height();
        // Query the distance from the camera to the object in the center of the image
        float dist_to_center = depth.get_distance(width / 2, height / 2);
        std::cout << "The camera is facing an object " << dist_to_center << " meters away \n";
    }

    return EXIT_SUCCESS;
[... 167 lines cut ...]
The camera is facing an object 2.188 meters away 
The camera is facing an object 2.19 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
The camera is facing an object 2.192 meters away 
```
(after around 1-2 min the distance value stops changing at all, even if I put my hand in front of camera)
As you can see the data updates at an erratic rate instead of a regular interval. I also notice that instead of the laser regularly turning on/off at 160ms intervals, there are 'gaps' where it fails to turn on. After some time (maybe 15 mins) the laser just goes permanently off (I haven't been able to easily reproduce the last part though). PFA a gif of the laser on/off.
If I *disable* hardware sync in this same piece of code (set value as 0.0f) I get a correct stream without any issues.

The behaviour is same when I attach the camera to jetson nano and run the program on it. In fact on the jetson nano, even through the realsense-viewer, I'm not able to get a good hardware-sync stream. Atleast the realsense-viewer on my laptop is able to run the camera correctly in hardware sync mode, which gives me hope that this is a library issue and not an issue with the hardware/electronics. I have 3 L515 cameras and I've tested on all of them, they give the same behaviour.

If someone has managed to run the hardware mode correctly on ubuntu, please do share your library versions (librealsense version, kernel version, l515 firmware version). Thanks.

[image]

MartyG-RealSense, 2021-07-13:
Hi @korroktheslavemaster   The first point I would highlight is that your pipe start instruction is **pipe.start()** without any instruction inside the brackets.  Your **cfg** definition should be placed inside the brackets, otherwise the custom configuration will be ignored and the default configuration of the particular camera model will be applied instead when the pipeline starts.  So the pipe start instruction should be:

**pipe.start(cfg);**

In regard to the 3.3V signal, have you confirmed that it remains stable and constant throughout the session, even when the performance of the script is apparently declining?

The main issue though may be the white-paper document's suggestion that INTER_CAM_SYNC_MODE is set to '1'.  That is the Master camera configuration.  The paper states that **all** cameras should be set to Slave in a multi-camera system.  The INTER_CAM_SYNC_MODE value for Slave is '2'.

Could you try changing '1' to '2' and see whether it makes a difference, please.

korroktheslavemaster, 2021-07-13:
Hi
Thanks for the quick response
1. The `pipe.start(cfg)` did change the program behaviour a little bit. Instead of multiple print statements per value, now I'm getting only a single print statement. However the laser pulses are still just as erratic and eventually the data stops coming in. PFB the new logs.
```
The camera is facing an object 2.08875 meters away 
The camera is facing an object 2.08425 meters away 
The camera is facing an object 2.09025 meters away 
The camera is facing an object 2.0895 meters away 
The camera is facing an object 2.08825 meters away 
The camera is facing an object 2.09175 meters away 
The camera is facing an object 2.08675 meters away 
The camera is facing an object 2.08525 meters away 
The camera is facing an object 2.091 meters away 
The camera is facing an object 2.08875 meters away 
The camera is facing an object 2.089 meters away 
The camera is facing an object 2.08725 meters away 
The camera is facing an object 2.0915 meters away 
The camera is facing an object 2.087 meters away 
The camera is facing an object 2.08875 meters away 
The camera is facing an object 2.086 meters away 
The camera is facing an object 2.08875 meters away 
The camera is facing an object 2.09 meters away 
The camera is facing an object 2.0915 meters away 
The camera is facing an object 2.08875 meters away 
The camera is facing an object 2.08725 meters away 
The camera is facing an object 2.0875 meters away 
The camera is facing an object 2.092 meters away 
The camera is facing an object 2.089 meters away 
The camera is facing an object 2.08825 meters away 
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x5583df7cb090):
    Frame didn't arrive within 15000
```
2. Yes I have confirmed that even while the camera is showing weird behaviour, the hardware signal is correct. Also the camera works perfectly fine through `realsense-viewer` in inter cam sync mode for long periods of time.
3. Not sure what your point is, I'm using the value 1 only.

MartyG-RealSense, 2021-07-13:
1.  Adapting **rs-distance** in this way may not be the best way to test a multi-camera setup (even though you are currently only testing with one camera), since the original script is a single-camera application.  Multiple-camera scripts (like **rs-multicam**) use different code that builds a list of all the attached cameras and can use that list to communicate with the cameras individually.  So even if you got hardware sync working, in its current state the script may only use the first camera of the set that it identifies and ignore the rest of the cameras.

Another complicating factor is that the **rs-distance** example is 'C' programming code and not C++ (as the Viewer is).  If you are using C++ then it may be better to use a C++ program as the foundation for your test script.   The text-only C++ example **rs-hello-realsense**, which also prints the distance of the center of the camera view, may work better.

https://github.com/IntelRealSense/librealsense/tree/master/examples/hello-realsense

2.  Thanks very much for the confirmation.

3.  I was just 'thinking out loud' to explain the difference between the 400 Series and L515 hardware sync systems.  Future readers of this case could be confused unless it was explained, since under the 400 Series version of the sync system, '1' is the Master configuration and not Slave.  :)

korroktheslavemaster, 2021-07-13:
This is not my intended use case, I was just using a simplified example to show the issue. My actual application uses realsense-ros and consumes the pointcloud topics published by 3 different cameras. I was able to narrow down the issue to the base librealsense library itself, which is why I opened the issue here and not on `realsense-ros`. Also my code is the C++ version only, I think I shared the wrong link.

korroktheslavemaster, 2021-07-13:
In the multicamera example from the documentation, I could not find which versions of librealsense and L515 firmware they have tested it on. Is it possible to get those versions so I can try to duplicate the exact setup? @MartyG-RealSense

MartyG-RealSense, 2021-07-13:
If you are using RealSense ROS, Intel's multiple camera point cloud tutorial for generating point clouds and stitching them together into a combined cloud may meet your needs instead of using hardware sync. It describes how to use all the cameras on the same computer or spread them across more than one computer.  It was designed for the 400 Series cameras but may be adaptable for L515.

https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

If you would still like to use Inter Cam Sync Mode, it can also be controlled from within RealSense ROS instead of setting it up in librealsense.

https://github.com/IntelRealSense/realsense-ros/issues/984#issuecomment-707729244

Whilst SDK and firmware are not listed in the L515 hardware sync paper, as it was published in early April 2021 I would speculate that it may have been developed on February's **2.42.0** and firmware **1.5.4.1** (the L515 firmware that was recommended for 2.42.0).  This is based on 2.43.0 not being released until 21 March 2021.

korroktheslavemaster, 2021-07-13:
I require inter-cam sync since otherwise there is a lot of interference as the cameras are pointing towards each other. I have already controlled inter cam sync mode using realsense-ros which is where I first faced this issue. However the issue is there even when I skip the realsense-ros layer and use the native librealsense library.
I suspect there is some kind of memory leak in the firmware which is causing the camera to stop responding after a while. Can someone from the development team for hardware sync help me out with this?

MartyG-RealSense, 2021-07-13:
Although the white paper is currently dated April 2021, there is a reference in the link below to it being posted in August 2020, which would make the probable SDK combination 2.36.0 / 1.4.1.0

https://github.com/IntelRealSense/librealsense/issues/6641#issuecomment-673587955

In regard to advice from L515 hardware sync specialists, you would likely need to have a Non Disclosure Agreement (NDA) with Intel to access such assistance.  If you do not have an NDA then I will act as your RealSense support contact.

korroktheslavemaster, 2021-07-14:
I tried by downgrading firmware to 1.4.1.0 and SDK combination to 2.40 (could not get a prebuilt binary for 2.36 and I remember having issues running L515 at all with < 2.40 SDK). Still facing the exact same issue.
Please let me know how to proceed.

MartyG-RealSense, 2021-07-14:
Okay, so in summary ... you want to use hardware sync in RealSense ROS to compensate for interference when multiple L515 cameras are facing towards each other.  Your external sync trigger device provides a consistently stable signal but you are experiencing instabilities in your program's output.

Since your preference is to use RealSense ROS, it is probably best to forget about librealsense sync scripts and focus only on ROS in order to avoid confusion, and test with only one L515 camera at first.

Can you confirm please that when you defined **inter_cam_sync_mode** in ROS, you set it to '1' (slave).

If you did set inter_cam_sync_mode to '1', how did you test hardware sync in ROS to determine whether or not it was working correctly?

korroktheslavemaster, 2021-07-15:
Yes, I set it to 1 in ROS through rqt_reconfigure. I faced the exact same behavior - the LIDAR laser is blinking irregularly as in the gif above, and the data on the pointcloud topic of ROS comes at erratic rates and eventually stops coming at all.
I would like to point out that I am able to get a good stream in hardware sync mode when I use `realsense-viewer` on my laptop. So I can't see this as being a USB connection issue.

MartyG-RealSense, 2021-07-15:
I researched your case further and located a recent case of a RealSense ROS user who had been having problems with hardware sync with two L515s on a Jetson Nano.  They set inter_cam_sync_mode using dynamic_reconfigure.

They recently posted the details of what they did to get it working.

https://github.com/IntelRealSense/realsense-ros/issues/1950#issuecomment-878626019

MartyG-RealSense, 2021-07-21:
Hi @korroktheslavemaster  Do you require further assistance with this case, please?  Thanks!

korroktheslavemaster, 2021-07-21:
Yes, I'm still working on the suggestion on the other thread, will let you
know

closed 2021-08-18 by MartyG-RealSense