realsenseai/librealsense #9386: Low FPS on L515 camera

alexandru-oana, 2021-07-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (01.05.08.01) | 
| Operating System & Version |   {Ubuntu 20.04}  | 
| Kernel Version (Linux Only)    |  (5.8.0-59)                                         | 
| Platform                                 | Intel  |
| SDK Version                            |  { 2.48 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others}                          | 

I have an issue regarding low fps performance of L515.
I've used the same camera with same librealsense setup on two different machines, both with Ubuntu 20.04.

The machine that has low FPS performance issue has kernel: 5.8.0-59-generic.
The other one, on which the camera runs fine has 5.4.0-77. 
Also, i mention that both machines kernel has the uvc patch.

I've tried using librealsense 2.47 and 2.48 on the machine running 5.8.0-59-generic.

The only warning that i see on both machines is in dmesg:
Failed to query GET_CUR UVC control 1 on unit 3: -32 ( exp. 1024 )
So, I've tried to disable auto-exposure, but i don't think that's the root cause, because i get the same warning on the machine that is running normal.

My setup is this:
ros2 launch realsense2_camera rs_launch.py color_width:=1920 color_height:=1080 color_fps:=15. depth_width:=1024 depth_height:=768 depth_fps:=$depth. infra_width:=1024 infra_height:=768 infra_fps:=30.

[INFO] [launch]: All log files can be found below /opt/ros/foxy/.ros/ros_log/2021-07-13-13-02-35-98171
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [54]
[realsense2_camera_node-1] [INFO] [1626181356.270150072] [RealSenseCameraNode]: RealSense ROS v3.2.0
[realsense2_camera_node-1] [INFO] [1626181356.270235202] [RealSenseCameraNode]: Built with LibRealSense v2.48.0
[realsense2_camera_node-1] [INFO] [1626181356.270249215] [RealSenseCameraNode]: Running with LibRealSense v2.48.0
[realsense2_camera_node-1] [INFO] [1626181356.343920549] [RealSenseCameraNode]: Device with serial number f02xxxx \was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1626181356.343945963] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1626181356.343949912] [RealSenseCameraNode]: Device with name Intel RealSense L515 was found.
[realsense2_camera_node-1] [INFO] [1626181356.344108165] [RealSenseCameraNode]: Device with port number 4-1 was found.
[realsense2_camera_node-1] [INFO] [1626181356.344114133] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1626181356.345972094] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [INFO] [1626181356.346539777] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-1] [INFO] [1626181356.346547190] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1626181356.346552631] [RealSenseCameraNode]: Device Name: Intel RealSense L515
[realsense2_camera_node-1] [INFO] [1626181356.346556962] [RealSenseCameraNode]: Device Serial No: f02xxxxx
[realsense2_camera_node-1] [INFO] [1626181356.346561214] [RealSenseCameraNode]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1626181356.346565828] [RealSenseCameraNode]: Device FW version: 01.05.08.01
[realsense2_camera_node-1] [INFO] [1626181356.346569913] [RealSenseCameraNode]: Device Product ID: 0x0B64
[realsense2_camera_node-1] [INFO] [1626181356.346573726] [RealSenseCameraNode]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO] [1626181356.346577554] [RealSenseCameraNode]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1626181356.346581342] [RealSenseCameraNode]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1626181356.346588667] [RealSenseCameraNode]: Device Sensors: 
[realsense2_camera_node-1] [INFO] [1626181356.347023769] [RealSenseCameraNode]: L500 Depth Sensor was found.
[realsense2_camera_node-1] [INFO] [1626181356.348161052] [RealSenseCameraNode]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1626181356.348235317] [RealSenseCameraNode]: Motion Module was found.
[realsense2_camera_node-1] [INFO] [1626181356.348244727] [RealSenseCameraNode]: (Infrared, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1626181356.348250609] [RealSenseCameraNode]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1626181356.348256373] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1626181356.348261492] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1626181356.348267924] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1626181356.348272653] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1626181356.348278861] [RealSenseCameraNode]: num_filters: 0
[realsense2_camera_node-1] [INFO] [1626181356.348284455] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1626181356.353021235] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1626181356.354068702] [RealSenseCameraNode]: depth stream is enabled - width: 1024, height: 768, fps: 30, Format: Z16
[realsense2_camera_node-1] [INFO] [1626181356.354311670] [RealSenseCameraNode]: infra stream is enabled - width: 1024, height: 768, fps: 30, Format: Y8
[realsense2_camera_node-1] [INFO] [1626181356.356030774] [RealSenseCameraNode]: color stream is enabled - width: 1920, height: 1080, fps: 15, Format: RGB8
[realsense2_camera_node-1] [INFO] [1626181356.356055031] [RealSenseCameraNode]: setupPublishers...
[realsense2_camera_node-1] [INFO] [1626181356.358910295] [RealSenseCameraNode]: setupStreams...
[realsense2_camera_node-1] [INFO] [1626181356.625144922] [RealSenseCameraNode]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO] [1626181356.774198356] [RealSenseCameraNode]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1626181356.854173052] [RealSenseCameraNode]: 
[realsense2_camera_node-1] [WARN] [1626181356.854228210] [RealSenseCameraNode]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.

Here is the performance:
ros2 topic hz /camera/color/image_raw
average rate: 8.837
	min: 0.065s max: 0.199s std dev: 0.04161s window: 10
average rate: 7.738
	min: 0.065s max: 0.267s std dev: 0.05741s window: 17
average rate: 6.604
	min: 0.065s max: 0.336s std dev: 0.08268s window: 22
average rate: 6.690
	min: 0.065s max: 0.336s std dev: 0.08466s window: 29
average rate: 6.221
	min: 0.065s max: 0.336s std dev: 0.08700s window: 34
average rate: 6.339
	min: 0.064s max: 0.336s std dev: 0.08673s window: 41
average rate: 7.168

Also, instead of 30 fps i have much less.

Any suggestions ?
Thanks

MartyG-RealSense, 2021-07-13:
Hi @alexandru-oana  There was a case this week involving unstable performance on librealsense 2.48.0 with kernel 5.8.0-59-generic that was resolved by the RealSense user choosing kernel **5.8.0-43-generic** instead.

https://github.com/IntelRealSense/librealsense/issues/9346#issuecomment-878534334

Your custom resolution / FPS settings look fine and the launch has accepted them as valid.

Do you see any improvement in performance if you add **initial_reset:=true** to the launch instruction to reset the camera at launch?

alexandru-oana, 2021-07-14:
@MartyG-RealSense  Thanks for the fast answer.
No, if i use initial_reset:=true i still get low performance and 
Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-xxxx

I will try with kernel .8.0-43-generic

MartyG-RealSense, 2021-07-20:
Hi @alexandru-oana  Do you have an update about this case that you can provide, please?  Thanks!

alexandru-oana, 2021-07-20:
Hi @MartyG-RealSense 
Didn't manage to test against 5.8.0-43 due to some technical issues, but is somehow irrelevant, at least for me, because compatibility with 5.10 is the one that i need. I've tested against 5.10 and again, same problem with low performance.

Any plans to support 5.10 LTS ?
Thanks

MartyG-RealSense, 2021-07-20:
It has been shown to be possible to use kernels newer than the most recent officially supported one (5.8 at the time of writing this).  As you have experienced though, the results may be unpredictable.  It is possible that 5.10 may be officially supported at a future date since the SDK does add support for further kernels over time, though I do not have a time estimate that I can provide for the implementation of such support.

If your computer needs to use kernel 5.10 then an approach that may work for you is to build librealsense from source code with the **RSUSB** installation method and then build the RealSense ROS wrapper from source code.  The RSUSB installation process requires an internet connection but is not dependent on Linux versions or kernel versions and does not require kernel patching.  This approach would allow you to have kernel 5.10 on your computer for non-RealSense applications that require it whilst still enabling librealsense to work correctly.

MartyG-RealSense, 2021-07-26:
Hi @alexandru-oana  Do you require further assistance with this case, please?  Thanks!

alexandru-oana, 2021-07-27:
Not for the moment. My use case is not very straight forward and i cannot make use of your suggestion that implies compiling librealsense ( is time consuming ).
I appreciate very much you quick responses !
Thanks, please close this issue.

closed 2021-07-27 by alexandru-oana

MartyG-RealSense, 2021-07-27:
Thanks very much for the update.

I would add that if compile time is a problem for you then you can add a '-j'instruction to the CMake build instruction to speed up the build process by taking advantage of multiple cores in a CPU.  The higher the processor's specification, the higher the j-value it can use.

To work out a suitable j-number, you can find out the number of cores in your CPU and double it.  For example, a good j-value for a common quad-core CPU may be **8**, whilst a higher-end processor could use **16**.  

An example of a build instruction with a j-value of '8' would be as follows:

sudo make uninstall && make clean && **make -j8** && sudo make install