realsenseai/librealsense #9392: about get_option and set_option function of exposure time.

vislero, 2021-07-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i} | 
| Firmware Version                   | 05.12.14.50 | 
| Operating System & Version |  Linux(Ubuntu 18.04)| 
| Kernel Version (Linux Only)    |  5.4.0-77-generic                                   | 
| Platform                                 | PC |
| SDK Version                            |          v.2.48.0                | 
| Language                            |  {C++ / OpenCV}                          | 
| Segment			|  {Robot                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello.

We are testing with d-435i , T-265 about exposure time.

In test, we used get_option(RS2_OPTION_EXPOSURE) to get exposure time , and set_option(rs2_option::RS2_OPTION_EXPOSURE, exposure_value) to set exposure time to deivce.

When we checked the processing time of the above functions, T-265 is faster than D-435i, and D-435i's processing time is almost 100 times T-265's.

So, could you tell me why processing time of the D-435i is so slow? And are they any other methods faster than get_option, set_option?

thank you.

RealSenseSupport, 2021-07-16:
Please take a look at #4589 in which the similar topic was discussed in regard to the elapsed time for manually changing the exposure time. Thank you.

RealSenseSupport, 2021-07-23:
Duplicate of #4589

closed 2021-07-23 by RealSenseSupport