realsenseai/librealsense #9393: RULE_LAUNCH_LINK should not be set to ccache

browdus, 2021-07-14:
`CMake/global_config.cmake` sets `set_property(GLOBAL PROPERTY RULE_LAUNCH_LINK ccache)` if ccache is enabled and present. This is problematic for a few reasons:
1. ccache cannot cache linking operations (see https://github.com/mozilla/sccache/issues/550, https://crascit.com/2016/04/09/using-ccache-with-cmake/, https://ccache.dev/)

From the ccache website:

> Other types of compilations (multi-file compilation, **linking**, etc) will silently fall back to running the real compiler.

While setting `RULE_LAUNCH_LINK` to ccache is relatively benign, it provides no benefits. It also adds an unnecessary, admittedly small, overhead to those operations.

2. Passing linking operations to other compiler wrappers is not always so benign. This makes using ccache's `CCACHE_PREFIX` while compiling the realsense library impossible in some configurations.

In summary, setting `RULE_LAUNCH_LINK` to ccache provides no benefits, but actively breaks some build chain configurations.

Solution: simply remove the line of code that sets `RULE_LAUNCH_LINK`.

MartyG-RealSense, 2021-07-15:
Hi @browdus  The best way to approach this request may be to add a note to installation documentation that suggests this method for those who encounter compilation problems.  As such, I have added a **Documentation** tag to this case so that it can be considered by the RealSense team at a future date.  This case should be kept open in the meantime.  Thanks very much for the suggestion!

browdus, 2021-07-15:
Documentation might help, but I don't think there is any downside to just removing it. You wouldn't need to document this potential issue in that case either.

At least for our project, it would be nice to not have to modify any of the realsense cmake content (and maintain those modifications across updates). We do want to use ccache, but realsense fails to build out of the box for us because of its `RULE_LAUNCH_LINK` setting.

Edit: maybe there is some cmake magic we can use to override the realsense ccache config with out own, so as to avoid modifying realsense?

MartyG-RealSense, 2021-07-15:
I have highlighted your request for a CMake change to Intel.

closed 2026-10-23 by MartyG-RealSense