realsenseai/librealsense #9395: Updating FW on 2 RealSense cameras - one works, other is put into recovery mode

etorbenson, 2021-07-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | 05.11.01.100, 05.12.12.100 | 
| Operating System & Version | Win 10 (multiple PCs, 1909 and newer) | 
| Kernel Version (Linux Only)    |  N/A | 
| Platform                                 | PC |
| SDK Version                            |  N/A, standalone Windows FW tool latest version                        | 
| Language                            |  N/A  | 
| Segment			|  others                          | 

### Issue Description
We have devices in the field that require a fully unattended solution to perform RealSense FW upgrades. These devices have two RealSense D4xx cameras attached. Running the standalone Windows FW upgrade tool found [here](https://github.com/IntelRealSense/librealsense/releases/tag/v2.47.0), latest release, we are able to upgrade one of the cameras and have it report the correct new FW version. Targeting the second camera in the unit after the first succeeded causes the second to be put into Recovery Mode. It takes several attempts using the "-r" switch of the FW update tool to get the camera back online.

### Expected Behavior
We can use the FW upgrade utility sequentially to update first one, then the other, camera, and have both cameras accept FW and reconnect successfully.

### Observed Behavior
The first camera updates, but the second ends up in Recovery Mode. (This has happened on several different devices with the same results.)

### Steps to Reproduce
1. Start with a Windows device with two working RealSense cameras attached.
2. Run `Intel.RealSense.FW.Update.exe -l` to list the two working, connected cameras.
3. Run `Intel.RealSense.FW.Update.exe -s <serial1> -f <FW file>` to flash FW on the first camera. **This succeeds.**
4. After the camera has come back online, run `Intel.RealSense.FW.Update.exe -s <serial2> -f <FW file>` to target the second camera. **This fails. Camera is in recovery mode.**
5. Run `Intel.RealSense.FW.Update.exe -r -f <FW file>` to attempt recovery. Camera will still not come back online
6. Repeat several times...camera will finally upgrade and come back online.

### Annotated Transcript of what we see
List cameras pre-upgrade:

```
PS C:\> .\Intel.RealSense.FW.Update.exe -l

connected devices:
1) Name: Intel RealSense D435, serial number: xxxxxxxxxxxxxxxx, update serial number: zzzzzzzzzzzzz, firmware version: 05.11.01.100, USB type: 3.2
2) Name: Intel RealSense D435, serial number: yyyyyyyyyyyyyyyy, update serial number: aaaaaaaaaaaaa, firmware version: 05.11.01.100, USB type: 3.2
```
Upgrade first camera, then list devices (camera took a while to come back up, but came up with new FW.):
```
S C:\CMCMigrationPackage> .\Intel.RealSense.FW.Update.exe -s xxxxxxxxxxxxxxxx -f .\Signed_Image_UVC_5_12_12_100.bin

search for device with serial number: xxxxxxxxxxxxxxxx

updating device:
Name: Intel RealSense D435, serial number: xxxxxxxxxxxxxxxx, update serial number: zzzzzzzzzzzzz, firmware version: 05.11.01.100, USB type: 3.2
 14/07 10:53:15,685 ERROR [2680] (types.h:315) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."

firmware update started

firmware update progress: 100[%]

firmware update done
PS C:\> .\Intel.RealSense.FW.Update.exe -l

connected devices:
1) Name: Intel RealSense D435, serial number: yyyyyyyyyyyyyyyy, update serial number: aaaaaaaaaaaaa, firmware version: 05.11.01.100, USB type: 3.2
PS C:\> .\Intel.RealSense.FW.Update.exe -l

connected devices:
1) Name: Intel RealSense D435, serial number: xxxxxxxxxxxxxxxx, update serial number: zzzzzzzzzzzzz, firmware version: 05.12.12.100, USB type: 3.2
2) Name: Intel RealSense D435, serial number: yyyyyyyyyyyyyyyy, update serial number: aaaaaaaaaaaaa, firmware version: 05.11.01.100, USB type: 3.2
```
Target second camera, gets stuck in Recovery mode:
```
PS C:\CMCMigrationPackage> .\Intel.RealSense.FW.Update.exe -s yyyyyyyyyyyyyyyy -f .\Signed_Image_UVC_5_12_12_100.bin

search for device with serial number: yyyyyyyyyyyyyyyy

updating device:
Name: Intel RealSense D435, serial number: yyyyyyyyyyyyyyyy, update serial number: aaaaaaaaaaaaa, firmware version: 05.11.01.100, USB type: 3.2
 14/07 10:54:56,721 ERROR [10092] (types.h:315) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."

firmware update started

firmware update progress: 100[%]

firmware update done
PS C:\CMCMigrationPackage> .\Intel.RealSense.FW.Update.exe -l

connected devices:
1) Name: Intel RealSense D435, serial number: xxxxxxxxxxxxxxxx, update serial number: zzzzzzzzzzzzz, firmware version: 05.12.12.100, USB type: 3.2
2) Name: Intel RealSense D4xx Recovery, serial number: unknown, update serial number: aaaaaaaaaaaaa, firmware version: unknown, USB type: unknown
```
Try several attempts to get it out of recovery mode, reports success but fails, then finally succeeds:
```
PS C:\> .\Intel.RealSense.FW.Update.exe -f .\Signed_Image_UVC_5_12_12_100.bin -r

update to FW: .\Signed_Image_UVC_5_12_12_100.bin

recovering device:
Name: Intel RealSense D4xx Recovery, serial number: unknown, update serial number: aaaaaaaaaaaaa, firmware version: unknown, USB type: unknown

firmware update started

firmware update progress: 100[%]

firmware update done

recovery done
PS C:\> .\Intel.RealSense.FW.Update.exe -l

connected devices:
1) Name: Intel RealSense D435, serial number: xxxxxxxxxxxxxxxx, update serial number: zzzzzzzzzzzzz, firmware version: 05.12.12.100, USB type: 3.2
2) Name: Intel RealSense D4xx Recovery, serial number: unknown, update serial number: aaaaaaaaaaaaa, firmware version: unknown, USB type: unknown
```
Repeat this a few times and finally it works:
```
PS C:\CMCMigrationPackage> .\Intel.RealSense.FW.Update.exe -f .\Signed_Image_UVC_5_12_12_100.bin -r

update to FW: .\Signed_Image_UVC_5_12_12_100.bin

recovering device:
Name: Intel RealSense D4xx Recovery, serial number: unknown, update serial number: aaaaaaaaaaaaa, firmware version: unknown, USB type: unknown

firmware update started

firmware update progress: 100[%]

firmware update done

recovery done
PS C:\CMCMigrationPackage> .\Intel.RealSense.FW.Update.exe -l

connected devices:
1) Name: Intel RealSense D435, serial number: xxxxxxxxxxxxxxxx, update serial number: zzzzzzzzzzzzz, firmware version: 05.12.12.100, USB type: 3.2
PS C:\CMCMigrationPackage> .\Intel.RealSense.FW.Update.exe -l

connected devices:
1) Name: Intel RealSense D435, serial number: xxxxxxxxxxxxxxxx, update serial number: zzzzzzzzzzzzz, firmware version: 05.12.12.100, USB type: 3.2
PS C:\> .\Intel.RealSense.FW.Update.exe -l

connected devices:
1) Name: Intel RealSense D435, serial number: xxxxxxxxxxxxxxxx, update serial number: zzzzzzzzzzzzz, firmware version: 05.12.12.100, USB type: 3.2
2) Name: Intel RealSense D435, serial number: yyyyyyyyyyyyyyyy, update serial number: aaaaaaaaaaaaa, firmware version: 05.12.12.100, USB type: 3.2
```

MartyG-RealSense, 2021-07-15:
Thanks very much @etorbenson for your highly detailed report!

For future readers of this case, the support discussion for it can be found at https://github.com/IntelRealSense/librealsense/issues/7095#issuecomment-879995800

MartyG-RealSense, 2021-07-21:
Hi @etorbenson  Do you require further assistance with anything covered in this particular case, please?  Thanks!

etorbenson, 2021-07-21:
Thanks for following up. I've been testing this for a couple days now and here's what I found for future searchers:

- I found one component of our software solution was still running and holding one of two cameras open; part of the process was to kill the entire software stack and I missed one piece. This seems to have allowed the FW upgrade to complete successfully most of the time. Even with the software running, the cameras do accept firmware updates, however, the results appear unpredictable. 
- I also found that the FW upgrade succeeds more frequently when the camera you just upgraded is allowed to fully boot back up and reconnect, i.e. it shows up when you list the available RealSense cameras. So, in between upgrades, I give my script a short pause, then wait until both cameras are fully reporting back before starting the second upgrade.

Bottom line seems to be this - if anything is accessing the camera(s) during the update, the behaviour I'm describing can occur. I appreciate the effort on your end (hopefully) ruling out an issue with the tools; it pointed me back to my internal process.

closed 2021-07-21 by etorbenson

MartyG-RealSense, 2021-07-22:
Thanks again @etorbenson for the excellent documentation of your results  :)