realsenseai/librealsense #9413: Intel RealSense D415 - Troubleshoot when several other serial devices are connected to rpi4

gabrielcorreiab, 2021-07-15:
I have multiple devices communicating with rpi4 using serial communication. Everything was working fine, but when I connected the Intel RealSense D415 camera to the RPI4 all the sensors stopped working. As I understand it, the script to control the Intel RealSense D415 camera looks for all serial ports during the initialization step. This situation broke or reserved all the serial ports, causing my sensors to stop working. 

Connected to the rpi serial ports I have:
- 1 sonar sensor
- 1 telemetry device
- 1 Pixhawk device
- 1 Intel RealSense camera D415 (connected to usb3 port)

Running the script that only communicates with Sonar, Telemetry and Pixhawk everything works fine. When I run the script that only communicates with the camera everything works fine. If I put the camera code into the first script, everything crashe because the camera code must be trying to interact with all the serial ports, making all the communications shut-down. As I understood the camera code does not specify a serial port. He runs all the serial ports looking for the camera. How can I solve this? I still think it does this all the time, because I tried to put the camera on serial port 0 and its initialization code right at the start, to ensure that after the first scan of all serial ports, I could start communicating with the others sensors, but that didn't work either.

The code that make everything crash:
self.pipeline = rs.pipeline()
self.pipeline.start()

Debug already done:
- If I only use the other sensors that communicate using serial protocol with rpi4, the sensors work fine.
- If I only use the camera with rpi4, the camera works fine
- If I connect everything to rpi4 (camera + sensors) but only run the script that communicates with the sensors, the sensors work fine.
The problem is now:
-  If I connect everything to rpi4 (camera + sensors) but run the script that communicates with the sensors + camera, everything stops working.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | Open RealSense Viewer - 05.12.10 |  
| Operating System & Version |   Linux - raspbian 10 (buster)  | 
| Kernel Version (Linux Only)    | 5.10.11-v7l+                                         | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                           |  SDK 2.0 (v2.48.0)                        | 
| Language                                | python | 
| Segment			         | Robot | 

### Issue Description
Intel RealSense D415 - Troubleshoot when several other serial devices are connected to rpi4

MartyG-RealSense, 2021-07-15:
This is a continuation of a case first posted on the Intel Support RealSense forum.

https://support.intelrealsense.com/hc/en-us/community/posts/4403601913619-Intel-RealSense-D415-Troubleshoot-when-several-other-serial-devices-are-connected-to-rpi4

***********

Hi @gabrielcorreiab   Let's continue your case here on the RealSense GitHub.  In the most recently posted comment of the above case, i provided links for researching the setting of static serial ports for each device.  

https://www.raspberrypi.org/forums/viewtopic.php?t=300068
https://www.freva.com/2019/06/20/assign-fixed-usb-port-names-to-your-raspberry-pi/

Was that information helpful to you, please?

MartyG-RealSense, 2021-07-22:
Hi @gabrielcorreiab  Do you require further assistance with this case, please?  Thanks!

gabrielcorreiab, 2021-07-23:
> Hi @gabrielcorreiab Do you require further assistance with this case, please? Thanks!

Unfortunately still not discover and solve the problem.

Also, we analyzed the linux system tasks that are running when we run the camera script to try to understand if there are any tasks in the linux system running that occupy/block all the serial ports but we didn't find anything.

However we tested it with another camera from another manufacturer and managed to run the script together with the other sensors and maintain the serial communication with all of them.

MartyG-RealSense, 2021-07-23:
So in summary, the fact that other cameras worked with all of your sensors confirmed your theory that the RealSense camera specifically was disruptive to the serial communications when plugged in?

gabrielcorreiab, 2021-07-24:
Yes, the fact that we have tested it with another camera leads me to conclude that the  Intel RealSense D415 camera is interfering with the serial communication with the other sensors/devices connected to the remaining serial ports of the Raspberry pi. Any suggestions to overcome this problem?

gabrielcorreiab, 2021-07-24:
> So in summary, the fact that other cameras worked with all of your sensors confirmed your theory that the RealSense camera specifically was disruptive to the serial communications when plugged in?

Yes, the fact that we have tested it with another camera leads me to conclude that the Intel RealSense D415 camera is interfering with the serial communication with the other sensors/devices connected to the remaining serial ports of the Raspberry pi. Any suggestions to overcome this problem?

MartyG-RealSense, 2021-07-24:
I went through the case again carefully from the start and looked at it from all angles.  I am not certain it is related to the serial ports being crashed by the RealSense camera.  I note that you stated that the problem occurs when the pipeline start code is called in the script:

```
self.pipeline = rs.pipeline()
self.pipeline.start()
```

When the camera is inserted into the USB port but its streams are not enabled, it should only be using a small amount of power.  The power draw from the camera would ramp up when the pipeline is started and the streams are enabled.  Because you are not specifying a custom stream configuration in your pipeline setup, the D415 would enable the set of streams of its default configuration, which on the D415 model would be 1280x720 depth and 1280x720 RGB, both at 30 FPS.  

If your Pi 4 does not have sufficient power available to cover the needs of that stream configuration then the camera may disconnect, and perhaps set off a chain of failure in the script.

You could test this theory by specifying a humble custom configuration that initiates a depth stream on;y at 640x480 and 30 FPS.

```
self.pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
profile = self.pipeline.start(config)
```

closed 2021-08-05 by MartyG-RealSense