realsenseai/librealsense #9417: "RuntimeError: Frame didn't arrive within 5000" with D415 and D435

PaulaScharf, 2021-07-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 and D435 | 
| Firmware Version                   | 05.12.14.50 and 5.12.12.100| 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC (Optiplex 7080 with Intel I5)  |
| SDK Version                           |  2.48.0.3381                         | 
| Language                            |  Python v3.7                       | 
| Segment			|  ~AR                       | 

### Issue Description
When retrieving the color stream of our D415 and D435 camera we often get the error:

``` RuntimeError: Frame didn't arrive within 5000 ```

This usually happens when we start our system and do the first request of the color frame but sometimes also after the system was already running for a while. The error seemingly shows up randomly. It not just happens in our python code but also from time to time in the Realsense Viewer (the video visibly lags and sometimes shows an error message that the frame wasn't received). We have been running into this issue for half a year now, so this was already the case for older firmware and SDK versions.

We have tried unplugging our camera or doing a hardware reset in our python code, but it usually does not solve the issue. Changing the USB cable also doesn't do anything. Sometimes it does however help to restart our PC. We usually just take a break when it happens and after a while it will often work again. It is really annoying though.

We are aware of other issues with this error (#9270, #8691, #5181, #2168, #6628, and #927) but none of the proposed solutions have worked for us.

MartyG-RealSense, 2021-07-15:
Hi @PaulaScharf  Are you using the cameras one at a time or both of them at the same time, please?

Also, if you have auto-exposure enabled then does your program's start-up reliability improve if you skip the first several frames with a **for** instruction such as the one below?

```
for index in range(5):
pipeline.wait_for_frames()
```

PaulaScharf, 2021-07-15:
Thank you for the reply :)

We are just using one at a time depending of our setup. These days we are mostly using D415.

I will try skipping the first few frames. Can I leave this issue open or should I reopen it in case the error happens again?

MartyG-RealSense, 2021-07-15:
It is fine to leave the issue open.  Generally, an issue would be closed if you close it yourself (or ask us to close it) or we do not receive a comment from you for a couple of weeks.

PaulaScharf, 2021-07-16:
The error has happened again even though we are now skipping the first few frames. I also tried turning the auto-exposure on and off but that doesn't help. What really worries me is that the error is also visible in the Realsense Viewer, so I don't think it is an issue of our python code.

PaulaScharf, 2021-07-16:
In the Realsense Viewer we also currently get an error like this:
```
Notification from Hardware
  Received the following Notification:
    Timestamp: 1626442553117.083008
    Severity: Error
    Description: USB CAM overflow
```

MartyG-RealSense, 2021-07-16:
What resolution and FPS speed are you using, please?  **USB CAM Overflow** suggests to me that the USB port may be having difficulty with handling the incoming data in a timely manner and the SDK's 'frame queue' is getting jammed up with frames instead of being able to process new frames as they arrive.

Also, are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice, please?

MartyG-RealSense, 2021-07-22:
Hi @PaulaScharf  Do you require further assistance with this case, please?  Thanks!

PaulaScharf, 2021-07-22:
Hi @MartyG-RealSense, yes, sorry, we are only in the lab once or twice a week to work with the cameras.

We have tried several different USB-cables now:
- both of our official 1m USB cables that came with the camera
- two high end 3m cables (USB3.2 Gen 1, transfer rate of up to 5Gbps, max current of 3A)
- a random 1m cable that was lying around (dont know the specs)

All of these have occasionally produced the error 5000. What kind of specs for the usb cable do you think are necessary to work with the cameras?

Also, we noticed that the cameras get very hot when using them. Maybe heat is an issue?

MartyG-RealSense, 2021-07-22:
Information about USB cable specification can be found on page 85 of the current edition of the data sheet document for the 400 Series cameras.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

The official 1 meter cable supplied with the camera has been validated by Intel for use with RealSense cameras and so should provide a stronger likelihood of good performance than a cable of your own choice.  'Industrial grade' high quality cables such as those provided by the company Newnex can work well with the cameras too though.

https://www.newnex.com/realsense-3d-camera-connectivity.php

If the camera's casing becomes hot to the touch after only a few minutes of the camera being enabled then this could indicate a bad USB cable or a glitch on the USB port of the computer / computing device.  If the internal ASIC and Projector temperatures of the camera rise above 42 degrees C then problems with a stream are more likely to emerge.

PaulaScharf, 2021-07-22:
Ok, thank you, I guess it has to be a problem with our PC then. We also tried a few times with another notebook (I dont know the specs) and we also had this error. I will try the camera with my own PC at home later and report back to you.

MartyG-RealSense, 2021-07-22:
Thanks very much @PaulaScharf - I look forward to your next test results.  Good luck!

MartyG-RealSense, 2021-08-02:
Hi @PaulaScharf  Do you have an update about this case that you can provide, please?  Thanks!

PaulaScharf, 2021-08-02:
Hi, I have tested it a few times for longer periods on my home PC now and have so far not encountered the error. I have tried both our Python code and the realsense viewer. I will try a few more times, but I guess it has to be a problem of the work PC and Notebook then.

What I did notice though was heat: After a few minutes of running our code the camera gets really hot (I dont have a thermostat but its so hot that I can not touch it for more than a second). It also gets hot from running the realsense viewer but I would say its maybe a bit less than when running our code.

PaulaScharf, 2021-08-07:
Thank you for your support so far. We are thinking that it might be an issue with the motherboard or the installed drivers (The motherboard we are using is a Dell E93839 in case someone else also runs into this error). Maybe we have to get in touch with the manufacturers of the PCs. I will close the issue for now.

closed 2021-08-07 by PaulaScharf

MartyG-RealSense, 2021-08-07:
Thanks very much @PaulaScharf for the update - good luck, and please do feel free to re-open this case if you require further advice about it.