realsenseai/librealsense #9419: The window could not be refreshed properly on Jetson AGX

Ligcox, 2021-07-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |05.12.13.50| 
| Operating System & Version |   Ubuntu 18.04.5 LTS   | 
| Kernel Version (Linux Only)    |  4.9.201-tegra (buildbrain@mobile-u64-4551)                                       | 
| Platform                                 |  NVIDIA Jetson  |
| SDK Version                            |  2.25.0                          | 
| Language                            |  python                           | 
| Segment			|  Robot | 

### Issue Description
I tried to use D435 high speed mode(`https://dev.intelrealsense.com/docs/high-speed-capture-mode-of-intel-realsense-depth-camera-d435`) to create an application to identify the high speed moving object and display the recognition result on my Jetson AGX. The circles close to D435 cam will be red, and the circles far away will be white.

In my framework, which is a Python-based multithreaded program, RealSense works perfectly on a Windows platform. But in Jetson, RealSense seemed to work fine, but when I tried to convert the captured image into a colormap and display it through cv.imshow(), OpenCV didn't work . As I can see from the debuger, the interpreter is stuck at the cv2.imshow () function and is blocking several other threads (but not all). When I comment out the RealSense task code, the other functions work fine. I Make sure RealSense is working at Jetson, because I'm trying to save image array as a .npy file before reading and display the file separately. Files can display images as normal

Because of the complexity of my framework, I had to separate out the RealSense part and rewrite it, and it worked fine on Windows. But when I tried to put the task on Linux, I still had an unexpected problem: cv2.imshow created the window properly, but no image refresh (after a while the window was unresponsive, of course), but it didn't block the program. I show the average of the last 50 frames at the end part of the program. Even if the display is not normal, it will still refresh properly. You can focus on the RUN methods as follows.
```

import cv2
import time
import copy
import threading
import numpy as np

import platform
sysstr = platform.system()
if(sysstr =="Windows"):
    import pyrealsense2 as rs
elif(sysstr == "Linux"):
    import pyrealsense2.pyrealsense2 as rs

# The object sequence
# The format is [latest center point coordinates, time, [distance sequence element 1, distance sequence element 2, ````], distance latest value]
sequence = []

DEBUG_MODEL = True
class AutoDodge(object):
    def __init__(self, hide_controls=False):
        self.name = "auto_dodge"
        self.pipeline = rs.pipeline()
        self.config = rs.config()

        self.config.enable_stream(
            rs.stream.depth, 0, 848, 100, rs.format.z16, 300)
        self.pipeline.start(self.config)

        self.stop_flag = False
        self.realsense_task = threading.Thread(target=self.run, name="realsense_task")
        self.realsense_task.start()

    def update(self, p_centre, d):
        new_data = True
        n_time = time.time()
        for num, i in enumerate(sequence):
            if n_time - i[1] >= 0.5:
                sequence.pop(num)
                continue

            distance = pow(i[0][0]-p_centre[0], 2)+pow(i[0][1]-p_centre[1], 2)
            if distance < 10:
                i[0] = p_centre
                i[1] = n_time
                i[2].append(d)

                for j in range(1, len(i[2])):
                    i[3] += i[2][j] - i[2][j-1]

                i[3] /= len(i[2])
                new_data = False

                if i[3] <= 0:
                    return True

        if new_data:
            sequence.append([p_centre, n_time, [d], 0])

        return False

    def process(self, depth_image):
        depth_threshold = depth_image.mean()*0.95
        depth_image[depth_image > depth_threshold] = 0
        depth_image_copy = copy.deepcopy(depth_image)

        depth_image_copy[depth_image_copy != 0] = 2
        depth_binary = depth_image_copy.astype('uint8')
        _, depth_binary = cv2.threshold(depth_binary, 1,
                                        255, cv2.THRESH_BINARY)

        contours, _ = cv2.findContours(
            depth_binary, cv2.RETR_EXTERNAL, cv2.CHAIN_APPROX_SIMPLE)

        for cnt in contours:
            area = cv2.contourArea(cnt)
            perimeter = cv2.arcLength(cnt, True)
            if area <= 10:
                continue

            (x, y), radius = cv2.minEnclosingCircle(cnt)
            center = (int(x), int(y))
            radius = int(radius)

            if 2*3.14*pow(radius, 2)/area <= 3:
                left = np.min(cnt[:, :, 0])
                right = np.max(cnt[:, :, 0])
                top = np.min(cnt[:, :, 1])
                buttom = np.max(cnt[:, :, 1])
                _p = ((left, top), (right, buttom))

                roi = depth_image_copy[top:buttom+1, left:right+1]

                p_centre, stddv = cv2.meanStdDev(cnt)
                _mean, _stddv = cv2.meanStdDev(roi[roi != 0])

                if DEBUG_MODEL:
                    depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(
                        depth_image, alpha=0.085), cv2.COLORMAP_JET)
                    img = copy.deepcopy(depth_colormap)
                    img = cv2.circle(
                    img, center, radius, (0, 0, 255), 2)
                    if self.update((p_centre[0][0], p_centre[1][0]), _mean[0][0]):
                        img = cv2.rectangle(img, (left, top), (right, buttom), (0,0,255),thickness=5)
                    else:
                        img = cv2.rectangle(img, (left, top), (right, buttom), (255,255,255),thickness=5)

                    final = np.concatenate((depth_colormap, cv2.cvtColor(depth_binary, cv2.COLOR_GRAY2BGR), img), axis=0)
                    cv2.imshow("realsense", final)
            cv2.waitKey(1)

    def run(self):
        idx = 0
        s_t = time.time()
        while not self.stop_flag:
            frames = self.pipeline.wait_for_frames()
            depth_frame = frames.get_depth_frame()
            if not depth_frame:
                continue
            depth_image = np.asanyarray(depth_frame.get_data())
            
            self.process(depth_image)
            idx += 1

            if idx % 50 == 0:
                print(1/((time.time()-s_t)/50))
                s_t = time.time()

    def stop(self):
        self.stop_flag = True
        self.realsense_task.join()

if __name__ == "__main__":
    a = AutoDodge()
    a.run()

```

On Jetson, RealSenseViewer was able to open my camera normally and display a USB3.2 connection. I tried to rewrite the official demo and run it. The following program will run normally and refresh the window.
```

import cv2
import numpy as np

import platform
sysstr = platform.system()
if(sysstr == "Windows"):
    import pyrealsense2 as rs
# On the Linux platform, packages compiled from realsense source code may exist in the following
elif(sysstr == "Linux"):
    import pyrealsense2.pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 0, 848, 100, rs.format.z16, 300)
pipeline.start(config)

while True:
    frames = pipeline.wait_for_frames()
    depth_frame = frames.get_depth_frame()
    if not depth_frame:
        continue
    depth_image = np.asanyarray(depth_frame.get_data())
    depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(
        depth_image, alpha=0.085), cv2.COLORMAP_JET)

    cv2.imshow('RealSense', depth_colormap)
    cv2.waitKey(1)
cv2.destroyAllWindows()

```

I suspect it's due to  the pyrense2 package, I tried cmake source file through DPYTHON_EXECUTABLE, either -DPYTHON_EXECUTABLE=/usr/bin/python3.6 or -DPYTHON_EXECUTABLE=/usr/bin/python3 or -DPYTHON_EXECUTABLE=/usr/bin/python. or Could it be OpenCV package?
some package info:
numpy                         1.18.5
opencv-python                 4.5.1.48
pyrealsense2                 2.48.0

MartyG-RealSense, 2021-07-15:
Hi @Ligcox  If the problem was with DPYTHON_EXECUTABLE then it is unlikely that your test script based on the official demo would run successfully (unless you had the two .so files librealsense2.so and pyrealsense2.so placed in the same folder as your pyrealsense2 project).

If the demo script adaptation is able to display its graphics successfully then this would suggest that **cv2.imshow** is working okay on your Jetson, at least for that particular script.

The fact that your Python project works normally on Windows but not on Jetson is a complicating factor, as it would suggest that the cause is outside of the scripting.

Is the Windows version of your project also using OpenCV **4.5.1.48** please? 

Could you also confirm that your project continues creating frames even if the graphics are not being rendered?

Ligcox, 2021-07-15:
> Hi @Ligcox If the problem was with DPYTHON_EXECUTABLE then it is unlikely that your test script based on the official demo would run successfully (unless you had the two .so files librealsense2.so and pyrealsense2.so placed in the same folder as your pyrealsense2 project) as your Python 3.6 installation would be unable to be located.
> 
> If the demo script adaptation is able to display its graphics successfully then this would suggest that **cv2.imshow** is working okay on your Jetson, at least for that particular script.
> 
> The fact that your Python project works normally on Windows but not on Jetson is a complicating factor, as it would suggest that the cause is outside of the scripting.
> 
> Is the Windows version of your project also using OpenCV **4.5.1.48** please?
> 
> Could you also confirm that your project continues creating frames even if the graphics are not being rendered?

Thank you very much for seeing my question!

I can confirm that I use the same versions of OpenCV and NumPy on Windows and Jetson.

pyrealsense2 is installed in a folder, /usr/lib/python3/dist-packages/pyrealsense2, that contains six files:
pybackend2.cpython-36m-aarch64-linux-gnu.so
pybackend2.cpython-36m-aarch64-linux-gnu.so.2
pybackend2.cpython-36m-aarch64-linux-gnu.so.2.48.0
pyrealsense2.cpython-36m-aarch64-linux-gnu.so
pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.48
pyrealsense2.cpython-36m-aarch64-linux-gnu.so.2.48.0

One thing that has always puzzled me is that I have to pass `impoer pyrealsense2.realsense2 as rs` to use pyrealsense2 on Jetson. I wonder if it was a problem of .so file.

As I mentioned earlier, in my project (and in the Demo I mentioned above), if I don't use cv2.imshow(), I can get the correct frames. The reason for this is that I saved the frame as a .npy file before cv2.imshow(), and I have verified on both Windows and Linux, and the result is that there is no problem with the obtained frames. As I mentioned, by changing the variable DEBUG_MODEL from True to False, the program will run normally. However, no matter what the value of DEBUG_MODEL is, it does not affect the normal operation of the program (the FPS printing part at the end of the program can run normally).

MartyG-RealSense, 2021-07-15:
The link below suggests that if **waitKey(1)** is used (as your script does) then only 1 frame is shown and then the display automatically closes, whilst if waitKey(0) is used then the display is visible infinitely until a key is pressed.

https://stackoverflow.com/questions/51143458/difference-in-output-with-waitkey0-and-waitkey1/51143586

Ligcox, 2021-07-15:
> The link below suggests that if **waitKey(1)** is used (as your script does) then only 1 frame is shown and then the display automatically closes and refreshes to the next frame, whilst if waitKey(0) is used then the display is visible infinitely in a 'pause' state until a key is pressed.
> 
> https://stackoverflow.com/questions/51143458/difference-in-output-with-waitkey0-and-waitkey1/51143586

As you said, the parameters of the cv2.inshow function are like this, but my current work requires me to determine several key parameters by looking at successive frames, so I have to refresh the window constantly, but the window freezes When the program is running.

MartyG-RealSense, 2021-07-15:
The next test that I would recommend performing is that if you are drawing shapes on top of the image with your circle and rectangle code then comment out that code and just draw the raw image to test whether the drawing functions are causing disruption.

[image]

Ligcox, 2021-07-15:
> The next test that I would recommend performing is that if you are drawing shapes on top of the image with your circle and rectangle code then comment out that code and just draw the raw image to test whether the drawing functions are causing disruption.
> 
> [image]

I went ahead and simplified the demo above, removing all the image processing, and now the program should look pretty simple.What the program is doing right now is opening a separate thread to read the image and convert to colormap and make three copies to display.

The result of running on Windows is that the window refreshes normally, but OpenCV creates two Windows. One window is all black, it doesn't look like it's refreshing anymore; another normal display image.

The result of Jetson's run is that the window is still frozen after creation.

The modified procedure is as follows:
```

import cv2
import time
import copy
import threading
import numpy as np

import platform
sysstr = platform.system()
if(sysstr =="Windows"):
    import pyrealsense2 as rs
elif(sysstr == "Linux"):
    import pyrealsense2.pyrealsense2 as rs

DEBUG_MODEL = True
class AutoDodge(object):
    def __init__(self, hide_controls=False):
        self.name = "auto_dodge"
        self.pipeline = rs.pipeline()
        self.config = rs.config()

        self.config.enable_stream(
            rs.stream.depth, 0, 848, 100, rs.format.z16, 300)
        self.pipeline.start(self.config)

        self.stop_flag = False
        self.realsense_task = threading.Thread(target=self.run, name="realsense_task")
        self.realsense_task.start()

    def process(self, depth_image):
        depth_threshold = depth_image.mean()*0.95
        depth_image[depth_image > depth_threshold] = 0
        depth_image_copy = copy.deepcopy(depth_image)

        if DEBUG_MODEL:
            depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(
                depth_image, alpha=0.085), cv2.COLORMAP_JET)

            final = np.concatenate((depth_colormap, depth_colormap, depth_colormap), axis=0)
            cv2.imshow("realsense", final)

    def run(self):
        idx = 0
        s_t = time.time()
        while not self.stop_flag:
            frames = self.pipeline.wait_for_frames()
            depth_frame = frames.get_depth_frame()
            if not depth_frame:
                continue
            depth_image = np.asanyarray(depth_frame.get_data())
            
            self.process(depth_image)
            idx += 1

            # Record the average FPS for the last 50 frames
            if idx % 50 == 0:
                print(1/((time.time()-s_t)/50))
                s_t = time.time()

            key = cv2.waitKey(1)
            if key == ord("q"):
                self.stop()

            
    def stop(self):
        self.stop_flag = True
        self.realsense_task.join()

if __name__ == "__main__":
    task = AutoDodge()
    task.run()
```

MartyG-RealSense, 2021-07-16:
I am not experienced in Python multithread programming, so my knowledge of this subject is limited.  I researched the problem further though and located a pyrealsense2 case where a RealSense user separated into individual threads the tasks of retrieving RGB video and writing the video.  The recording was freezing up though.

An Intel team member at RealSense Support tested their code and experienced no problems with it, and suggested that the problem might be with the RealSense user's computer processing the frames too slowly.  So they tried the code on a different computer, and sometimes it worked and sometimes it failed.

The Intel support team member then suggested that the slowness may be due to inefficiencies in the Python Threading module, and the Multiprocessing module may have better performance.  The user confirmed that they achieved better performance with Multiprocessing instead of Threading.

The link to the complete case is here:

https://support.intelrealsense.com/hc/en-us/community/posts/360033158094-Recording-Video-in-60fps-with-threading-and-queue-Python-

And below is their original Python threading script before they switched to multiprocessing.

```
import pyrealsense2 as rs
import numpy as np
import datetime as dt
import time
import multiprocessing as mp
import os
#import mpio
import cv2
from queue import Queue
import threading as th

class RecordingJob(th.Thread):

    def __init__(self, video_name, queue):
        super(RecordingJob, self).__init__()
        self.stop_flag = th.Event()
        self.stop_flag.set()
        self.video_name = video_name
        self.queue = queue

    def run(self):
        self.num = 0
        self.fourcc = cv2.VideoWriter_fourcc(*'XVID')
        self.out = cv2.VideoWriter(self.video_name, self.fourcc, 60, (640, 480))
        while self.stop_flag.isSet():
            #print(self.queue.qsize())
            #self.num += 1
            if not self.queue.empty():
                self.out.write(self.queue.get())
                #self.queue.get()
                self.num += 1
        time.sleep(1)
        print(self.num)
        #print(self.queue.qsize())
        self.out.release()
        print('Thread Exitted')

    def stop(self):
        self.stop_flag.clear()

if __name__ == '__main__':

    directory_path = '0702test'
    if not os.path.exists(directory_path):
        os.mkdir(directory_path)

    rgb_video_name = directory_path+'/rgb.avi'
    f = open(directory_path+'/time.txt','w')
    #rgb_queue = mp.Queue()
    rgb_queue = Queue()

    #p = mpio.RGBRecordingClass(rgb_video_name, rgb_queue)
    #p.start()
    t = RecordingJob(rgb_video_name, rgb_queue)
    t.start()

    pipeline = rs.pipeline()
    config = rs.config()
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 60)

    profile = pipeline.start(config)
    frame_count = 0
    sensors = profile.get_device().query_sensors()

    #print('Sensors are '+str(sensors))

    for sensor in sensors:
        if sensor.supports(rs.option.auto_exposure_priority):
            #print('Start setting AE priority.')
            aep = sensor.get_option(rs.option.auto_exposure_priority)
            print('Original AEP = %d' %aep)
            aep = sensor.set_option(rs.option.auto_exposure_priority, 0)
            aep = sensor.get_option(rs.option.auto_exposure_priority)
            print('New AEP = %d' %aep)

    try:
        print('Start Recording %s' %(str(dt.datetime.now())))
        date_start = time.time()

        while True:
            frames = pipeline.wait_for_frames()

            color_frame = frames.get_color_frame()
            f.write(str(time.time()*1000)+'\n')
            if not color_frame:
                print('No color frame')

            color_image = color_frame.get_data()
            color_image = np.asanyarray(color_image)
            '''
            cv2.imshow('frame', color_image)
            cv2.waitKey(1)
            '''
            
            rgb_queue.put(color_image)

            frame_count += 1

            total_elapsed = time.time()-date_start
            print('Frame num: %d, fps: %d' %(frame_count, frame_count/total_elapsed))

            if frame_count >= 18000:

                break

    except Exception as e:
        print('Error is ', str(e))
    finally:
        print(str(dt.datetime.now()))
        pipeline.stop()
        print('frame_count=',str(frame_count))
        #p.stop()
        t.stop()
        #cv2.destroyAllWindows()
        f.close()
        print('end')
```

Ligcox, 2021-07-16:
@MartyG-RealSense  Thank you very much for your advice!

Because I need to share a lot of data between different threads, if I implement parallelism through a process, the changes may require more work. I've tried to get around this problem for the time being by calling class attribute, although this can cause a bit of trouble later.

However, I still don't think this is due to performance reasons, since the program's refresh window on the main thread is working fine. Thanks again for your advice and I will try to fix my program through it.

MartyG-RealSense, 2021-07-16:
You are very welcome @Ligcox - good luck!

closed 2021-07-28 by MartyG-RealSense