realsenseai/librealsense #9421: Stuck when calibrating IMU

aaumar, 2021-07-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Ubuntu 20.03  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | Jetson Nano  |
| SDK Version                            |  2.47.0                       | 
| Language                            |  Python                          | 
| Segment			|  Robot                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am trying to calibrate my imu using calibration tool provided by the SDK. But, I always stuck in this condition

> waiting for realsense device...
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  939622074541
  Product Line:  D400
  Firmware version:  05.12.05.00
Start interactive mode:
FOUND ACCEL with fps=63
FOUND GYRO with fps=200
*** Press ESC to Quit ***
Align to direction:  [ 0. -1.  0.]   Mounting screw pointing down, device facing out
   Status.rotate:           [-0.0215 -0.0003 -0.0089]:                 [ True  True  True]

MartyG-RealSense, 2021-07-15:
Hi @aaumar  This guidance is asking for the camera to be put on a flat surface with the bottom of the camera resting on the surface and the front of the camera facing straight ahead.  You then rotate the camera with your hands until the three values of **Status.rotate** match as closely as possible to the three values of **Align to direction**.  

When the camera orientation is detected to be a match to the **Align to direction**, a counter that is represented as a lengthening line of dots should begin.  

When the line of dots successfully reaches a certain length, the calibration tool should automatically proceed to the next stage and ask you to orient the camera to a different **Align to direction** position.  This process is repeated until all orientations have been successfully completed. 

Are you able to get as far as the orientation being matched and the line of dots starting to increase, please?

aaumar, 2021-07-16:
Hi @MartyG-RealSense, it seems the calibration tool is fine after I restart my jetson. 

But, there is another problem coming after finished doing 6 calibration poses (just after confirming to write the results to the camera). The error message is

> Error: failed to set power state
Segmentation fault (core dumped)

MartyG-RealSense, 2021-07-16:
I have seen a small number of past cases of this error when performing IMU calibration.  The majority of them were able to complete calibration and receive a 'Done' confirmation shown in the message below (indicating that the calibration had been written to the camera) before the segmentation fault occurred.  

[image]

One Ubuntu user who just received the same confirmation-less message that you did changed to a different Ubuntu machine and received the version above (Done, and then the segmentation fault).

The IMU of the camera is calibrated in the factory during manufacturing though, so it may not be a serious problem if you cannot get to the 'Done' state on your particular computer.

closed 2021-07-22 by aaumar