realsenseai/librealsense #9422: MACOSX_RPATH is not specified for the following targets: realsense2-gl

nabulsi, 2021-07-16:
### Issue Description
Camera Model D455
Mac Big Sur 11.4

I am following the installation steps as described in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

I am getting this warning when I run cmake:
`cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false`

Warning:

> CMake Warning (dev):
>   Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
>   --help-policy CMP0042" for policy details.  Use the cmake_policy command to
>   set the policy and suppress this warning.
> 
>   MACOSX_RPATH is not specified for the following targets:
> 
>    realsense2-gl

I tried adding `-DBUILD_WITH_TM2=false` as well as `-DCMAKE_BUILD_TYPE=Release`, but both didn't help

Thanks

MartyG-RealSense, 2021-07-16:
Hi @nabulsi  The link below features a recent case where the **MACOSX_RPATH is not specified for the following targets: realsense2-gl** error occurred when building on a Mac with the Apple Silicon M1 processor.  The RealSense user in that case found that adding **set(CMAKE_MACOSX_RPATH 1)** to the CMakeLists.txt file resolved the error.

https://github.com/IntelRealSense/librealsense/issues/9319

nabulsi, 2021-07-16:
Hi @MartyG-RealSense Thanks for the response. However, my laptop doesn't have an M1 processor. It's just intel i7.

nabulsi, 2021-07-16:
I just tried to use the pre-release version (i.e. 2.48.0), and I added this to the CMakeList.txt `set(CMAKE_MACOSX_RPATH 1)`
I didn't get the error this time after running cmake, but after running `make -j2`, I got this error:
```
Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY
  OPENSSL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.21.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.21.0/share/cmake/Modules/FindOpenSSL.cmake:574 (find_package_handle_standard_args)
  CMakeLists.txt:387 (find_package)
```
And the make this fails at 48%. Why is it complaining about openssl although I have configured it when running cmake?

MartyG-RealSense, 2021-07-16:
The issue of the **Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR** error is addressed in comments in the MacOS installation case https://github.com/IntelRealSense/librealsense/issues/8090

https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-836892782
https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-844905403

nabulsi, 2021-07-17:
Thanks Marty. I was able to bypass the errors & warnings so far by following what you referenced above. I am now able to build and compile successfully. However, when I run the realsense-viewer tool, I get this error all the time and the image I see in the tool doesn't seem to be clean. It has some duplicates for example. Is that related to what I was seeing before?
`WARNING [0x7000098cb000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60`

MartyG-RealSense, 2021-07-17:
A continuously generating **control_transfer returned** warning can indicate that there is a serious communication problem with the device (as the **operation timed out** information would suggest).  Do you still receive the messages if you go to the **More** option near the top of the RealSense Viewer's options side-panel and select the **Reset Hardware** option from its drop-down menu?

nabulsi, 2021-07-17:
I tried resetting but the issue is still there. This is what I was getting before hitting the reset HW button:
```
17/07 16:41:09,407 WARNING [0x700005383000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60
 17/07 16:41:09,558 WARNING [0x700005383000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60
 17/07 16:41:09,816 WARNING [0x700005383000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60
 17/07 16:41:10,176 WARNING [0x700005383000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60
....
```

Now I hit reset and the tool crashes:
 ```
17/07 16:41:15,365 ERROR [0x700005383000] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 17/07 16:41:15,365 ERROR [0x70000558f000] (sensor.cpp:512) acquire_power failed: failed to set power state
 17/07 16:41:15,365 ERROR [0x70000558f000] (error-handling.cpp:78) Error during polling error handler: failed to set power state
 17/07 16:41:16,369 ERROR [0x700005383000] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 17/07 16:41:16,369 ERROR [0x70000558f000] (sensor.cpp:512) acquire_power failed: failed to set power state
 17/07 16:41:16,369 ERROR [0x70000558f000] (error-handling.cpp:78) Error during polling error handler: failed to set power state
 17/07 16:41:17,370 ERROR [0x700005383000] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 17/07 16:41:17,370 ERROR [0x70000558f000] (sensor.cpp:512) acquire_power failed: failed to set power state
 17/07 16:41:17,370 ERROR [0x70000558f000] (error-handling.cpp:78) Error during polling error handler: failed to set power state
[1]    39772 illegal hardware instruction  ./realsense-viewer
```

After restarting the viewer, I get the same timeout errors

MartyG-RealSense, 2021-07-17:
And you still experience **control_transfer returned** errors if you unplug the micro end of the USB cable from the camera, turn the connector around the opposite way and re-insert it?  (USB-C cables are two-way insertion on the camera end).

nabulsi, 2021-07-17:
I get the same errors. Nothing changes.
FYI, my laptop has only USB-C. So, I am either using a convertor (original from Apple) and the cable that came with the camera, or a cable with USB-C on both ends

MartyG-RealSense, 2021-07-17:
Cables with USB-C on both ends can work with RealSense cameras.  Type-C cables can have quality issues though when using a cable of your own choice, and cable quality is important with RealSense because of the volume of data bandwidth that is transmitted through it.  A cable with quality issues can cause performance degradation or camera disconnections.  So there may be an issue in the C-to-C cable, the convertor or both.

You could try selecting depth only in the Viewer with a low resolution of 640x480 and see how that impacts the number of warnings generated.

nabulsi, 2021-07-17:
I am using the same cable on another Windows laptop and it works perfectly. I just tried with another high quality cable (from Dell) as well as with a lower resolution as you suggested but nothing changed.

MartyG-RealSense, 2021-07-17:
What firmware version is your D455 camera using, please?  You can find this information by clicking on the **Info** option in the Viewer side-panel.

nabulsi, 2021-07-17:
Thanks for your continuous support. 
Firmware version is 05.12.14.50

MartyG-RealSense, 2021-07-17:
You are very welcome!

As you built the tools and examples, are you able to run the **rs-capture** example program to see how the range of multiple stream types that it simultanously displays performs in comparison to the Viewer?

nabulsi, 2021-07-17:
this is what I get when I run the example:
```
./rs-capture
 17/07 18:58:02,124 ERROR [0x1126a7e00] (handle-libusb.h:95) failed to claim usb interface: 5, error: RS2_USB_STATUS_ACCESS
 17/07 18:58:02,125 ERROR [0x1126a7e00] (handle-libusb.h:95) failed to claim usb interface: 5, error: RS2_USB_STATUS_ACCESS

#### RS Frame Rate Printer ####
Depth[0]: 15.5/30 [FPS] || Accel[0]: 20.6/63 [FPS] || Gyro[0]: 20.4/200 [FPS] || Color[0]: 20.6/30 [FPS] ||  17/07 18:59:47,806 ERROR [0x700009b44000] (ds5-options.cpp:88) Asic Temperature value is not valid!
```
Also, the depth view remains static (nothing changes) as opposed to the RGB view.

MartyG-RealSense, 2021-07-17:
I note that you are apparently not receiving error messages such as **Frames not received within 5000** that would indicate a prolonged time-out period on the connection.  It makes me wonder whether the camera connection is dropping but coming back within a 5-second period, thus enabling the stream to automatically continue without having to manually re-start the stream.

Do you have access to a USB-C mains electricity powered hub that the camera could be attached to in order to test whether it is a USB power stability issue that could be resolved with a powered hub instead of drawing power from the Mac's USB port?

nabulsi, 2021-07-17:
I have a USB-A powered hub (i.e. all ports are USB-A 3.2 ports). I connected the camera to it using the cable that is included with it from intel, and then plugged the USB-A output of the hub to an adapter and then to the USB-C port of the Mac: no difference in the viewer or capture behavior.

MartyG-RealSense, 2021-07-17:
Thanks very much.  Could you try the **rs-hello-realsense** example program next.  This is a program that outputs a depth measurement as a plain-text value with no OpenGL graphics, so this will provide indication of whether there is a graphics related glitch on your Mac.

nabulsi, 2021-07-17:
That works fine without errors:
```
./rs-hello-realsense
The camera is facing an object 1.135 meters away
```

MartyG-RealSense, 2021-07-17:
There was an unofficial MacOS installation guide that suggested installing GLFW3 (the OpenGL graphics that librealsense uses) with the command below:

**brew install homebrew/versions/glfw3**

The guide can be found here:

https://gist.github.com/svenevs/f00ed3898d2af6248921b63254aa8cc1

nabulsi, 2021-07-17:
`brew install homebrew/versions/glfw3`

`Error: homebrew/versions was deprecated. This tap is now empty and all its contents were either deleted or migrated`

MartyG-RealSense, 2021-07-17:
Okay, thanks for trying (I have not got a Mac at my location to test with).

Let's try **rs-pointcloud** next as that also uses OpenGL.  If you get poor performance / errors in that then it would lend weight to the theory that the problem is a graphics-related one.

nabulsi, 2021-07-17:
Running rs-pointcloud doesn't show any errors at all. But is it normal that parts of the image are black?
[image]

MartyG-RealSense, 2021-07-17:
That looks like a good pointcloud to me.  It is normal to have a rough black border in 3D point cloud images.  For comparison, here is an **rs-pointcloud** image from my Windows computer.

[image]

nabulsi, 2021-07-17:
Good to know that this is normal. Thanks for sharing.
Is there anything else we can investigate? I know it is taking a lot of your time.

MartyG-RealSense, 2021-07-17:
It's no trouble at all, I promise.  :)

Comparing the difference between rs-capture that had problems and rs-pointcloud that didn't, a difference between them is that both of them access depth and RGB, but rs-pointcloud does not make use of the IMU (Motion Module).   

Did you have the Motion Module enabled when using the RealSense Viewer?

nabulsi, 2021-07-17:
No. I mostly tested with stereo and rgb. Even one of them alone will cause errors to be thrown.

MartyG-RealSense, 2021-07-17:
So far, we have estalished that the camera can function normally on your Mac in certain programs (rs-hello-realsense and rs-pointcloud).  Are you planning to write your own application to use the camera with or do you need to use a pre-made tool such as the Viewer for your project?

nabulsi, 2021-07-17:
hello-realsense was throwing erros as well.
I am planning to write my own application, but I am afraid with the existence of these errors that I will probably get issues later.
I will need to go now and will resume tomorrow. Please let me know if you have any more ideas and I will reply tomorrow. Thanks!

MartyG-RealSense, 2021-07-17:
Okay, thanks again for your patient testing.  If you let me know tomorrow about the kind of application that you are planning to write then I may be able to point you to scripting references that you can test for stability on your Mac.  Thanks again!

nabulsi, 2021-07-19:
Hi @MartyG-RealSense I believe that the issue I am facing `(handle-libusb.h:95) failed to claim usb interface: 5, error: RS2_USB_STATUS_ACCESS` has to do with some setting in the operating system. It seems several people reported the same issue, with some in Linux and some in Mac, but for the cam 515:

- https://github.com/IntelRealSense/librealsense/issues/7302
- https://github.com/IntelRealSense/realsense-ros/issues/1408

For me, it's related to D455.
Today, I tried with another powered hub (supplied it with 90W) and then used the cable that came with the camera (USB-C and USB-A endings). Then connected the hub to the laptop via USB-C. Still the same problem happens.

MartyG-RealSense, 2021-07-19:
I researched the case again from the beginning to seek new insights.

-  The camera hardware is essentially able to function despite the errors, suggesting that a USB communication related issue at the camera end of the connection is possible (errors such as 'control_transfer returned' and 'failed to claim usb interface' would also suggest communication issues).

- It is not a depth-only problem, as you stated that the problem can occur with RGB on its own too.

- It does not seem to be a problem with the powered USB hub, since the same error occurs when connecting the camera directly to the Mac with a USB C to C cable.

- A common factor in both cabling approaches (direct and hub) is the USB-C port on the side of the camera and the USB-C port on the Mac.  Is there any history of issues with the Mac's USB-C port when connecting non-RealSense devices to it?

nabulsi, 2021-07-19:
Thanks for reviewing again all the inputs.
No issues with the HW. It's s fairly new laptop. I tried with different ports in the laptop as well.

MartyG-RealSense, 2021-07-20:
Basically, everything seems to be okay except that programs have device communication errors when they are running, as described in https://github.com/IntelRealSense/librealsense/issues/9422#issuecomment-881807310

These can be difficult to solve as they may not have an obvious source.  Sometimes they just go away if a clean reinstall of the SDK is done, without finding out what was causing the errors.

MartyG-RealSense, 2021-07-26:
Hi @nabulsi  Do you require further assistance with this case, please?  Thanks!

nabulsi, 2021-07-26:
Hi, the issue is still there, but I gave up. I am currently using another Windows laptop. So, please feel free to close this issue. I'll reopen if something new comes up. Thanks for your support.

MartyG-RealSense, 2021-07-27:
Ok, thanks very much for the update.   I'm sorry to hear that your Mac issue wasn't able to be resolved.  On your suggestion, I will close the issue.  Good luck!

closed 2021-07-27 by MartyG-RealSense