realsenseai/librealsense #9423: How to stop enable_to_record_file()?

hyebin-rdv, 2021-07-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |  ubuntu 18.04 | 
| Kernel Version (Linux Only)    |5.4.0-77-generic | 
| Platform                                 | PC |
| SDK Version                            | 2.45.0| 
| Language                            |c++ | 

### Issue Description
I'm saving files using the enable_to_record_file function. I want to stop recording when a certain command comes from here.
`if ( number_test == 1){
    m_cfg.enable_record_to_file("/home/hyebin/bagfiles/hyebin.bag");
    fprintf(stderr, "--------%d\n", number_test);
} else if ( number_test == 0) {
    m_pipe.stop();
    fprintf(stderr, "--------%d\n", number_test);
} `

If number_test is 0, I want to stop recording and restart recording when number_test is 1.

MartyG-RealSense, 2021-07-16:
Hi @hyebin-rdv   As you have already created code to listen for a key input of '0' or '1', what you are seeking to achieve should be straightforward to implement using the stop and start recording code in the SDK's C++ **rs-record-playback** example program.

https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback#record-and-playback

hyebin-rdv, 2021-07-16:
I have already seen that document. But it did not help me. 

>if ( number_test == 1){
            m_pipe.stop();
            m_cfg.enable_record_to_file("/home/hyebin/bagfiles/" + name + ".bag");
        } else if ( number_test == 0) {
            m_rs_devices.as<rs2::recorder>().pause();
            m_pipe.stop();
            m_cfg.disable_all_streams();
            auto m_pipe = std::make_shared<rs2::pipeline>(); 
        }

how to stop record..?

MartyG-RealSense, 2021-07-16:
Converting the code in the example to the style of your script above I believe that the recording stop instruction would look like this:

```
m_pipe.stop();  // Stop the pipeline that holds the file and the recorder
auto m_pipe = std::make_shared<rs2::pipeline>();
```

You should not need to use **m_cfg.disable_all_streams();** as stopping the pipeline should stop all active streams anyway.

hyebin-rdv, 2021-07-18:
Then is there a way to keep streaming and create bag recordings only?
Stream keeps coming in, but it starts recording, saves, and restarts.
Because if you I pipe.stop(), the existing stream will be cut off.

MartyG-RealSense, 2021-07-18:
If you have the option of using ROS to record a bag file then you could use the **-split** command of the **rosbag record** tool of ROS.  This allows you to define a condition for when the current bag file will stop recording and a new file will automatically be created to continue the recording without stopping.  A recording can be set to stop after a certain amount of time has passed or when the bag file recording reaches a certain file size.

http://wiki.ros.org/rosbag/Commandline#rosbag_record

[image]

hyebin-rdv, 2021-07-20:
Can I get a video from a bag file created using rosbag_record?

MartyG-RealSense, 2021-07-20:
If you needed to convert a bag file created with rosbag record to video then the **rosbag2video** tool may be helpful, though it is written in Python language rather than C++.

https://github.com/mlaiacker/rosbag2video

If you are currently publishing a ROS topic that uses RGB then the instruction **rosbag record --all** should capture all currently active topics into a rosbag.  Alternatively, you can specify a precise list of topics to record to a bag, as described in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1947#issuecomment-874755451

hyebin-rdv, 2021-07-20:
I got it. But I don't know the code to represent it like this.

> /camera/align_to_color/parameter_descriptions
/camera/align_to_color/parameter_updates
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_color/image_raw/compressed
/camera/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressed/parameter_updates
/camera/aligned_depth_to_color/image_raw/compressedDepth
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/camera/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/camera/color/camera_info
/camera/color/image_raw
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/extrinsics/depth_to_color
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates

MartyG-RealSense, 2021-07-20:
To create an instruction to record a list of topics with rosbag record, you just put **rosbag record** at the start and then the list of topics after it.

[image]

hyebin-rdv, 2021-07-20:
nono, I know how to record but my `rostopic list` doesn't show that.
https://github.com/IntelRealSense/realsense-ros 

If I use 'roslaunch realsense2_camerars_camera.launch' here, I can get the list I want.

I want the same list to come out of the code I've written now. How can I use that roslaunch code with my code?

MartyG-RealSense, 2021-07-20:
Okay, I think I understand now.  You want to record the list of topics below but you do not have those topics available when using the rs_camera.launch roslauch instruction.

**/rdvcam0/camera/rgb/image_mask
/rdvcam0/camera/rgb/image_raw
/rdvcam0/rdv_rcnn_topic/pc_all
/rdvcam0/rdv_rcnn_topic/plane_points
/rdvcam0/rdv_rcnn_topic/plane_points_color_mask
/rdvcam0/rdv_rcnn_topic/plane_points_color_mask_filter
/rosout
/rosout_agg**

Could you launch rs_camera in one terminal and the perform the launch that produces the above non-RealSense topic list in a separate ROS terminal?  And then use rosbag record in a third ROS terminal to record the above list to a bag with:

**rosbag record /rdvcam0/camera/rgb/image_mask /rdvcam0/camera/rgb/image_raw /rdvcam0/rdv_rcnn_topic/pc_all
/rdvcam0/rdv_rcnn_topic/plane_points /rdvcam0/rdv_rcnn_topic/plane_points_color_mask /rdvcam0/rdv_rcnn_topic/plane_points_color_mask_filter /rosout /rosout_agg**

hyebin-rdv, 2021-07-21:
Thanks. Then can I get a video, not an image, using the 'rosbag record'? The image is obtained by recording like that, but the video is not obtained. Is there a way to add topic as a video?

hyebin-rdv, 2021-07-21:
Or do you know how to do two functions with one camera? 
"The device is busy!" So that the message doesn't appear.

MartyG-RealSense, 2021-07-21:
At this point, let's pause and confirm what you are trying to achieve so that there is not confusion.

Are you hoping to publish topics in ROS and then read those topics with a C++ ROS node script?  Or are you only using **rosbag record** now.

hyebin-rdv, 2021-07-21:
Anything. 

My ultimate goal is to record videos whenever I want.

My program is already using the realsense camera to capture images. I want to record a video whenever I want regardless of the capture motion.  I want to record a video whenever I want regardless of the capture motion. 

The enable_record_to_file function requires pipe.stop() to stop and restart recording.
Topic saves images only when the topic is called. The same goes for rosbag.

This is what I want to ultimately get "video" from using topics.

If I want to get a video using topics, do I have to keep calling it into an infinite loop?

hyebin-rdv, 2021-07-22:
Ok, Thank you! It is so helpful!

closed 2021-07-22 by hyebin-rdv

MartyG-RealSense, 2021-07-22:
You are very welcome @hyebin-rdv - thanks for the update!