realsenseai/librealsense #9424: Undefined 'extern' symbol when linking SDK statically

MikhailUskin, 2021-07-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version |  Ubuntu 18.04 / GCC-9.4  | 
| Kernel Version (Linux Only)    |  4.9.201-tegra | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                           | 2.48.0 | 
| Language                            |  C++ | 
| Segment			|  VR/AR | 

### Issue Description
I'm compiling and linking SDK statically with shared object. When this shared object is loaded by executable, I see the following error: "dlerror system message: /home/mrs/repo/cv-core/build/bin/libcvcore.so: undefined symbol: _ZN14rs2rosinternal8TIME_MINE". Static libraries libfw.a, librealsense-file.a, librealsense2.a are linked to 'libcvcore.so'.

I tried linking SDK to 'libcvcore.so' with option '--no-as-needed' but it didn't help. Some undesirable workaround I found for this problem is declaring 'TIME_MIN' in time.h as 'static' instead of 'extern' to avoid implicit dependency of one static library (librealsense-file.a) from another (librealsense2.a) as I understand.

MartyG-RealSense, 2021-07-16:
Hi @MikhailUskin  There are few information resources available about building librealsense as a static library, and I have not previously encountered the error that you quoted.

It's good that you found a workaround.  If you would prefer to use a more 'correct' method though, the best reference that I know of for building librealsense statically by setting **BUILD_SHARED_LIBS** to False is at the link below.

https://github.com/IntelRealSense/librealsense/issues/4709

If the guidance in that link does not resolve your problem then I will be happy to continue working through the problem with you.

MikhailUskin, 2021-07-16:
Thanks, I saw issue [#4709](https://github.com/IntelRealSense/librealsense/issues/4709):
* __BUILD_SHARED_LIBS__ is set to False;
* Also I tried link the libraries in different order.

Nothing helped me

MikhailUskin, 2021-07-16:
Wrapping SDK linked libraries so that "--whole-archive <sdk_libraries> --no-whole-archive" solved the problem

MartyG-RealSense, 2021-07-16:
Excellent news @MikhailUskin that you were able to develop a solution - thanks so much for sharing the solution with the RealSense community!

closed 2021-07-23 by MartyG-RealSense