realsenseai/librealsense #9425: Openvino examples compilation error on Raspberry Pi

dgorissen, 2021-07-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 5.12.13 | 
| Operating System & Version |   Linux (Raspbian Buster)  | 
| Kernel Version (Linux Only)    |  5.10.17-v7l+                                         | 
| Platform                                 |  Raspberry Pi 4  |
| SDK Version                            |   v2.48.0                          | 
| Language                            |   C++ / cmake                           | 
| Segment			|  Robot                           | 

### Issue Description
Camera & regular examples are working fine. I have a working local install of openvino (pkg: l_openvino_toolkit_runtime_raspbian_p_2021.4.582) and wanted to compile the openvino examples.

I added the following flags:

```
 -DBUILD_OPENVINO_EXAMPLES=ON \
 -DINTEL_OPENVINO_DIR=/home/dgorissen/openvino \
 -DOpenCV_DIR=/home/dgorissen/openvino/opencv \
 -DCMAKE_CXX_FLAGS=-isystem\ /home/dgorissen/openvino/inference_engine \

```

I found I had to add the last line for else cmake could not find `inference_engine.hpp` but it should not be needed I would think.

If I then compile I get the following compilation error. Any thoughts?

```
...
Consolidate compiler generated dependencies of target pyrealsense2
[ 51%] Built target pyrealsense2
Consolidate compiler generated dependencies of target rs-face-vino
[ 52%] Building CXX object wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/rs-face-vino.cpp.o
In file included from /home/dgorissen/librealsense/wrappers/openvino/./rs-vino/base-detection.h:7,
                 from /home/dgorissen/librealsense/wrappers/openvino/./rs-vino/object-detection.h:6,
                 from /home/dgorissen/librealsense/wrappers/openvino/face/rs-face-vino.cpp:8:
/home/dgorissen/librealsense/wrappers/openvino/./rs-vino/openvino-helpers.h:189:5: error: expected class-name before ‘{’ token
  189 |     {
      |     ^
/home/dgorissen/librealsense/wrappers/openvino/./rs-vino/openvino-helpers.h:190:14: error: ‘void openvino_helpers::error_listener::onError(const char*)’ marked ‘override’, but does not override
  190 |         void onError( char const * msg ) noexcept override
      |              ^~~~~~~
/home/dgorissen/librealsense/wrappers/openvino/face/rs-face-vino.cpp: In function ‘int main(int, char**)’:
/home/dgorissen/librealsense/wrappers/openvino/face/rs-face-vino.cpp:41:12: error: ‘class InferenceEngine::Core’ has no member named ‘SetLogCallback’
   41 |     engine.SetLogCallback( error_listener );
      |            ^~~~~~~~~~~~~~
make[2]: *** [wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/build.make:76: wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/rs-face-vino.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1112: wrappers/openvino/face/CMakeFiles/rs-face-vino.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
(failed reverse-i-search)`': ^C librealsense/
```

Cmake 3.16.3. Compiler info:

```
dgorissen@cbrain:~/openvino $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 8.3.0-6+rpi1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
```

MartyG-RealSense, 2021-07-16:
Hi @dgorissen  There is a past RealSense case in which listener errors were occurred when trying to install OpenVINO on a device with an Arm processor (like the Raspberry Pi has).  

https://github.com/IntelRealSense/librealsense/issues/8266 

The OpenVINO system requirements specify an Intel processor.  In that case though, the RealSense user found that they could successfully build OpenVINO 2021 from source code on Arm.  But the RealSense OpenVINO examples are currently validated for OpenVINO 2019 and 2020.  So that user decided to use OpenVINO 2020.

dgorissen, 2021-07-16:
Thanks @MartyG-RealSense. ARM (or at least Raspberry Pi) does [seem to be a first class supported platform ](https://docs.openvinotoolkit.org/latest/openvino_docs_install_guides_installing_openvino_raspbian.html) with recent releases. Its not crucial for me at the moment so I will skip the openvino integration for now but probably good if in the future the realsense examples could be updated.

MartyG-RealSense, 2021-07-16:
Thanks very much @dgorissen - I will highlight your suggestion about OpenVINO 2021 support for examples to Intel.

MartyG-RealSense, 2021-07-23:
Hi @dgorissen  Bearing in mind that I have highlighted your suggestion about OpenVINO 2021 support for examples to Intel, do you require further assistance with this case please?  Thanks!

dgorissen, 2021-07-23:
No, thank you. Will keep an eye out for future releases.

MartyG-RealSense, 2021-07-23:
Okay, thanks very much for the update @dgorissen - I will close this case now.  Thanks again!

closed 2021-07-23 by MartyG-RealSense

dgorissen, 2021-08-12:
Thanks @MartyG-RealSense !