realsenseai/librealsense #9426: Python import error on M1 Mac Big Sur

aman3014, 2021-07-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430 | 
| Operating System & Version |   MacOS  | 
| SDK Version                            |  2.2.48  | 
| Language                            |  python                 | 

### Issue Description
I'm running an M1 Mac with Big Sur 11.4.

I followed the installation steps at https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk and added the `-DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl` flag to solve the issue described in #8090.

However, when trying to import pyrealsense2 in a python file (in a Conda environment which has access to the python binaries generated by Xcode), I get the following error:

`ImportError: dlopen(path/pyrealsense2/pyrealsense2.cpython-38-darwin.so, 2): no suitable image found.  Did find:
	path/pyrealsense2/pyrealsense2.cpython-38-darwin.so: mach-o, but wrong architecture
	/librealsense/build/wrappers/python/Debug/pyrealsense2.2.48.0.cpython-38-darwin.so: mach-o, but wrong architecture`

Thanks!

MartyG-RealSense, 2021-07-16:
Hi @aman3014   I would recommend starting with investigating the problem from the perspective of Conda, since it is rarely used with the Python wrapper and so may introduce issues that have not been encountered in other cases.

For example, the build method applied by one Conda user was to input the command **conda develop build/wrappers/python/**

https://github.com/IntelRealSense/librealsense/issues/6922#issuecomment-663976346

MartyG-RealSense, 2021-07-23:
Hi @aman3014  Do you require further assistance with this case, please?  Thanks!

aman3014, 2021-07-23:
Hey @MartyG-RealSense, sorry for my late response.

I tried conda develop command but I'm still facing the same issue, unfortunately.

MartyG-RealSense, 2021-07-23:
Googling for the term **mach-o, but wrong architecture** demonstrates that this error seems to be an issue with Mac M1 (due to the Arm architecture) that has been encountered on a range of different platforms, rather than a problems that is specific to RealSense.

My understanding of a successful case of building the Python wrapper for M1 in https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-844905403 was that the CMake flag **-DBUILD_PYTHON_BINDINGS:bool=true** was included in the CMake instruction to build librealsense, so that the SDK and te Python wrapper were built at the same time.  Is that the method that you used to build the Python wrapper please?

aman3014, 2021-07-26:
Yes, I did include that flag.

MartyG-RealSense, 2021-07-26:
Do you have the option to update to the modern version of Python for Mac that supports M1 and Big Sur?

https://www.techradar.com/news/you-can-now-run-python-on-apple-m1-macos

aman3014, 2021-07-27:
I tried using a native python 3.8.10 install (the universal2 installer here: https://www.python.org/downloads/release/python-3810/) without conda but I get the same error unfortunately. Then I followed the following steps: 

```
brew install libusb pkg-config
brew install homebrew/core/glfw3
brew install cmake
mkdir build && cd build
sudo xcode-select --reset
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl -DBUILD_PYTHON_BINDINGS:bool=true -G Xcode
```
- Build ALL in Xcode, then:
`ln -s /../librealsense2/build/wrappers/python/Debug/ python_site_packages_dir/pyrealsense2
`
- In python,
`from pyrealsense2 import pyrealsense2`

and I get the error mentioned in the first post.

When I tried with conda, I additionally did `conda develop build/wrappers/python/` but it still gave the same error at the end.

MartyG-RealSense, 2021-07-27:
I decided to take the approach of searching for the Apple M1 error **cpython-38-darwin.so: mach-o, but wrong architecture** without including the RealSense part of the error, since there are no previous RealSense related occurrences of this error.  One contributor suggested trying the following procedure:

1.  Go to the **Applications** folder and right-click on **Terminal App > Get Info**

2.  Tick the **Open with Rosetta** option.  This option enables a Mac with Apple Silicon to use apps built for a Mac with an Intel processor.

3.  Build your project.

aman3014, 2021-07-29:
I obtain the following error during the Xcode Build:

```
Ld /Users/aman/Code/librealsense/build/Debug/librealsense2.2.48.0.dylib normal (in target 'realsense2' from project 'librealsense2')
    cd /Users/aman/Code/librealsense
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -target x86_64-apple-macos11.3 -dynamiclib -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk -L/Users/aman/Code/librealsense/build/Debug -F/Users/aman/Code/librealsense/build/Debug -filelist /Users/aman/Code/librealsense/build/librealsense2.build/Debug/realsense2.build/Objects-normal/x86_64/realsense2.2.48.0.LinkFileList -install_name /librealsense2.2.48.0.dylib -Xlinker -object_path_lto -Xlinker /Users/aman/Code/librealsense/build/librealsense2.build/Debug/realsense2.build/Objects-normal/x86_64/realsense2.2.48.0_lto.o -Xlinker -no_deduplicate -dynamiclib -Wl,-headerpad_max_install_names -install_name @rpath/librealsense2.2.48.dylib -framework CoreFoundation -framework IOKit /Users/aman/Code/librealsense/build/common/fw/Debug/libfw.a /Users/aman/Code/librealsense/build/third-party/realsense-file/Debug/librealsense-file.a /Users/aman/Code/librealsense/build/libusb_install/lib/libusb.a -lobjc -framework CoreFoundation -framework IOKit -Xlinker -no_adhoc_codesign -compatibility_version 2.48.0 -current_version 2.48.0 -Xlinker -dependency_info -Xlinker /Users/aman/Code/librealsense/build/librealsense2.build/Debug/realsense2.build/Objects-normal/x86_64/realsense2.2.48.0_dependency_info.dat -o /Users/aman/Code/librealsense/build/Debug/librealsense2.2.48.0.dylib

ld: warning: ignoring file /Users/aman/Code/librealsense/build/libusb_install/lib/libusb.a, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
Undefined symbols for architecture x86_64:
  "_libusb_alloc_transfer", referenced from:
      librealsense::platform::usb_request_libusb::usb_request_libusb(libusb_device_handle*, std::__1::shared_ptr<librealsense::platform::usb_endpoint>) in request-libusb.o
  "_libusb_bulk_transfer", referenced from:
      librealsense::platform::usb_messenger_libusb::bulk_transfer(std::__1::shared_ptr<librealsense::platform::usb_endpoint> const&, unsigned char*, unsigned int, unsigned int&, unsigned int) in messenger-libusb.o
  "_libusb_cancel_transfer", referenced from:
      librealsense::platform::usb_messenger_libusb::cancel_request(std::__1::shared_ptr<librealsense::platform::usb_request> const&) in messenger-libusb.o
      librealsense::platform::usb_request_libusb::~usb_request_libusb() in request-libusb.o
  "_libusb_claim_interface", referenced from:
      librealsense::platform::handle_libusb::claim_interface(unsigned char) in device-libusb.o
  "_libusb_clear_halt", referenced from:
      librealsense::platform::usb_messenger_libusb::reset_endpoint(std::__1::shared_ptr<librealsense::platform::usb_endpoint> const&, unsigned int) in messenger-libusb.o
  "_libusb_close", referenced from:
      librealsense::platform::handle_libusb::~handle_libusb() in device-libusb.o
  "_libusb_control_transfer", referenced from:
      librealsense::platform::usb_messenger_libusb::control_transfer(int, int, int, int, unsigned char*, unsigned int, unsigned int&, unsigned int) in messenger-libusb.o
  "_libusb_detach_kernel_driver", referenced from:
      librealsense::platform::handle_libusb::claim_interface(unsigned char) in device-libusb.o
  "_libusb_exit", referenced from:
      librealsense::platform::usb_context::~usb_context() in context-libusb.o
  "_libusb_free_config_descriptor", referenced from:
      librealsense::platform::usb_device_libusb::usb_device_libusb(libusb_device*, libusb_device_descriptor const&, librealsense::platform::usb_device_info const&, std::__1::shared_ptr<librealsense::platform::usb_context>) in device-libusb.o
      librealsense::platform::get_subdevices(libusb_device*, libusb_device_descriptor) in enumerator-libusb.o
  "_libusb_free_device_list", referenced from:
      librealsense::platform::usb_context::~usb_context() in context-libusb.o
  "_libusb_free_transfer", referenced from:
      librealsense::platform::usb_request_libusb::usb_request_libusb(libusb_device_handle*, std::__1::shared_ptr<librealsense::platform::usb_endpoint>)::$_0::operator()(libusb_transfer*) const in request-libusb.o
  "_libusb_get_bus_number", referenced from:
      librealsense::platform::get_device_path(libusb_device*) in enumerator-libusb.o
  "_libusb_get_config_descriptor", referenced from:
      librealsense::platform::usb_device_libusb::usb_device_libusb(libusb_device*, libusb_device_descriptor const&, librealsense::platform::usb_device_info const&, std::__1::shared_ptr<librealsense::platform::usb_context>) in device-libusb.o
      librealsense::platform::get_subdevices(libusb_device*, libusb_device_descriptor) in enumerator-libusb.o
  "_libusb_get_device_address", referenced from:
      librealsense::platform::get_device_path(libusb_device*) in enumerator-libusb.o
  "_libusb_get_device_descriptor", referenced from:
      librealsense::platform::get_device_path(libusb_device*) in enumerator-libusb.o
      librealsense::platform::usb_enumerator::query_devices_info() in enumerator-libusb.o
      librealsense::platform::usb_enumerator::create_usb_device(librealsense::platform::usb_device_info const&) in enumerator-libusb.o
  "_libusb_get_device_list", referenced from:
[... cut ...]
      librealsense::platform::handle_libusb::~handle_libusb() in device-libusb.o
  "_libusb_submit_transfer", referenced from:
      librealsense::platform::usb_messenger_libusb::submit_request(std::__1::shared_ptr<librealsense::platform::usb_request> const&) in messenger-libusb.o
  "_libusb_unref_device", referenced from:
      librealsense::platform::usb_device_libusb::~usb_device_libusb() in device-libusb.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

MartyG-RealSense, 2021-07-29:
Someone else who was building librealsense for M1 had a similar error message.  They didn't use the **Open with Rosetta** option to make the program x86/x64 compatible.  Instead, they followed the advice in the link below and then their build on M1 was successful.

https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-837115091

I do apologize for all of the difficulties that you have been having with building for M1.

aman3014, 2021-08-03:
Hey @MartyG-RealSense, I don't need to use librealsense anymore in the near future so I'll close the issue for now.

Hopefully Intel and/or Apple will provide better compatibility and documentation for the M1 macs in the future :)

Thanks for your help!

closed 2021-08-03 by aman3014

MartyG-RealSense, 2021-08-03:
Thanks very much @aman3014 for the update - I hope that we see you again in the future!