realsenseai/librealsense #9427: Intel RealSense D455 and Unity wrapper

lafith, 2021-07-16:
### Issue Description [Question]
Can I use the Unity wrapper for SDK with Intel RealSense D455 camera? Is it possible to stream live feed from left and right lens to Unity using this wrapper?

MartyG-RealSense, 2021-07-16:
Hi @lafith  D455 support in the RealSense Unity wrapper was added in SDK version **2.40.0**.  In that SDK version, Intel provided an update to the Unity wrapper to enable its use with Unity 2018 to 2020.  Before SDK 2.40.0, the D455 was supported in Unity 2017.

Like with librealsense scripting, you can define stream profiles to have index numbers to differentiate more than one stream of the same type.

[image]

lafith, 2021-07-16:
Ok, thank you. This was helpful.

lafith, 2021-08-23:
> Hi @lafith D455 support in the RealSense Unity wrapper was added in SDK version **2.40.0**. In that SDK version, Intel provided an update to the Unity wrapper to enable its use with Unity 2018 to [...]

Hi,
Other than obtaining depth and color feed as you have mentioned, is it possible to obtain color feed from the left and right cam present in D455?

MartyG-RealSense, 2021-08-23:
You can only obtain color from the **left** infrared imager of D455.  The right infrared imager is monochrome.  The D455 also needs firmware version **5.12.8.200** or newer installed to support this color from left infrared capability.

https://github.com/IntelRealSense/librealsense/issues/7897

closed 2021-08-23 by lafith