realsenseai/librealsense #9428: D435 - Resource temporarily unavailable

banuprathap, 2021-07-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   Ubuntu 14.04  | 
| Kernel Version (Linux Only)    |  4.4                                        | 
| Platform                                 | Intel Com Xpress  |
| SDK Version                            | 2.40.0                         | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
We have three D435 cameras connected to a USB3 root hub on the com express. We power these USB ports externally with dedicated 1A per port on our mother board.

We stream these cameras at 1280x720 @15FPS. After extensive testing, we discovered that the sensors stop streaming after a long while. We haven't spent anytime in understanding the duration but this can be sometime between several minutes to a day or two. We are working around this issue by resetting power to USB ports (other users have reported similar issues).

We started noticing that the sensor would sometime not connect back and fails with `Resource temporarily unavailable` exception.

Here is a snippet of how we initialize the device.

```
rs2::context m_ctx;
m_ctx.set_devices_changed_callback(bind(&RsSensorServer::changeDeviceCallback, this, _1));

void RsSensorServer::changeDeviceCallback(rs2::event_information& info)
{
    if(info.was_removed(m_device))
    {
        m_device = rs2::device();
    }
    if(!m_device)
    {
        rs2::device_list new_devices = info.get_new_devices();
        if (new_devices.size() > 0)
        {
            initDeviceHandle(new_devices);
            // if we have matching device start streaming
            if(m_device) startDevice();
        }
    }
}

void RsSensorServer::initDeviceHandle(rs2::device_list list)
{
    if(!m_device)
    {
        // process only if we don't already have initialized       
        if(list.size() == 0)
        {
           LOG() << "No Realsense devices were found!";
        }
        else
        {
            // iterate over all available devices
            for(size_t i = 0; i < list.size(); i++)
            {
                rs2::device dev;
                try
                {
                    dev = list[i];
                }
                catch(const std::exception& e)
                {
                    LOG() << "Device " << i+1 << "/" << list.size() 
                                        << " failed with exception: " << e.what();
                    continue;
                }
                <do more stuff here> 
             }
    }
}
```

Do you have any insights on why this is happening?

MartyG-RealSense, 2021-07-16:
Closed as a duplicate of https://github.com/IntelRealSense/librealsense/issues/9385

MartyG-RealSense, 2021-07-19:
Hi @banuprathap  Would it be possible to provide hardware specifications for the particular COM (Computer On Module) that you are using in your project, please?

A look at typical specifications of Computer On Modules makes me aware that they may be sufficiently low-end to be unsuited to having three cameras attached to them, especially if those cameras are all enabled simultaneously.  This could cause a drain on the computing device's resources that could lead to the **Resource temporarily unavailable** exception.  

Whilst RealSense 400 Series cameras have onboard Vision Processor D4 hardware to perform processing of data within the camera so that the computer's CPU does not have to do all of the processing work, the specification of the computer hardware does matter too in setups where there are multiple cameras attached to the same computer.

MartyG-RealSense, 2021-07-25:
Hi @banuprathap  Do you require further assistance with this case, please?  Thanks!

banuprathap, 2021-07-27:
Hi. Sorry for the delayed response. Here is a link to specs of the processor we use.
https://www.congatec.com/us/products/com-express-type-6/conga-tc170/

In summary, it is a 6th gen i7 with 8GB RAM. I'm not convinced it is due to host compute availability in our case.

MartyG-RealSense, 2021-07-27:
Thanks very much for the processor details.  A 2018 Intel seminar about multiple RealSense 400 Series cameras stated that an i7 processor was recommended for using 4 of the cameras simultaneously on the same computer.  So even taking into account that this advice was given a couple of years after the 6th generation was released, your processor's specification should be suitable for 3 cameras.

How does the SDK's C++ multiple camera example program **rs-multicam** perform in comparison in regard to stability?

https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam

You could test this example on Ubuntu by entering **rs-multicam** into the Ubuntu terminal.  If the examples and tools were built when librealsense was installed, they should be in the **usr/local/bin** folder.

banuprathap, 2021-08-05:
Hi Marty. I haven't had a chance to test the rs-multicam demo yet. But on a cursory look, it seems like there will be a UI rendering overhead.

I'll give it a shot and report back here.

MartyG-RealSense, 2021-08-05:
Thanks very much @banuprathap for the update - I look forward to your next report.  Good luck!