realsenseai/librealsense #9432: kinfu vector subscript out of range

Tony-le-han, 2021-07-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<48>.<0> }                          | 
| Language                            |  {C/C#/opencv/}                          | 
| Segment			|  {Kinect Fusion
[image]
[image]
 }                          | 

### Issue Description
<vector subscript out of range .>

MartyG-RealSense, 2021-07-18:
Hi @Tony-le-han   This specific error has occurred previously in https://github.com/IntelRealSense/librealsense/issues/4850 on RealSense SDK version **2.25.0**, so this reduces the likelihood that the problem is related to the SDK version that you are using. 

Have you made any modifications to the **rs-kinfu.cpp** script of the Kinfu wrapper please, such as editing the stream resolution, format or FPS in the cfg instruction?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/kinfu/rs-kinfu.cpp#L190

Tony-le-han, 2021-07-18:
> 嗨@Tony-le-han 此特定错误之前已在 RealSense SDK 版本**2.25.0**上的[#4850 中](https://github.com/IntelRealSense/librealsense/issues/4850)发生，因此这出现了问题与您使用的 SDK 版本的相关原因。****
> 
> **请问**Kinfu包装的**RS-kinfu.cpp**脚本有没有修改过，比如CFG指令中的流分辨率，格式或者FPS的修改？
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/kinfu/rs-kinfu.cpp#L190

并没有修改过，并出现了以下警告
[image]

MartyG-RealSense, 2021-07-18:
Are you able to run any of the OpenCV example programs without errors?

您是否能夠無錯誤地運行任何 OpenCV 示例程序？

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv#list-of-samples

Tony-le-han, 2021-07-18:
> 您是否能够无错误地运行任何 OpenCV 示例程序？
> 
> 您是否能够无错误地运行任何 OpenCV 示例程序？
> 
> https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv#list-of-samples

可以运行其他示例程序，只有rs-kinfu出错

MartyG-RealSense, 2021-07-18:
Because there is not a clear solution for the error that you posted, lets go through the steps of the rs-kinfu installation procedure.

**1.  Install OpenCV and opencv_contrib as described in the official guide.  Note that version 4 of OpenCV is required.**

It sounds as though you have done this if the OpenCV examples are working correctly.

**2.  In step 5, along with configuring OPENCV_EXTRA_MODULES_PATH, check the OPENCV_ENABLE_NONFREE option.**

If you used the official OpenCV installation guide to install OpenCV using CMake,  did you enable the OPENCV_ENABLE_NONFREE option?

**3.  When configuring CMake for librealsense, in addition to checking BUILD_CV_EXAMPLES, check the BUILD_CV_KINFU_EXAMPLE flag.  Also, ensure that BUILD_GRAPHICAL_EXAMPLES is checked.**

When building the librealsense SDK with CMake, did you include BUILD_CV_EXAMPLES and BUILD_GRAPHICAL_EXAMPLES in your CMake build instruction?

******************

由於您發布的錯誤沒有明確的解決方案，讓我們完成 rs-kinfu 安裝過程的步驟。

**1. 按照官方指南中的說明安裝 OpenCV 和 opencv_contrib。 請注意，需要 OpenCV 版本 4。**

如果 OpenCV 示例工作正常，聽起來好像您已經這樣做了。

**2. 在第 5 步中，在配置 OPENCV_EXTRA_MODULES_PATH 的同時，檢查 OPENCV_ENABLE_NONFREE 選項。**

如果您使用官方 OpenCV 安裝指南使用 CMake 安裝 OpenCV，您是否啟用了 OPENCV_ENABLE_NONFREE 選項？

**3. 為 librealsense 配置 CMake 時，除了檢查 BUILD_CV_EXAMPLES 外，還要檢查 BUILD_CV_KINFU_EXAMPLE 標誌。 此外，請確保選中 BUILD_GRAPHICAL_EXAMPLES。**

使用 CMake 構建 librealsense SDK 時，您是否在 CMake 構建指令中包含 BUILD_CV_EXAMPLES 和 BUILD_GRAPHICAL_EXAMPLES？

Tony-le-han, 2021-07-19:
> 由于您发布的错误没有明确的解决方案，让我们完成 rs-kinfu 安装过程的步骤。
> 
> **1. 按照官方指南中的说明安装 OpenCV 和 opencv_contrib。请注意，需要 OpenCV 的第 4 版。**
> 
> 如果 OpenCV 示例工作正常，听起来好像您已经这样做了。
> 
> **2. 在第 5 步中，在配置 OPENCV_EXTRA_MODULES_PATH 的同时，选中 OPENCV_ENABLE_NONFREE 选项。**
> 
> 如果您使用官方 OpenCV 安装指南使用 CMake 安装 OpenCV，您是否启用了 OPENCV_ENABLE_NONFREE 选项？
> 
> **3. 为librealsense配置CMake时，除了检查BUILD_CV_EXAMPLES外，还要检查BUILD_CV_KINFU_EXAMPLE标志。此外，确保选中 BUILD_GRAPHICAL_EXAMPLES。**
> 
> 使用 CMake 构建 librealsense SDK 时，您是否在 CMake 构建指令中包含 BUILD_CV_EXAMPLES 和 BUILD_GRAPHICAL_EXAMPLES？
> 
> 由于您发布的错误没有明确的解决方案，让我们完成 rs-kinfu 安装过程的步骤。
> 
> **1. 按照官方指南中的说明安装 OpenCV 和 opencv_contrib。 注意，需要 OpenCV 版本 4。**
> 
> 如果 OpenCV 示例工作正常，听起来就好像您已经腐化了。
> 
> **2.在第5步中，在配置OPENCV_EXTRA_MODULES_PATH的同时，检查OPENCV_ENABLE_NONFREE选项。**
> 
> 如果您使用官方 OpenCV 安装指南使用 CMake 安装 OpenCV，您是否可以启用 OPENCV_ENABLE_NONFREE 选项？
> 
> **3.为librealsense配置CMake时，除了检查BUILD_CV_EXAMPLES外，还要检查BUILD_CV_KINFU_EXAMPLE 标志。另外，请确保勾选BUILD_GRAPHICAL_EXAMPLES。**
> 
> 使用 CMake 构建 librealsense SDK 时，您是否在 CMake 构建内存中包含 BUILD_V_EXAMPLES 和 BUILD_GRAPHICAL_EXAMPLES？

我检查了您所说的，并未出错。
我把之前的文件，配置全部删除，重新安装了一遍。
在运行示例时出现了如下错误：
[image]
我在依赖项中添加了:msvcrtd.lib.结果能正常运行。
[image]
非常感谢您耐心的解答。

closed 2021-07-19 by Tony-le-han

MartyG-RealSense, 2021-07-19:
Thank you very much for sharing the knowledge that adding **msvcrtd.lib** to the dependencies solved your problem.

非常感謝您分享將 msvcrtd.lib 添加到依賴項解決了您的問題的知識。