realsenseai/librealsense #9437: how to get the RGB exposuretime when camera is in AutoExposure in C++

shiguangzqz, 2021-07-19:
I want to get the RGB Camera's exposure time in AutoExposure Mode.
I try to get this by this code 
`
auto dev=_profile.get_device();
auto val=dev.query_sensors()[1].get_option(rs2_option::RS2_OPTION_EXPOSURE);
`
However , the val under different illumination conditions did not change

so，can you show me the right code ？Thanks!

MartyG-RealSense, 2021-07-19:
Hi @shiguangzqz   When auto-exposure is enabled, the exposure value does not update.  The exposure value can be retrieved from camera **metadata** with the instruction **RS2_FRAME_METADATA_ACTUAL_EXPOSURE** if support for metadata is enabled.  However, the **RGB** exposure value cannot be obtained whilst auto-exposure is enabled.

https://github.com/IntelRealSense/librealsense/issues/2549#issuecomment-431863920
https://github.com/IntelRealSense/librealsense/issues/7267

shiguangzqz, 2021-07-21:
OK。thanks

closed 2021-07-21 by shiguangzqz