realsenseai/librealsense #9438: 人体3D扫描

Tony-le-han, 2021-07-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<48>.<0> }                          | 
| Language                            |  {C/C#//opencv/pcl/}                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
我想问一下：是否可以通过深度相机和转台相结合，利用kinfu 实现对人体的3D扫描，并输出点云。
[image]

MartyG-RealSense, 2021-07-19:
Translation: Human body 3D scan.  I want to ask: Is it possible to use kinfu to realize 3D scanning of the human body by combining a depth camera and a turntable, and output a point cloud.

**********

Hello @Tony-le-han  If you needed to use two cameras then in theory it would be possible to do this with KinectFusion (kinfu) by calibrating the extrinsics of the two cameras together.  It has been done in the past, though I could not find any example of it being achieved in kinfu with RealSense depth cameras.

Since rs-kinfu can build up a point cloud as a camera is moved, you could perhaps use one camera and have it move up and down on a motorised rail as the human subject rotates on the turntable, so that the full head to foot height of the human is captured in 360 degrees.

Alternatively, if you are able to consider a non-kinfu solution then stitching together of pointclouds from two RealSense 400 Series cameras into a single cloud has been demonstrated in ROS in the guide linked to below.

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

****************

你好@Tony-le-han 如果您需要使用兩個攝像頭，那麼理論上可以通過將兩個攝像頭的外在參數一起校準來使用 KinectFusion (kinfu) 來實現。 過去已經完成了，儘管我找不到任何使用 RealSense 深度相機在 kinfu 中實現的示例。

由於 rs-kinfu 可以在相機移動時建立點雲，您也許可以使用一個相機，讓它在人類主體在轉盤上旋轉時在電動軌道上上下移動，以便整個頭到腳的高度 360 度捕捉人類的影像。

或者，如果您能夠考慮非 kinfu 解決方案，那麼在下面鏈接的指南中已在 ROS 中演示了將來自兩個實感 400 系列攝像機的點雲拼接到單個雲中。

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

MartyG-RealSense, 2021-07-26:
Hello @Tony-le-han  Do you require further assistance with this case, please?  Thanks!

你好@Tony-le-han 請問這個案例需要進一步的幫助嗎？ 謝謝！

websocketing, 2021-07-31:
> 你好@Tony-le-han 请问这个案例需要进一步的帮助吗？谢谢！
> 
> 你好@Tony-le-han请问这个案例需要进一步的帮助吗？谢谢！

Does your company's camera support size measurement after human body scanning?

websocketing, 2021-07-31:
@MartyG-RealSense  Does your company's camera support size measurement after human body scanning?

MartyG-RealSense, 2021-07-31:
Hi @websocketing  I have replied at your question at https://github.com/IntelRealSense/librealsense/issues/9536

closed 2021-08-06 by MartyG-RealSense