realsenseai/librealsense #9440: Dropped frame. alloc_frame(...) returned nullptr

liuluyangdev, 2021-07-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Android  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Android 11  |
| SDK Version                            |  2.48.0                        | 
| Language                            |  Java                       | 
| Segment			|  Smartphone}                         | 

### Issue Description
I want to get DepthFrame Distance, here is my code:

`

                        FrameSet fs = f.as(Extension.FRAMESET);
                        if (fs != null) {
                            Frame depth_frame = fs.first(StreamType.DEPTH).releaseWith(frameReleaser);
                            DepthFrame depthFrame = depth_frame.as(Extension.DEPTH_FRAME);
                            final float depthValue = depthFrame.getDistance(depthFrame.getWidth() / 2, depthFrame.getHeight() / 2);
                            depthFrame.releaseWith(frameReleaser);
                            Log.d(TAG, "onCameraFrame--depthValue: " + depthValue);

                            try (FrameSet processed = fs.applyFilter(mColorizer)) {
                                if (mCount % 5 == 0) {
                                    mDetect.onCameraFrame(fs, handler, myView);
                                }
                                mGLSurfaceView.upload(processed);
                            }
                        }

`
error:
Dropped frame. alloc_frame(...) returned nullptr
Dropped frame. alloc_frame(...) returned nullptr
Dropped frame. alloc_frame(...) returned nullptr
......

MartyG-RealSense, 2021-07-19:
Hi @liuluyangdev  This kind of 'tombstone' error in Android applications that involves **null pointer dereference** is one of the most difficult to diagnose and cases of it have often not had a clear solution.

In one of these cases, a RealSense user shared their method of frame release that they said worked for them in their particular project.

https://github.com/IntelRealSense/librealsense/issues/5424#issuecomment-703303393

Alternatively, you could try the RealSense SDK's example program **java_example** which prints the distance to the pixel in the center of the depth frame.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/java_example

liuluyangdev, 2021-07-20:
@MartyG-RealSense ， How to get the distance of the person (face) by DepthFrame?

liuluyangdev, 2021-07-20:
@MartyG-RealSense The problem with nullptr can be solved by adding "depth_frame.close();"

liuluyangdev, 2021-07-20:
Anyone know?

liuluyangdev, 2021-07-20:
`

getDistance(int x, int y)

`

I don`t know what it means in terms of params

MartyG-RealSense, 2021-07-20:
The x and y variables refer to the x and y coordinates of 2D pixels, and getDistance uses those values to obtain a z-depth value for that pixel.

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#point-coordinates

liuluyangdev, 2021-07-22:
@MartyG-RealSense Hi, why do I have to Request Permissions every time I reboot my phone. Is there any way to Grant only once? Thanks.

[image]

liuluyangdev, 2021-07-22:
Is there a default Grant method?

MartyG-RealSense, 2021-07-22:
Is your application linked to an **AAR** file that handles camera permissions, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/android.md#build-a-realsense-application

liuluyangdev, 2021-07-23:
@MartyG-RealSense Thank you for your support. How do I grant permissions by default to a device that is not rooted? My Application version is librealsense-2.48.0.aar.

MartyG-RealSense, 2021-07-23:
I don't honestly have this knowledge myself.  A developer of the RealSense Android wrapper offers advice about permissions on unrooted devices in the link below though.

https://github.com/IntelRealSense/librealsense/issues/3385#issuecomment-481723287

liuluyangdev, 2021-07-29:
I have no ideas.

closed 2021-08-09 by MartyG-RealSense