realsenseai/librealsense #9442: Minimum size code for interfacing with D435i

oym1994, 2021-07-19:
| Required Info                         |                                                                |
| Camera Model                       |  D435i | 
| Operating System & Version |   Ubuntu18.04  | 
| Language                            |  C++                   | 

Hi, I am using Realsense D435i in my project and it's required to decrease the code size as much as possible and ROS is forbidden. I want to know whether there is such a minimum size version which can be used to interface with Realsense D435i to reading and setting RGB-D and IMU data.  I prefer to use it as the manner to use a USB camera with IMU.

Thank you for your kind help!    Looking for your kind response!

MartyG-RealSense, 2021-07-19:
Hi @oym1994  if you are building the SDK from source code with CMake then you could potentially reduce the project size noticably by building the SDK with debug symbols and other content stripped out.

https://github.com/IntelRealSense/librealsense/issues/7982

oym1994, 2021-07-20:
Thanks for your good advice! 
Then how about the building and running dependency? Another requirement is to try to delete those unnecessary dependency in my mini-PC platform and leave those dependency which is necessary for basic function(reading and setting RGB-D and IMU data). My original solution is to extract those necessary code from the complete project and then find out the related necessary dependency. But I am afraid it's a time-consuming task with potential risk and there may exist other better way.

MartyG-RealSense, 2021-07-20:
Using the CMake build instruction **-DBUILD_WITH_OPENMP=false** may help trim the build, as described in the link below.

https://github.com/IntelRealSense/librealsense/issues/744#issuecomment-340014751

If you do not need example programs and tools such as the RealSense Viewer then you could also exclude these from the build with **-DBUILD_EXAMPLES=false -DBUILD_GRAPHICAL_EXAMPLES=false**

oym1994, 2021-07-21:
Get it! Thanks again for your kind response! Maybe it is not wise to trim the source code.

MartyG-RealSense, 2021-07-26:
Hi @oym1994  Do you require further assistance with this case, please?  Thanks!

oym1994, 2021-07-27:
> Hi @oym1994 Do you require further assistance with this case, please? Thanks!

No, thank you!  Only the bad news that we can't build from source for the internet problem. So we have to utilise the library.

MartyG-RealSense, 2021-07-27:
Thanks very much for the update!   If you mean that you cannot build from source using the **RSUSB** method (which requires an internet connection), then you can instead build from source by patching the kernel with the method in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

closed 2021-08-08 by MartyG-RealSense