realsenseai/librealsense #9444: vmap allocation for size 1847296 failed: use vmalloc=<size> to increase size.

banuprathap, 2021-07-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   Ubuntu 14.04  | 
| Kernel Version (Linux Only)    |  4.4                                        | 
| Platform                                 | Intel Com Xpress  |
| SDK Version                            | 2.40.0                         | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
Hi. We ran into a strange issue while unplugging and plugging back the sensor. See logs attached below.

```
kernel: [ 7213.449252] usb 8-3: new SuperSpeed USB device number 8 using xhci_hcd
kernel: [ 7213.465698] usb 8-3: New USB device found, idVendor=8086, idProduct=0b07
kernel: [ 7213.465701] usb 8-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
kernel: [ 7213.465703] usb 8-3: Product: Intel(R) RealSense(TM) Depth Camera 435 
kernel: [ 7213.465705] usb 8-3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
kernel: [ 7213.465706] usb 8-3: SerialNumber: 832513024132
kernel: [ 7213.468226] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb8/8-3/8-3:1.0/input/input146
RealsenseDriver: DEBUG: file: src/RsSensorServer.cpp, line: 497
RealsenseDriver: Device with SerialNumber: 832112070837 name Intel RealSense D435 FW version 05.12.09.00 product ID 0B07 was found with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb8/8-3/8-3:1.0/video4linux/video21
RealsenseDriver: DEBUG: file: src/RsSensorServer.cpp, line: 507
RealsenseDriver: Device USB type: 3.2
RealsenseDriver: DEBUG: file: src/RsSensorServer.cpp, line: 529
RealsenseDriver: Loaded advanced mode settings from file
RealsenseDriver: DEBUG: file: src/RsSensorServer.cpp, line: 624
RealsenseDriver: Enabled decimation_filter
RealsenseDriver: DEBUG: file: src/RsSensorServer.cpp, line: 686
RealsenseDriver: EnableDepth
RealsenseDriver: DEBUG: file: src/RsSensorServer.cpp, line: 716
RealsenseDriver: Open sensor
kernel: [ 7219.174219] vmap allocation for size 1847296 failed: use vmalloc=<size> to increase size.
```

MartyG-RealSense, 2021-07-20:
Hi @banuprathap  There were not any past librealsense cases of this particular warning, so I widened my research to occurrences of it in Ubuntu in general.  You could try increasing the **virtual memory allocation** (vmalloc) size using the guidance in the link below.

http://thinking-electron.blogspot.com/2015/05/how-to-increase-vmalloc-size-vmalloc.html 

If the method in that link does not work for you, you can find further information by googling for the term **ubuntu increase vmalloc size**

banuprathap, 2021-07-20:
Hi @MartyG-RealSense. Increasing the vmalloc is not helping with the issue. No matter how high I set it, the process eventually consumes max limit and fails with `vmap allocation failed` error.

I noticed that every time the device is disconnected and reconnected, the memory footprint (as reported under `/proc/<PID>/status` increases and eventually hits the max limit.

We detect the device disconnects using the `context.set_devices_changed_callback()` functionality and I suspect that this call is causing the memory leak, as also reported in several other threads.

I noticed that your recommendation was to not use the context object to query devices. Do you have any example of how we can detect a device disconnect and reconnect without context?

Thanks for all your help!

MartyG-RealSense, 2021-07-21:
A type of memory leak that occurs when starting and stopping the pipeline has been a known issue.  It was reported by a RealSense team member to have been addressed in librealsense version **2.48.0**.

https://github.com/IntelRealSense/librealsense/issues/8091#issuecomment-871036563

The most recent reference I have made to **set_devices_changed_callback** that I know of at the time of writing this was the one in the link below.

https://github.com/IntelRealSense/librealsense/issues/8836#issuecomment-820365268

banuprathap, 2021-07-27:
What can I do/provide to help the realsense team troubleshoot the memory leak we are noticing within the SDK?

MartyG-RealSense, 2021-07-27:
You could try SDK version **2.48.0** to see whether that version corrects the leak that you are experiencing, as a type of memory leak that occurs when starting the pipeline after a stop was reported by a RealSense team member to have been addressed with fixes in 2.48.0.  If that version corrects your problem then it would save the time and labor of trying to develop a fix yourself.

I carefully researched your question about disconnection and reconnection but could not find an example that did not use context (ctx) to do so.

banuprathap, 2021-07-28:
I tried version 2.48.0 but the issue persists. Do you know what else I can try?

MartyG-RealSense, 2021-07-28:
@banuprathap  Thanks very much - good luck!

banuprathap, 2021-07-28:
Confirming issue is resolved in v2.48.0 and closing the issue. Thank you for the continued assistance.

closed 2021-07-28 by banuprathap