realsenseai/librealsense #9445: cannot connect via Thunderbolt 3

christian-rauch, 2021-07-19:
| | |
|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|922612070857|
|**Firmware Version**|05.12.14.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|923123023876|
|**Firmware Update Id**|923123023876|

I am unable to connect via a Thunderbolt 3 cable (especially this one: https://www.amazon.co.uk/Belkin-Thunderbolt-USB-C-Certified-40Gbps/dp/B072KPBKS3). The RealSense Viewer reports errors such as `USB SCP overflow` and `USB CAM overflow`.

The same cable works when I connect the D435 to an Android phone.

MartyG-RealSense, 2021-07-19:
Hi @christian-rauch  Are you having the problem when you attempt to connect the camera to a **computer** instead of an Android phone?  If this is correct, which computer are you using the camera with, please (PC, Mac, Raspberry Pi, Nvidia Jetson, etc)

christian-rauch, 2021-07-19:
Yes, the problem I described is when connecting to a PC, specifically a Dell XPS 15 7590. My point with Android was that the cable itself works. I tried this on Windows (same PC) but got the same issues.

I can see via `lsusb` that the camera is connected and can even use it as a regular webcam (e.g. in `cheese`). Just the realsense driver seems to have issues with it.

MartyG-RealSense, 2021-07-20:
Your findings about Thunderbolt 3 cabling working on Android are replicated by the RealSense user in the link below.

https://github.com/IntelRealSense/librealsense/issues/4091#issuecomment-590974048

Perhaps if the camera has to be connected to a **USB 3 OTG** micro USB port on the Android device via an OTG cable or adapter then that intermediary interface acts as a buffer for high volumes of incoming data that prevents the USB port on the Android from being overwhelmed.

In regard to use of RealSense and Thunderbolt with PC computers, there is not enough information available to draw a firm conclusion, though amongst those few cases the reports of success with Thunderbolt outnumbered the cases that had probems.

If the camera can be detected in the Viewer, does performance improve if you reset the camera before you begin streaming by using the **Reset Hardware** menu option under the Viewer's **More** option?

[image]

christian-rauch, 2021-07-20:
I reset the hardware, but the outcome is the same.

This looks like a "classic" bandwidth issue. I can receive images from the "RGB Camera" in the RealSense Viewer using either the highest resolution (1920x1080 @ 8Hz) or the highest FPS (424x240 @ 60Hz). Initially, I get the message that no frames have been received, but after about 1~2 sec. I receive colour frames.

In the default configuration (640x480 @ 15Hz) I cannot receive frames from the "Stereo Module". However, when reducing the resolution to 256x144 @ 90Hz I eventually receive some depth frames, albeit at a very low frequency.

@MartyG-RealSense I noticed that your screenshot shows a USB 3.2 connection. My RealSense viewer only shows a USB 2.1 connection. When using a regular USB-A to USB-C cable, this is shown as USB 3.2 connection and I can use the highest resolution or framerate for the "Stereo Module".

MartyG-RealSense, 2021-07-20:
If the USB overflow messages disappear when using a very low resolution then that would match my expectations that accompany such errors: that the USB port might be being overwhelmed with more data than it could process in a timely manner when using higher resolutions.  The bandwidth limitations of USB 2.1 would certainly exacerbate that.  Intel have an article at the link below about using 400 Series cameras on a USB 2 connection that highlights this.

https://www.intelrealsense.com/usb2-support-for-intel-realsense-technology/

Unless the high-speed capability of a Thunderbolt 3 cable is vital to your application then it may be more reliable to just use a regular USB-C cable.

christian-rauch, 2021-07-20:
> Unless the high-speed capability of a Thunderbolt 3 cable is vital to your application then it may be more reliable to just use a regular USB-C cable.

Since USB 3 is sufficient for using the D435 and Thunderbolt 3 has a higher bandwidth than USB 3, a Thunderbolt 3 connection should be more than capable to transfer all the images. The Thunderbolt 3 cable is not the issue per se as it works with my Android device.

I also tried this with a USB 3.1 cable with USB-C connectors (there are also USB-C cables that only support USB 2) and this works as expected. The Thunderbolt 3 cable with a much higher bandwidth should not be the limiting factor. To this end, I still think that there is an issue with either the USB/Thunderbolt setup and/or the realsense driver as the bandwidth should be sufficient.

If you have ideas on how to debug this, I am happy to compile and run development versions of the driver and collect logs to solve this "Thunderbolt 3 connection" issue.

MartyG-RealSense, 2021-07-20:
I will seek advice from Intel about your Thunderbolt 3 question.  Thanks very much for your patience.

MartyG-RealSense, 2021-07-23:
Hi @christian-rauch  I have received advice from Intel.  Could you perform a **USB trace** on your Dell XPS 15 7590 and provide us with the results please?  Thanks!

https://wiki.kubuntu.org/Kernel/Debugging/USB

christian-rauch, 2021-07-23:
On my system, the device is connected to bus 1, amongst others:
```
Bus 001 Device 004: ID 27c6:5395 Shenzhen Goodix Technology Co.,Ltd. Fingerprint Reader
Bus 001 Device 028: ID 8086:0b07 Intel Corp. Intel(R) RealSense(TM) Depth Camera 435 
Bus 001 Device 003: ID 8087:0029 Intel Corp. 
Bus 001 Device 005: ID 0c45:6723 Microdia Integrated_Webcam_HD
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
and I logged with `sudo cat /sys/kernel/debug/usb/usbmon/1u > D435_USB_log.txt`. The documentation does say how to log specific devices, so the log probably contains all the devices on bus 1.

This is the log file I get after starting the real-sense viewer and trying to open the stereo and colour camera: [D435_USB_log.txt](https://github.com/IntelRealSense/librealsense/files/6868232/D435_USB_log.txt).

MartyG-RealSense, 2021-07-23:
I will feed your kindly provided log information back to Intel.  Thanks for your patience!

MartyG-RealSense, 2021-07-23:
Intel also said that you could perform a **USB protocol trace** from the physical USB bus, though this would require the purchase of **USB analyzer** hardware like the one in the link below and so would be an optional step.

https://teledynelecroy.com/protocolanalyzer/usb/advisor-t3

MartyG-RealSense, 2021-07-29:
Hi @christian-rauch  Do you require further assistance with this case, please?  Thanks!

christian-rauch, 2021-07-29:
We bought dedicated USB3 USB-C cables for the camera. So the Thunderbolt 3 issue does not occur with these cables.

However, I would still like to see the driver working with Thunderbolt 3 cables (so that others can avoid this issue in the future).

MartyG-RealSense, 2021-07-29:
Thanks very much @christian-rauch for the update!  I will feed back your findings to Intel.

christian-rauch, 2021-08-04:
I had similar issues with the Azure Kinect and opened an issue in the Azure Kinect SDK repo. There it was pointed out that the specific Thunderbolt 3 cable from Belkin does not support USB3: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1646#issuecomment-890282125.
In summary, it may be common that active Thunderbolt 3 cables only support USB2, while passive Thunderbolt 3 cables will support USB2 and USB3.

This would explain why it did not work with the standard resolutions on my laptop. However, I am still baffled that it worked with the standard resolutions on a different Windows 10 machine without issues.

In any case, I don't think this can be fixed in the driver easily and I switched to a dedicated USB3 cable anyway. Maybe it would be useful to have this information somewhere in the documentation and driver warnings when the bandwidth is not sufficient.

closed 2021-08-04 by christian-rauch

MartyG-RealSense, 2021-08-04:
I will pass on your further feedback to Intel.  Thanks again!