realsenseai/librealsense #9446: Crash related to concurrency issue in playback_sensor class

blackccpie, 2021-07-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 (playback ) | 
| Firmware Version                   | 05.11.11.100 (old replay!) | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.45.0                          | 
| Language                            |  C++                         | 
| Segment			|  others                          | 

### Issue Description
I encouter crash of my application referring to this line:
https://github.com/IntelRealSense/librealsense/blob/156c1bd163fbc735d07a98970a504408d5a6a9e3/src/media/playback/playback_sensor.h#L130
The crash happens at the end of my program, when I stop the pipeline. I presume there is a concurrency issue related to the attribute `m_user_callback` being reset here:
https://github.com/IntelRealSense/librealsense/blob/156c1bd163fbc735d07a98970a504408d5a6a9e3/src/media/playback/playback_sensor.cpp#L161
In the `playback_sensor::stop` method, the `m_user_callback` modification is protected by a mutex, but in the `playback_sensor::handle_frame` the attribute is accessed without visible sync protection.
I don't have very concise sample code to reproduce this now, as it is non-deterministic and related to multi-threading, but I'll post it here as sson as I have one.
Last thing, the bug happens in **non-realtime playback mode**, and doesn't seem to occur in realtime playback mode. 

Thanks for your help,

Regards,

Albert

PS: this bug may be related to issue #4897

MartyG-RealSense, 2021-07-20:
Hi @blackccpie  If you are playing back a bag file using **cfg.enable_device_from_file()** then you can set your script to stop playback automatically at the last frame of the bag instead of looping around to the start of the bag again if you place **repeat_playback=False** in the brackets of that instruction.  For example:

**cfg.enable_device_from_file(filename, repeat_playback=False); **

blackccpie, 2021-07-20:
Hi @MartyG-RealSense, in fact I doubt this issue is related to the looping feature. I already have playback looping disabled in my code, and the crash happens when I stop the playback before reaching the end of the bag.

MartyG-RealSense, 2021-07-20:
How about programming a mechanism to read the number of the end frame of the bag and then carrying out a check to not allow stop to activate if the current frame number is less than the number of the final frame?

The link below has an example of a bag frame-counting script in C++.

https://support.intelrealsense.com/hc/en-us/community/posts/360031847974/comments/360007849114

blackccpie, 2021-07-21:
@MartyG-RealSense : Thanks for your feedback. My problem is that I have unit tests that have been running for more than a year successfully with a previous sdk version, and now they are crashing with v2.45.0, so I'd rather first know if this is a real flaw that could be fixed and integrated in the upcoming sdk version soon. If it not planned to be corrected soon, then I'll consider your workaround, or sdk downgrade.

MartyG-RealSense, 2021-07-21:
I am not involved in SDK development activities so I do not have knowledge of fixes that are being worked on for upcoming SDK versions, unfortunately.

Nir-Az, 2021-07-22:
Hi @blackccpie ,
My name is Nir and I am from the SDK development team.

First thank you for your feedback, we really appreciate costumers feedback and glad to assist.
During the last 2 month this code section (dispatcher/playback-device/playback-sensor) had been refactored.
We are familiar with crash cases on this spots as the use cases over there has lots of corner cases, some were not handled correctly.
So far the latest code seems much more robust.

I suggest grabbing the latest development branch and rerun the UT's, if it works for you,  you can take the upcoming next release.

If you will still have issues please let up know and attach the log so we can investigate.

Good luck 👍

blackccpie, 2021-07-27:
Hi @Nir-Az,

Thanks for your suggestion, I'll try the development branch asap and will provide feedback.

blackccpie, 2021-07-28:
@Nir-Az : just made a quick test this morning, and with the current development branch I get no crash anymore. I'll wait for the next official release to confirm the issue is resolved.

MartyG-RealSense, 2021-08-09:
Case closed due to no further comments received.

closed 2021-08-09 by MartyG-RealSense

blackccpie, 2021-09-02:
The issue is fixed running with the latest release v2.49.0

MartyG-RealSense, 2021-09-02:
Great to hear that your issue is fixed, @blackccpie - thanks very much for the update!