realsenseai/librealsense #9447: Pyrealsense2 ERROR on JetsonNano

b2992872, 2021-07-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |  | 
| Operating System & Version |   inux (Ubuntu 16.04 )  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |                        | 
| Language                            |  python                    | 
| Segment			|    | 

I refer to this question Running pyrealsense2 on JetsonNano，cannot import pyrealsense2 after installation on NANO，as shown in the picture

https://github.com/IntelRealSense/librealsense/issues/6964

[image]
[image]
                     

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-07-20:
Hi @b2992872  A method for correcting this error on Jetson (it also works for Raspberry Pi) is to locate a pair of **.so files** named 'librealsense2.so' and 'pyrealsense2.so' and copy-and-paste them to the folder location where your pyrealsense2 project script is stored.  

If Python 3 is being used then the pyrealsense2.so file will likely have a more complex filename based on your Python 3 version number, such as 'pyrealsense2.cpython-35m-arm-linux-gnueabihf.so'.  For example, Python 3.5 may generate a filename with '35m' in it, whilst Python 3.7 may have '37m', and so on.

I believe that the .so files can be found at the folder location **librealsense > build > wrappers >python**.  In the case of the exotically named pyrealsense2.so file on a Python 3 installation, you may be able to find it by looking for a file whose name starts with **pyrealsense2** and ends in **.so**

[image]

b2992872, 2021-07-21:
I put them together as you said and executed the program, but the result seems to be the same
[image]

b2992872, 2021-07-21:
I arrived at the location you mentioned, but the file you mentioned is missing. Is it because of an error in the build process?
The failure of this step in the build process is the cause of this problem?
cmake ../-DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=... -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA
[image]

b2992872, 2021-07-22:
Yes, but the last step was unsuccessful.
 you can try a more advanced build, which builds the example programs and includes optimizations such as building with CUDA support for faster alignment processing on devices such as Jetson that are equipped with an Nvidia graphics GPU.

cmake ../-DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=... -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true

MartyG-RealSense, 2021-07-22:
**RSUSB_BACKEND=ON** requires an internet connection on the computer that the installation is being performed on, as it installs dependencies through the internet instead of through kernel patching.  Can you confirm that you have a working internet connection on that particular computer, please?

b2992872, 2021-07-23:
I make sure my device is connected to the internet。

MartyG-RealSense, 2021-07-23:
When you performed **md build && cd build** in the extracted source code folder, did you do so whilst in the folder containing the SDK files and not the empty folder, please?  In other words, were you in the folder **librealsense-2.39.0 > librealsense-2.39.0**

[image]

Also, the librealsense SDK version that you used (2.39.0) is quite old now.  Though the https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 guide suggests 2.38.0, this recommendation reflects the date when the guide was published and it should be fine to use a recent version from the SDK **Releases** page, such as **2.48.0**.

b2992872, 2021-07-25:
I followed your suggestion and used 2.48.0, the result is still the same, there will be an error I noted
[image]
[image]

MartyG-RealSense, 2021-07-25:
Let's check through the steps of the installation process for building librealsense and the Python wrapper together to see whether anything has been missed.

1.  You downloaded the source code folder of librealsense and navigated to the directory where the CMake files are located.

[image]

2.  Whilst in this directory, you used **mkdir build && cd build** or a similar command to create a **build** directory and navigate into it.

[image]

3.  Whilst in the **build** folder you input the CMake build command below to build librealsense and the Python wrapper together:

**cmake ../-DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=... -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA**

4.  When this command is run, librealsense builds successfully and at the end of the build process, you have a **wrappers > python** directory in your build folder.

[image]

5.  You tried to run a test script called **numpy.py** from the 'Desktop' location on your computer but received a **No module named pyrealsense2** error.

[image]

Running the script from the Desktop location would not work though unless you had done one of the following: 

(a) Copy and pasted the two .so files to the same location as your test script - the 'Desktop' location in this case.

or (b) Defined a **PYTHONPATH** so that Python could find pyrealsense2 no matter which location on your computer you ran your test script from.

You can define the PYTHONPATH by inputting the command below into your Ubuntu terminal whilst in the **build** folder:

**export PYTHONPATH=$PYTHONPATH:/usr/local/lib**

[image]

After inputting the PYTHONPATH, are you then able to run your **numpy.py** script from the Desktop, please?

b2992872, 2021-07-26:
You are right, but I did not generate two .so files when I wanted to build librealsense and Python

MartyG-RealSense, 2021-07-26:
It is not necessary to use the two .so files if a PYTHONPATH has been defined.  Can you confirm please whether you are able to run your **numpy.py** script successfully if you have defined a PYTHONPATH please?

b2992872, 2021-07-26:
Still after i tried Receive a No module named pyrealsense2 error，
[image]

b2992872, 2021-07-26:
I am on the desktop, open terminal, and enter export PYTHONPATH=$PYTHONPATH:/usr/local/lib
[image]

b2992872, 2021-07-26:
After I defined PYTHONPATH in the build data, I executed the numpy.py test script on the desktop, but it still failed to execute successfully.
[image]

MartyG-RealSense, 2021-07-26:
My understanding is that the **.bashrc** file is a hidden file located in the home directory.  Is that the bashrc file that you edited, please?

https://www.programmersought.com/article/25181343388/

Were you also able to define the PYTHONPATH by inputting the command below into your Ubuntu terminal whilst in the build folder instead of editing the .bashrc file?

**export PYTHONPATH=$PYTHONPATH:/usr/local/lib**

banuprathap, 2021-07-26:
@b2992872 My two cents here:

When you do `export PYTHONPATH=$PYTHONPATH:/usr/local/lib` from the command line, it will be applied only to the current shell. When you open a new shell, by opening a new tab or window, you'd have to type `export PYTHONPATH=$PYTHONPATH:/usr/local/lib` again.

One way to avoid doing it for each new shell, is to do what MartyG is recommending above. Hope this helps.

Cheers!

b2992872, 2021-07-28:
Give me time to try, thanks

b2992872, 2021-08-02:
import cv2
import re
import numpy as np
import jetson.inference
import jetson.utils
import argparse
import sys
import os
import io
import time
import json
import random
import pyrealsense2 as rs

# parse the command line

parser = argparse.ArgumentParser(description="Locate objects in a live camera stream using an object detection DNN.", formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.detectNet.Usage() +jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())

parser.add_argument("--network", type=str, default="ssd-mobilenet-v2",help="pre-trained model to load (see below for options)")

parser.add_argument("--threshold", type=float, default=0.5,help="minimum detection threshold to use")

parser.add_argument("--width", type=int, default=640,help="set width for image")

parser.add_argument("--height", type=int, default=480,help="set height for image")

opt = parser.parse_known_args()[0]
# load the object detection network

net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)

# Configure depth and color streams

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, opt.width, opt.height, rs.format.z16, 30)
config.enable_stream(rs.stream.color, opt.width, opt.height, rs.format.bgr8, 30)

# Start streaming

pipeline.start(config)

os.system("sudo sh -c 'echo 128 > /sys/devices/pwm-fan/target_pwm'")

press_key = 0
while (press_key==0):
	
   # Wait for a coherent pair of frames: depth and color

   frames = pipeline.wait_for_frames()
   depth_frame = frames.get_depth_frame()
   color_frame = frames.get_color_frame()
   if not depth_frame or not color_frame:
       continue	
   # Convert images to numpy arrays

   depth_image = np.asanyarray(depth_frame.get_data())
   color_image = np.asanyarray(color_frame.get_data())
   depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
   show_img = np.asanyarray(color_frame.get_data())
   
   gray = cv2.cvtColor(color_image, cv2.COLOR_BGR2GRAY)

   # convert to CUDA (cv2 images are numpy arrays, in BGR format)
   bgr_img = jetson.utils.cudaFromNumpy(show_img, isBGR=True)

	
   # convert from BGR -> RGB
   img = jetson.utils.cudaAllocMapped(width=bgr_img.width,height=bgr_img.height,format='rgb8')

   jetson.utils.cudaConvertColor(bgr_img, img)

   # detect objects in the image (with overlay)

   detections = net.Detect(img)
   for num in range(len(detections)) :
       score = round(detections[num].Confidence,2)
       box_top=int(detections[num].Top)
       box_left=int(detections[num].Left)
       box_bottom=int(detections[num].Bottom)
       box_right=int(detections[num].Right)
       box_center = detections[num].Center
       label_name = net.GetClassDesc(detections[num].ClassID)

   point_distance=0.0
   for i in range (10):
       
       point_distance =  depth_frame.get_distance(int(box_center[0]),int(box_center[1]))

       point_distance = np.round(point_distance , 3)

       distance_text = str(point_distance) + 'm  ' + str(score)+ '%'

       cv2.rectangle(show_img,(box_left,box_top),(box_right,box_bottom),(255,0,0),2)

       cv2.line(show_img,(int(box_center[0])-10, int(box_center[1])),(int(box_center[0]+10), int(box_center[1])),(0, 0, 255), 3)

       cv2.line(show_img,(int(box_center[0]), int(box_center[1]-10)),(int(box_center[0]), int(box_center[1]+10)),(0, 0, 255), 3)

       cv2.putText(show_img,label_name + ' ' + distance_text,(box_left+5,box_top+20),cv2.FONT_HERSHEY_SIMPLEX,0.4,(0,0,255),1,cv2.LINE_AA)

       cv2.putText(show_img,"{:.0f} FPS".format(net.GetNetworkFPS()),(int(opt.width*0.8), int(opt.height*0.1)),cv2.FONT_HERSHEY_SIMPLEX,1,(0,0,255),2,cv2.LINE_AA)

       display = cv2.resize(show_img,(int(opt.width*1.5),int(opt.height*1.5)))

       images = np.hstack((color_image, depth_colormap))
       #cv2.imshow('Detecting...',display)
       cv2.imshow('RealSense', images)

   keyValue=cv2.waitKey(1)
   if keyValue & 0xFF == ord('q'):
       press_key=1

# 关闭所有 OpenCV 窗口

cv2.destroyAllWindows()
pipeline.stop()

b2992872, 2021-08-02:
import cv2
import re
import numpy as np
import jetson.inference
import jetson.utils
import argparse
import sys
import os
import io
import time
import json
import random
import pyrealsense2 as rs

# parse the command line

parser = argparse.ArgumentParser(description="Locate objects in a live camera stream using an object detection DNN.", formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.detectNet.Usage() +jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())

parser.add_argument("--network", type=str, default="ssd-mobilenet-v2",help="pre-trained model to load (see below for options)")

parser.add_argument("--threshold", type=float, default=0.5,help="minimum detection threshold to use")

parser.add_argument("--width", type=int, default=640,help="set width for image")

parser.add_argument("--height", type=int, default=480,help="set height for image")

opt = parser.parse_known_args()[0]
# load the object detection network

net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)

# Configure depth and color streams

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, opt.width, opt.height, rs.format.z16, 30)
config.enable_stream(rs.stream.color, opt.width, opt.height, rs.format.bgr8, 30)

# Start streaming

pipeline.start(config)

os.system("sudo sh -c 'echo 128 > /sys/devices/pwm-fan/target_pwm'")

press_key = 0
while (press_key==0):
	
   # Wait for a coherent pair of frames: depth and color

   frames = pipeline.wait_for_frames()
   depth_frame = frames.get_depth_frame()
   color_frame = frames.get_color_frame()
   if not depth_frame or not color_frame:
       continue	
   # Convert images to numpy arrays

   depth_image = np.asanyarray(depth_frame.get_data())
   color_image = np.asanyarray(color_frame.get_data())
   depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
   show_img = np.asanyarray(color_frame.get_data())
   
   gray = cv2.cvtColor(color_image, cv2.COLOR_BGR2GRAY)

   # convert to CUDA (cv2 images are numpy arrays, in BGR format)
   bgr_img = jetson.utils.cudaFromNumpy(show_img, isBGR=True)

	
   # convert from BGR -> RGB
   img = jetson.utils.cudaAllocMapped(width=bgr_img.width,height=bgr_img.height,format='rgb8')

   jetson.utils.cudaConvertColor(bgr_img, img)

   # detect objects in the image (with overlay)

   detections = net.Detect(img)
   for num in range(len(detections)) :
       score = round(detections[num].Confidence,2)
       box_top=int(detections[num].Top)
       box_left=int(detections[num].Left)
       box_bottom=int(detections[num].Bottom)
       box_right=int(detections[num].Right)
       box_center = detections[num].Center
       label_name = net.GetClassDesc(detections[num].ClassID)

   point_distance=0.0
   for i in range (10):
       
       point_distance =  depth_frame.get_distance(int(box_center[0]),int(box_center[1]))

       point_distance = np.round(point_distance , 3)

       distance_text = str(point_distance) + 'm  ' + str(score)+ '%'

       cv2.rectangle(show_img,(box_left,box_top),(box_right,box_bottom),(255,0,0),2)

       cv2.line(show_img,(int(box_center[0])-10, int(box_center[1])),(int(box_center[0]+10), int(box_center[1])),(0, 0, 255), 3)

       cv2.line(show_img,(int(box_center[0]), int(box_center[1]-10)),(int(box_center[0]), int(box_center[1]+10)),(0, 0, 255), 3)

       cv2.putText(show_img,label_name + ' ' + distance_text,(box_left+5,box_top+20),cv2.FONT_HERSHEY_SIMPLEX,0.4,(0,0,255),1,cv2.LINE_AA)

       cv2.putText(show_img,"{:.0f} FPS".format(net.GetNetworkFPS()),(int(opt.width*0.8), int(opt.height*0.1)),cv2.FONT_HERSHEY_SIMPLEX,1,(0,0,255),2,cv2.LINE_AA)

       display = cv2.resize(show_img,(int(opt.width*1.5),int(opt.height*1.5)))

       images = np.hstack((color_image, depth_colormap))
       #cv2.imshow('Detecting...',display)
       cv2.imshow('RealSense', images)

   keyValue=cv2.waitKey(1)
   if keyValue & 0xFF == ord('q'):
       press_key=1

cv2.destroyAllWindows()
pipeline.stop()

closed 2021-08-02 by b2992872