realsenseai/librealsense #9448: Python save to disk example

chain-tech, 2021-07-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi, I have been exploring the node and python examples. I was able to play with node module to capture and save using the available examples.
However, for python wrapper, I don't get a straight forward answer to save depth and color images. I searched a lot in the issues as well but could not get a simple answer.
All i'm looking is a python equivalent of nodejs-save-to-disk.js that can save color and depth image. Please help

MartyG-RealSense, 2021-07-20:
Hi @chain-tech   The link below provides an example of Python scripting for exporting depth and color frames as PNG images using the **imwrite** instruction.

https://github.com/IntelRealSense/librealsense/issues/2877#issuecomment-446102880

A more advanced example that makes use of alignment is here:

https://github.com/IntelRealSense/librealsense/issues/3658

chain-tech, 2021-07-20:
Thanks @MartyG-RealSense , I have already gone through those links you shared.
I will try to work more with those examples.
I am wondering why the python save to disk part is more complicated compared to the node version and why there's predefined example in the list of sample examples.

MartyG-RealSense, 2021-07-20:
Another approach to color and depth file export is to save the color as a PNG and the depth as an array of scaled matrices in an **.npy** file, like in the Python script in the link below.

https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225

An alternative export method is to use the **save_single_frameset** instruction to save each individual frame as a bag file instead of an image file.

https://github.com/IntelRealSense/librealsense/issues/2588#issuecomment-453757755

MartyG-RealSense, 2021-07-27:
Hi @chain-tech  Do you require further assistance with this case, please?  Thanks!

chain-tech, 2021-07-30:
Hi @MartyG-RealSense , Thanks for your help on this so far. Closing the ticket now.

closed 2021-07-30 by chain-tech